lvgl/lvgl #6684: lv_canvas_set_px() shows no effect on framebuffer

tobiasjakobi-compleo, 2024-08-14:
### LVGL version

v9.1.0-381-g90cfb6c61

### What happened?

Hello,

having some trouble here with LVGL Git master. Actually these are most likely two problems.

The first is: lv_canvas_set_px() seemingly has no effect on my canvas and/or framebuffer.
The second ist: lv_draw_rect() is broken for pixel-sized rectangles (i.e. width and height equals 1)

I've attached some code from our application. Yes, I'm well aware that LVGL also includes a QR code subsystem, but doing these draw operations should work nonetheless.
Anyway, I should probably explain what works at the moment. If we set `kQRCodeScalingFactor` to `2`, then I get proper rendering.

For testing purpose we have an IMGui-based application that shows us the LVGL framebuffer. This is how it looks like:
[image]

The IMGui part is irrelevant here though, one could just as well show the fb by different means. It's just for convenience.

If I keep trying to drawing stuff using lv_draw_rect(), but I have `kQRCodeScalingFactor` set to `1`, i.e. the rects are pixels, then nothing is rendered. Hence I thought, might as well use lv_canvas_set_px() to set these pixels in this case. Turns out that then nothing is rendered as well. The framebuffer just stays white at the position where the QR code should appear.

I don't want to rule out that I'm doing something horribly wrong here, in particular with the API changes that come with LVGL 9. I can see that lv_canvas_set_px() does call invalidate on the canvas object. But maybe I need to do more here?

With best wishes,
Tobias

### How to reproduce?

```
const auto code = qrcodegen::QrCode::encodeText(qrCodeInput.c_str(), qrcodegen::QrCode::Ecc::HIGH);

const int code_size = code.getSize();
const std::int32_t size = 2 * kQRCodeBorderWidth + code_size * kQRCodeScalingFactor;

canvas_ = ::lv_canvas_create(screen_active_);
canvas_buf_.resize(::lv_canvas_buf_size(size, size, kColorBitsPerPixel, LV_DRAW_BUF_STRIDE_ALIGN));

::lv_canvas_set_buffer(canvas_, ::lv_draw_buf_align(&canvas_buf_[0], kColorFormat), size, size, kColorFormat);
::lv_obj_align(canvas_, left_align ? ::LV_ALIGN_TOP_LEFT : ::LV_ALIGN_TOP_RIGHT, 0, 0);

::lv_canvas_fill_bg(canvas_, ::lv_color_white(), ::LV_OPA_COVER);

if (kQRCodeScalingFactor != 1) {
  ::lv_draw_rect_dsc_init(&draw_rect_);
  ::lv_canvas_init_layer(canvas_, &layer_);

  draw_rect_.bg_color = ::lv_color_black();

  for (int code_y = 0; code_y < code_size; ++code_y) {
    const auto pos_y = kQRCodeBorderWidth + code_y * kQRCodeScalingFactor;

    draw_area_.y1 = pos_y;
    draw_area_.y2 = pos_y + kQRCodeScalingFactor - 1;

    for (int code_x = 0; code_x < code_size; ++code_x) {
      if (code.getModule(code_x, code_y)) {
        const auto pos_x = kQRCodeBorderWidth + code_x * kQRCodeScalingFactor;

        draw_area_.x1 = pos_x;
        draw_area_.x2 = pos_x + kQRCodeScalingFactor - 1;

        ::lv_draw_rect(&layer_, &draw_rect_, &draw_area_);
      }
    }
  }

  ::lv_canvas_finish_layer(canvas_, &layer_);
} else {
  // LVGL has problems when drawing rectangles that have pixel-size.
  //
  // Set the pixels through the canvas directly in this case.

  // TODO: this draws nothing
  // TODO: try using lv_canvas_get_draw_buf() or lv_canvas_copy_buf()
  for (int code_y = 0; code_y < code_size; ++code_y) {
    const auto pos_y = kQRCodeBorderWidth + code_y;

    for (int code_x = 0; code_x < code_size; ++code_x) {
      if (code.getModule(code_x, code_y)) {
        const auto pos_x = kQRCodeBorderWidth + code_x;

        // Opacity is not used, since we don't have a canvas with alpha component.
        ::lv_canvas_set_px(canvas_, pos_x, pos_y, ::lv_color_black(), 0);
      }
    }
  }
}
```

kisvegabor, 2024-08-24:
Hi,

Although I don't know the value of `kColorFormat` I've tried this simple code
```c
  LV_DRAW_BUF_DEFINE_STATIC(buf1, 20, 20, LV_COLOR_FORMAT_RGB565);
  LV_DRAW_BUF_INIT_STATIC(buf1);

  lv_obj_t * canvas = lv_canvas_create(lv_screen_active());
  lv_canvas_set_draw_buf(canvas, &buf1);
  lv_obj_center(canvas);
  lv_canvas_fill_bg(canvas, lv_color_white(), LV_OPA_COVER);

  lv_canvas_set_px(canvas, 5, 10, lv_color_hex(0xff0000), LV_OPA_COVER);
```

and it worked well:

[image]

tobiasjakobi-compleo, 2024-08-27:
@kisvegabor Which Git master did you test this with? Please let me know the hash. The color format is `LV_COLOR_FORMAT_RGB565`, but I've also reproduced this with the 32-bit formats. Have you also checked with `lv_draw_rect()`?

Any ideas how to debug this?

kisvegabor, 2024-08-29:
It worked with the `v9.2.0` tag. Have you tried with exactly [this code](https://github.com/lvgl/lvgl/issues/6684#issuecomment-2308150602)? 

Does it work if you manually set some values in the buffer? E.g. `buf1.data[33] = 0x85`

tobiasjakobi-compleo, 2024-09-17:
My plan is to continue working on this issue starting end of this week. Hopefully going to provide some feedback then.

tobiasjakobi-compleo, 2024-09-24:
OK, so I went to rewrite most of the code that related to this bug. Removed lots of legacy cruft. E.g. the canvas object was destroyed and then recreated for each draw operation. My guess is that this messed up the order in which objects were drawn in the first place.

Anyway, after this rewrite both bugs are now gone:
- `lv_canvas_set_px()` works like intended
- `lv_draw_rect()` works also for pixel-sized rects

Sorry for the noise! :cry:

closed 2024-09-24 by tobiasjakobi-compleo

kisvegabor, 2024-09-25:
Cool! Thank you for the feedback!