lvgl/lvgl #6687: lv_obj_set_flag and clear_flag should check if the flag is already set/unset

lorddevereux, 2024-08-15:
### Introduce the problem

I recently ran into a case where a rogue function was continually calling **lv_obj_remove_flag(widget, LV_OBJ_FLAG_HIDDEN)** on a few widgets.
Every call causes the object to be invalidated and therefore redrawn even though it hasn't actually changed. This caused a massive performance impact (3 FPS -> 30 FPS!). Whilst it should be up to the user to make sure they don't call the function unnecessarily, I think patching to prevent this will save someone debugging in future.

Posting as a discussion before submitting PR in case someone think of a good reason not to do it!

### Proposal

Add to lv_obj_add_flag ~line 233
`if (obj->flags & f)`
`        return;`

Add to lv_obj_remove_flag ~line 270
`if (!(obj->flags & f))`
`        return;`

kisvegabor, 2024-08-29:
Great catch!  Could you send a Pull request?

Just a minor change for `add` as it should work with ORed flags too (e.g. add `HIDDEN | IGNORE_LAYOUT`):
```c
if ((obj->flags & f) == f) /*Check if all flags are set*/
        return;
```

kisvegabor, 2024-09-13:
No stale

jurstu, 2024-09-15:
I believe I made the proposed changes. PR coming.

closed 2024-09-17 by kisvegabor