lvgl/lvgl #6690: Ability to permanently disable back button on all menus

conor-third, 2024-08-15:
### Introduce the problem

Currently you can disable the root back button for the root menu page but nested menus where the depth is >= 2 will always draw a back button. I have a particular use case where the navigation of the menus is done elsewhere and the back button is not required.

### Proposal

Potentially change the following line in lv_menu.c:

`if(menu->cur_depth >= 2 || menu->mode_root_back_btn == LV_MENU_ROOT_BACK_BUTTON_ENABLED)`

to use '&&' instead of '||'. The caveat being that you will have to call "lv_menu_set_mode_root_back_button" if you do want to have the built in back button for nested menus/pages. There potentially could be other side effects to this but I haven't gave it extensive thought.

kisvegabor, 2024-08-29:
Makes sense to me. Could you send a Pull request with suggested change?

kisvegabor, 2024-09-18:
Finally the PR is not merged as it would break backward compatibility. Thank you for looking into it anyways! :+1:

closed 2024-09-18 by kisvegabor