lvgl/lvgl #6696: Add dithering support to gradients

kisvegabor, 2024-08-17:
### Introduce the problem

Gradients can look odd with RGB565 color format. 

### Proposal

@kdschlosser discovered a simple way of handling dithering. See [here](https://github.com/lvgl/lvgl/issues/5764#issuecomment-1982784368). The best parts are:
- It's simple
- The current pixel's value doesn't depend on the previous pixels. Therefore we can render only parts of the gradient without any issue.

When the current gradient engine was implemented @zjanosy did some exploration and as far as I know it was pretty successful. Could you share your finding with us?

@hailrake-322 was interested in picking up this topic an working on the implementation.

zjanosy, 2024-08-19:
> When the current gradient engine was implemented @zjanosy did some exploration and as far as I know it was pretty successful. Could you share your finding with us?

Yes, I have implemented the dithering and integrated it into the rendering. It is working pretty well. I will create a PR.

zjanosy, 2024-09-04:
@kisvegabor Is the talk by Ivan not so interesting so meanwhile you can update Github issues? :-))

Back to the topic: it seems that there is some more to make a PR, because we'd need to add configuration settings and likely update the test cases as well. So I could send you (or @hailrake-322) the implementation (I think 3 source files), and you or someone could integrate it into the current LVGL project.

What do you think?

kisvegabor, 2024-09-04:
Opening a PR even for a half ready but functional implementation would be very useful.

hailrake-322, 2024-09-04:
@zjanosy @kisvegabor I don't have much free time right now, but I would like to try integrating dithering

zjanosy, 2024-09-04:
@hailrake-322 There are two sets of dithers, currently selectable with #if 0/1. The original dither caused a somewhat visible pattern, although it was symmetrical (which I think is an advantage because it does not introduce DC shift). I modified the pattern a little. Now it is unfortunately not as symmetrical as it used to be, however, it looks smoother for my eyes. Your mileage may vary.

The dithering can now be configured only at compile time in lv_conf.h, i.e., it is not possible to apply dither selectively only to the gradients.

Currently dither is added to both the software fill and RGB888 images.

kisvegabor, 2024-09-04:
Thank you @zjanosy :heart:

kdschlosser, 2024-09-04:
The `dither_rgb888_to_rgb565` is not just doing a dither It is also doing a conversion from RGB888 to RGB565 at the same time. Dithering an RGB888 buffer from 24 bit color to 16 bit color doesn't require converting the buffer from RGB888 to RGB565.

The function to dither from 24 bit color to 16 bit color is an in place modification of the RGB888 buffer data and it should not require having to enable or disable it via a macro. There are use cases where doing the dither without wanting to convert to RGB565. For example. if you have an RGB888 display but you want to use 16 bit images to lessen the amount of flash used to store the images.. STM32 MCU's don't have a a lot of available flash so this is something that a person may want to do.

Another thing is the `magic_dither` array. since you are going the conversion of an entire buffer using that array can the array be allocated on the stack? It only needs to be made available when the dithering is being done. I know that 256 bytes is not a HUGE amount of memory but it ends up being memory that can be used for other purposes when he function is not in use. I cannot imagine the amount of time it takes to create the buffer is going to make any kind of a sizeable impact in performance if it is done inside of the function on the stack.

zjanosy, 2024-09-04:
@kdschlosser 
> The function to dither from 24 bit color to 16 bit color is an in place modification of the RGB888 buffer data and it should not require having to enable or disable it via a macro.
The reason why I've added a define is that in the current version (i.e., in my implementation) dithering is built into the software renderer. It is not just a function which could be stripped by the linker if not used.
> Another thing is the magic_dither array. since you are doing the conversion of an entire buffer using that array can the array be allocated on the stack?
I'm not fond of allocating large structures on the stack, because you never know how deeply your call will be nested. Stack space is limited, especially if there are many tasks (because each task needs a separate stack, and SRAM is scarce). In general we have more flash than SRAM.
I did not do too much optimization, so maybe we could shave some cycles here and there. Nevertheless, the current code is working reasonably well on an ESP32-S3 rendering a gradient background on a 480x320 screen. The visuals are clearly improved by dithering, although there is still some banding with certain dark colors.

kdschlosser, 2024-09-04:
out of question, why did the size of the magic array grow from 192 bytes to 256 bytes?

kdschlosser, 2024-09-05:
I was looking at the code and it does look like the pixels are still being processed one at a time. The inner workings of LVGL is quite complex and it takes a bit for me to follow the data through the code. 

If the magic function was allocated on the stack it would be done in the `dither_rgb888_to_rgb565` function for the reasons you just stated.. but instead it would be done in the `lv_draw_sw_blend_color_to_rgb565` or the `rgb888_image_blend_with_dither` functions and then a pointer to the array would be passed to the `dither_rgb888_to_rgb565` function. This way it is not being created over and over again on the stack. It gets created a single time and then used for each and every pixel. I don't know how many times `lv_draw_sw_blend_color_to_rgb565` or `rgb888_image_blend_with_dither` gets called when updating a single widget.

I am actually really eager to give the whole gradient thing a try as I have been wanting conical and radial gradients for a while now...I am more interested in rendering gradients using the canvas and I cannot figure out how to go about doing that...

closed 2024-09-30 by lvgl-bot

FloHomi, 2024-11-21:
Hi There,
is there any update on this topic?
I am really interested and would like to use it right now :)

kisvegabor, 2024-11-21:
I've just contacted @zjanosy if he is open to pick it up again :crossed_fingers: