lvgl/lvgl #6698: HW rotation of the screen is hardcoded in tft_espi

Chalta, 2024-08-18:
### LVGL version

v9.0.0

### What happened?

It seems that that HW rotation of the screen is hardcoded in libraries\lvgl\src\drivers\display\ tft_espi

`dsc->tft->setRotation(3);   /* Landscape orientation, flipped */ `

So defining an LC_DISPLAY_ROTATION_90 or _270 makes no difference to the display orientation:

```
//esp32 v2.0.17 by Espressif Systems
#include <lvgl.h>         // v9.0.0 by kisvegabor
#include <TFT_eSPI.h>     // v.2.5.0    by Bodmer

#define TFT_ROTATION  LV_DISPLAY_ROTATION_90

void setup() {

....

lv_display_set_rotation(disp, TFT_ROTATION); //has no effect

...

}
```
But changing the line in the lib file `dsc->tft->setRotation(0) ;   /* Landscape orientation, UNflipped */ ` does work.

TFT-eSPI #define ILI9341_DRIVER // Generic driver for common displays

More here:
https://forum.lvgl.io/t/implicit-display-roation-when-using-tft-espi-library/15473

kisvegabor, 2024-08-21:
It should be handled [here](https://github.com/lvgl/lvgl/blob/master/src/drivers/display/tft_espi/lv_tft_espi.cpp#L85).

Does it work if you use LVGL's  `master` branch?

Chalta, 2024-08-28:
On 9.2 at least, it seems to work as intended, without modifications required!

I don't think it shouldn't make a difference other than simplicity, but I also made sure that I'm using directly and not redefining TFT_WIDTH and TFT_HEIGHT which are defined in TFT_eSPI\User_Setups from the device chosen in TFT_eSPI\User_Setup_Select.h, 

 disp = lv_tft_espi_create(**TFT_WIDTH**, **TFT_HEIGHT**, draw_buf, sizeof(draw_buf));

kisvegabor, 2024-08-29:
Do we use `TFT_WIDTH` and `TFT_HEIGHT` in LVGL somewhere or is it more of a note for future readers?

closed 2024-08-31 by Chalta