lvgl/lvgl #6700: Arabic with vowels is not displayed correctly.

jihunluv, 2024-08-19:
### LVGL version

v9.1.0

### What happened?

I have no sense of Arabic characters but found an issue with rendering vowels. 

Below is GIMP output of string اللُّغَة

[image]

Below is Freetype enabled  LVGL v9.1.0 (LV_USE_BIDI  / LV_USE_ARABIC_PERSIAN_CHARS flags are on)
[image]

Can you help with this problem?

jihunluv, 2024-08-19:
I guess related issue is already discussed in below topic.

https://forum.lvgl.io/t/alternative-text-shaping-engine/12414

Any support in near future?

kisvegabor, 2024-08-30:
LVGL's typesetting engine is okay, but don't cover all the cases. I've just opened an [issue](https://github.com/lvgl/lvgl/issues/6763) for the planned tasks of v9.3. Reworking the typesetting engine is not on the list now, but it will consider it for v9.4.

closed 2024-09-23 by lvgl-bot

jihunluv, 2024-09-23:
It is great news that you will consider the feature at least in v9.4. can you guess the timeline?

kisvegabor, 2024-09-23:
v9.3 is for early 2025, so v9.4 or v10 can come in the second half of 2025. But it's a large feature, I can promise anything now. :slightly_frowning_face: