lvgl/lvgl #6701: Support for LV_COLOR_FORMAT_I1 is incomplete

amanita-citrina, 2024-08-19:
### LVGL version

Commit 790aa5a529d1ac37afbb0a18c33a9ecf4d64bff8

### What happened?

Commit 790aa5a529d1ac37afbb0a18c33a9ecf4d64bff8 from 19-Jul-2024 added support for 1-bit color depth. I tried to use it for my 296x128 EPaper display. However, I encountered several problems.

1. Defining `LV_COLOR_DEPTH 1` in lv_conf.h resulted in compile errors. This was fixed (I think) by modifying lv_color.h:
[image]
2. I provided an update buffer of size 296*128/8=4736 bytes to `lv_display_set_buffers()`, which is enough for a full update of the screen at 1 bit/pixel. However each call of `flush_cb()` would only update an area of 296x4 pixels.
3. The first 8 bytes of `pxmap` in `flush_cb` were unassigned and must be skipped (bytes AA below). The correct pixel data begins afterwards.
[image]

The problem still exists in recent commit c52aa1012898bca217e35f4d5e632e6f73667d25.

### How to reproduce?

1. In `lv_conf.h`, `#define LV_COLOR_DEPTH 1`.
2. In `lv_color.h`, apply the modification described above.
3. Build and run with `main.c` below.
    - The terminal output shows that in each call of `flush_cb()`, only 296x4 pixels are updated.
    - View `screenout.bin` in a hex editor. On each call of `flush_cb()`, the first 8 bytes are AA, which is the initial buffer value. Therefore, these bytes are never written by LVGL.
4. Uncomment the line `pxMap += 8;` in main.c and repeat. Now the pixel data in `screenout.bin` is correct.

```
#include "lvgl/lvgl.h"
#include <time.h>
#include <stdint.h>
#include <string.h>
#include <stdio.h>

#if LV_COLOR_DEPTH!=1
    #error Set LV_COLOR_DEPTH=1 in lv_conf.h
#endif

// --------------------------------------
// Local const and variables
// --------------------------------------
static const uint16_t DISP_WIDTH  = 296u;
static const uint16_t DISP_HEIGHT = 128u;
static const uint32_t BUFFER_SIZE_BYTES = DISP_WIDTH*DISP_HEIGHT / 8u;

// --------------------------------------
// Local function declarations
// --------------------------------------
static void hal_init(void);
static void hello_world(void);
static void flush_cb(lv_display_t* display, const lv_area_t* area, uint8_t* px_map);

// --------------------------------------
// Function definitions
// --------------------------------------
static void hal_init(void)
{
    static uint8_t m_buffer[296*128/8];
    (void)memset(m_buffer, 0xAA, sizeof(m_buffer));

    lv_group_set_default(lv_group_create());
    lv_display_t* const display = lv_display_create(DISP_WIDTH, DISP_HEIGHT);
    lv_display_set_buffers(display, m_buffer, NULL, sizeof(m_buffer), LV_DISP_RENDER_MODE_PARTIAL);
    lv_display_set_flush_cb(display, &flush_cb);
    lv_display_set_default(display);
}

static void hello_world(void)
{
    lv_obj_t* const label = lv_label_create(lv_screen_active());
    lv_label_set_text(label, "Hello world!");
    lv_obj_set_style_text_color(lv_screen_active(), lv_color_hex(0x000000), LV_PART_MAIN);
    lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
}

static void flush_cb(lv_display_t* display, const lv_area_t* area, uint8_t* px_map)
{
    static uint8_t m_fileBuffer[296*128/8];

    printf("Area: %d %3d %3d %3d\n", area->x1, area->x2, area->y1, area->y2);

    // Skip 8 bytes that are unassigned
    // px_map += 8;
    for (int32_t y=area->y1; y<=area->y2; y++)
    {
        const uint32_t offset = ((y*DISP_WIDTH) + area->x1) / 8u;
        uint8_t* pBuffer = &m_fileBuffer[offset];
        for (int32_t x=area->x1; x<=area->x2; x+=8u)
        {
            *pBuffer = *px_map;
            ++pBuffer;
            ++px_map;
        }
    }

    if (lv_display_flush_is_last(display))
    {
        FILE* const fp = fopen("screenout.bin", "wb");
        if (fp!=NULL)
        {
            (void)fwrite(m_fileBuffer, 1u, BUFFER_SIZE_BYTES, fp);
            (void)fclose(fp);
        }
    }
    lv_display_flush_ready(display);
}

int main(void)
{
    lv_init();
    hal_init();
    hello_world();

    while (1)
    {
        const struct timespec time_5ms = {0u, 5000000u};
        lv_timer_handler();
        (void)nanosleep(&time_5ms, NULL);
    }

    return 0;
}
```

C47D, 2024-08-19:
Hi,

Dumb question, did you also set `LV_DRAW_SW_SUPPORT_I1` to 1?

amanita-citrina, 2024-08-20:
Yes, apart from LV_COLOR_DEPTH=1 it differs from lv_conf_template.h in just two minor points:
- LV_BUILD_EXAMPLES=0
- Font is Montserrat24 instead of Montserrat14

kisvegabor, 2024-08-21:
Hi,

I've opened a PR to fix these, but couldn't test them on hardware. 
https://github.com/lvgl/lvgl/pull/6714

Could you take a look?

amanita-citrina, 2024-08-21:
> [Verfolgungsbild][https://github.com/notifications/beacon/ABKK5XHDQHZD3CB3MWQIBYTZSROELA5CNFSM6AAAAABMXNKWQGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUJF7P22.gif]

Thanks. I'll try tomorrow if I have time.

amanita-citrina, 2024-08-22:
I checked the PR. The compile break has been resolved. Unfortunately, the other problems persist.

1. There are still 8 unassigned bytes (0xAA) at the beginning of `px_map` on each call of `flush_cb()`.
2. **NEW:** Every 37th byte is erroneously set to 0x00
3. The size of the provided buffer is 4736 bytes (296\*128/8), which is exactly the size needed to represent all 296\*128 pixels with one bit each. Nevertheless, `flush_cb()` updates only 296\*124 pixels in the first call, and the remaining 296\*4 pixels on the second call.

screenout.bin from main.c (as in the issue description):
[image]

By the way, may I suggest to update `lv_conf_template.h`, line 29, with the info about I1:
`/*Color depth: 1 (I1), 8 (A8), 16 (RGB565), 24 (RGB888), 32 (XRGB8888)*/`

liamHowatt, 2024-08-22:
Regarding the `px_map += 8`, those eight bytes are the color palette. It maps each possible pixel value (only 0 and 1 for I1) to a `lv_color32_t`. Skipping it is the correct thing to do in your case. Another way to skip it is `px_map += LV_COLOR_INDEXED_PALETTE_SIZE(LV_COLOR_FORMAT_NATIVE) * sizeof(lv_color32_t)`.

amanita-citrina, 2024-08-22:
OK. Is this special to I1? I did not notice this offset for LV_COLOR_DEPTH 8 and 16. Also, why is the color palette transferred in every partial update and not just the first one? (Actually, that makes it easier to skip.)

Nevertheless, it seems that the color palette is not set at all - the bytes are at their initial value 0xAA.

liamHowatt, 2024-08-22:
It's special to indexed color formats: I1, I2, I4, and I8. It would stay uninitialized when used as a display buffer I think. One use case of the palette is when a canvas with an indexed color format buffer is being rendered to an RGB display. Index values 0 and 1 can be assigned palette values (like black and green) using `lv_canvas_set_palette`.

kisvegabor, 2024-08-23:
In LVGL there is no difference between rendering a screen, rendering to layer, or to canvas. When not rendering to the screen with I1 format you need the palette so for consistency we handle the palette in this case too.  

 We can could skip the palette before flushing, but you would still need to provide a buffer large enough for the palette too. I'll update the docs in  https://github.com/lvgl/lvgl/pull/6714

xlla, 2024-08-23:
hi @kisvegabor , should I add this palette space to the buffer that assigned to function lv_display_set_buffers, 
for example, screen size 128x64 ,  factor 8, I1 format,  the buffer size should be 128 or 128 + 8

kisvegabor, 2024-08-23:
What does "factor 8" mean? Anyway, it should be `128 + 8`.

xlla, 2024-08-26:
the "factor 8" means use 1/8 of full screen buffer size.

kisvegabor, 2024-08-26:
Ah, I see. 

As https://github.com/lvgl/lvgl/pull/6714 works well I close this issue.

PR #6714 (2024-08-26): fix(i1): fix compiler and runtime issues with I1 rendering
### Description of the feature or fix

Fixes #6701

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig [... cut]
closed 2024-08-26 by kisvegabor