lvgl/lvgl #6702: Different behavior regarding LV_EVENT_VALUE_CHANGED event when step is changed in Spinbox

mvladic, 2024-08-19:
### LVGL version

v9.1 v8.4

### What happened?

When step in Spinbox is changed with the encoder then LV_EVENT_VALUE_CHANGED event is sent, but when it is changed with the touch interface no LV_EVENT_VALUE_CHANGED event is sent.

****

### How to reproduce?

See below, first I change Spinbox step with the encoder button and in LV_EVENT_VALUE_CHANGED event handler I log the step change, but when I press with the touch the step position is changed but no LV_EVENT_VALUE_CHANGED event is sent, thus nothing is printed in the log.

[image]

mvladic, 2024-08-19:
Here is the code responsible for this difference in behaviour:

[image]

kisvegabor, 2024-08-21:
Hi,

We will release v9.2 on Monday (August 26). Can you send a PR to fix this issue? If so, I'll try my best to include it in the release.

liamHowatt, 2024-08-23:
Fixed by #6719

closed 2024-08-23 by liamHowatt