lvgl/lvgl #6704: ESP32 IntegerDivideByZero exception since PR #6427

mr-sven, 2024-08-19:
### LVGL version

v9.x master

### What happened?

PR #6427 broke widget demo on ESP32.
```
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.
img_draw_core at /home/sven/esp32/esp32-8048S050C/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_img.c:318 (discriminator 5)
img_decode_and_draw at /home/sven/esp32/esp32-8048S050C/managed_components/lvgl__lvgl/src/draw/lv_draw_image.c:285
lv_draw_image_normal_helper at /home/sven/esp32/esp32-8048S050C/managed_components/lvgl__lvgl/src/draw/lv_draw_image.c:178
lv_draw_sw_image at /home/sven/esp32/esp32-8048S050C/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw_img.c:169
execute_drawing at /home/sven/esp32/esp32-8048S050C/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.c:479
execute_drawing_unit at /home/sven/esp32/esp32-8048S050C/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.c:344
render_thread_cb at /home/sven/esp32/esp32-8048S050C/managed_components/lvgl__lvgl/src/draw/sw/lv_draw_sw.c:451
prvRunThread at /home/sven/esp32/esp32-8048S050C/managed_components/lvgl__lvgl/src/osal/lv_freertos.c:360
vPortTaskWrapper at /home/sven/esp/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

I stepped through the commits and after merge of PR #6427 it crashes.

### How to reproduce?

Analytics Tab -> scroll down to the circle graphs

C47D, 2024-08-19:
Hi,

The division is marked with a comment. @XuNeo  is it expected to have a buffer stride equal to 0? `blend_w` and/or `lv_color_format_get_size(cf_final)` must be 0 for this to happen.

```c
        int32_t src_w = lv_area_get_width(img_coords);
        int32_t src_h = lv_area_get_height(img_coords);
        int32_t blend_w = lv_area_get_width(&blend_area);
        int32_t blend_h = lv_area_get_height(&blend_area);

        lv_color_format_t cf_final = cf;
        if((cf == LV_COLOR_FORMAT_L8) && (draw_dsc->recolor_opa > LV_OPA_MIN)) {
            cf_final = LV_COLOR_FORMAT_ARGB8888;
        }
        if(transformed) {
            if(cf_final == LV_COLOR_FORMAT_RGB888 || cf_final == LV_COLOR_FORMAT_XRGB8888) cf_final = LV_COLOR_FORMAT_ARGB8888;
            else if(cf_final == LV_COLOR_FORMAT_RGB565) cf_final = LV_COLOR_FORMAT_RGB565A8;
            else if(cf_final == LV_COLOR_FORMAT_L8) cf_final = LV_COLOR_FORMAT_AL88;
        }

        uint8_t * tmp_buf;
        uint32_t px_size = lv_color_format_get_size(cf_final);
        int32_t buf_h;
        if(cf_final == LV_COLOR_FORMAT_RGB565A8) {
            uint32_t buf_stride = blend_w * 3;
            buf_h = MAX_BUF_SIZE / buf_stride;
            if(buf_h > blend_h) buf_h = blend_h;
            tmp_buf = lv_malloc(buf_stride * buf_h);
        }
        else {
            uint32_t buf_stride = blend_w * lv_color_format_get_size(cf_final);
            buf_h = MAX_BUF_SIZE / buf_stride;      // <----- Faulty division
            if(buf_h > blend_h) buf_h = blend_h;
            tmp_buf = lv_malloc(buf_stride * buf_h);
        }
        LV_ASSERT_MALLOC(tmp_buf);
```

mr-sven, 2024-08-20:
16bit, so `LV_COLOR_FORMAT_RGB565`

XuNeo, 2024-08-20:
Hi,
Which demo can reproduce this crash? Could you please provide the steps to reproduce the issue?

`buf_stride` should never be zero.

XuNeo, 2024-08-20:
Aha, I get. The widget demo, `Analytics Tab -> scroll down to the circle graphs `

I will check it now.

XuNeo, 2024-08-20:
Could you please try https://github.com/lvgl/lvgl/issues/6704?

mr-sven, 2024-08-20:
That works, thanks

closed 2024-08-21 by XuNeo