lvgl/lvgl #6707: No custom swipe possible in LVGL 9

HokageM, 2024-08-20:
### Introduce the problem

I used a custom setting for `LV_INDEV_DEF_SCROLL_LIMIT` and `LV_INDEV_DEF_SCROLL_THROW`.

In LVGL 9, these defines are located inside the .cpp file, which prevents customization. Additionally, there is no function available to set the `scroll_limit` and `scroll_throw` for an lv_indev_t object.

### Proposal

I would suggest a function to set scroll_limit and scroll_throw:
```
lv_indev_set_scroll_limit()
lv_indev_set_scroll_throw()
```

kisvegabor, 2024-08-21:
These setters are really missing.[ One was just added](https://github.com/lvgl/lvgl/pull/6664) in past days. Do have time and interest to contribute by adding API calls for the missing ones?

HokageM, 2024-08-22:
Done :)

closed 2024-08-23 by HokageM

kisvegabor, 2024-08-23:
Arigato, mighty Hokage-sama! The Hidden Leaf Village owes you a debt of gratitude! :blush: