lvgl/lvgl #6708: For 1 bit color format, lv_color_format_get_size maybe wrong

xlla, 2024-08-20:
### LVGL version

v9.1.0

### What happened?

I am try to implement the 1 bit color renderer using lvgl 9 with lv_micropython port for eink display.

to reduce memory usage, I choose color format LV_COLOR_FORMAT_I1, and I want to know if lvgl have treat this format correctly, so I dive into the source.

I found the function lv_color_format_get_size use uint8_t to storage the result.
it maybe right in some circumstance, but the accuracy result should 0.125 byte.

in lib/lv_bindings/lvgl/src/draw/lv_draw_buf.c
function lv_draw_buf_goto_xy use it to calculate the target data.

`return data + x * lv_color_format_get_size(buf->header.cf);`

will result wrong target data.
for example, x is 7 , it should +0’s byte instead of +7’s byte

maybe it should invoke the underlay function to calculate the correct index

`return data + (( x * lv_color_format_get_bpp(buf->header.cf) + 7) >> 3)`

I know nothing about the implement of the lvgl, just found this issue by accident.
I don’t know which logic will call lv_color_format_get_size to calculate the byte index again, for color format LV_COLOR_FORMAT_I1, it will get wrong result.

if lv_color_format_get_size can return 0.125 then everything is right, I know it is impossible, float is inflate for mcu.

xlla, 2024-08-20:
I found another use case, in lv_draw_buf.c , lv_draw_buf_clear 

`uint8_t px_size = lv_color_format_get_size(header->cf);`
`bufc = lv_draw_buf_goto_xy(draw_buf, a->x1, a->y1);`
`line_length = lv_area_get_width(a) * px_size;`

for color format LV_COLOR_FORMAT_I1 , line_length will be 8x large than actual size.

I think here should use **lv_color_format_get_bpp** instead of _lv_color_format_get_size_

kisvegabor, 2024-08-21:
Hey,

In `master` we have already support `I1` rendering and these issues should be fixed as well. Could you take a look at it?

xlla, 2024-08-23:
Hi @kisvegabor , I'v merged related PR #6313 ,  #6345 ,  #6441 ,  #6618 , #6714,  the function lv_draw_buf_goto_xy got fixed, but the function lv_draw_buf_clear does not.

<pre>
    uint8_t px_size = lv_color_format_get_size(header->cf);
    uint8_t * buf = lv_draw_buf_goto_xy(draw_buf, a_clipped.x1, a_clipped.y1);
    uint32_t line_length = lv_area_get_width(&a_clipped) * px_size;
</pre>

as I mentioned above

> for color format LV_COLOR_FORMAT_I1 , line_length will be 8x large than actual size.

`lv_memzero(buf, line_length);`

for example,  a_clipped.x1 =0, a_clipped.x2 =7,  lv_area_get_width will return 8 , line_length will be 8 Bytes instead of 8 bit ( 1 Byte)  

in worst case, lv_memzero will erase buf out of range, maybe crash.

xlla, 2024-08-23:
another strange problem is the callback function flush_cb was called with wrong parameter area.x2, it is 1 more   
 than actual, I'v create  display_drv with width=128, height=64, buffer size is 1/8 of actual screen size.
start_x=0, start_y=0,

<pre>
    def flush(self, disp_drv, area, color_p):
        x1 = area.x1 + self.start_x
        x2 = area.x2 + self.start_x
        y1 = area.y1 + self.start_y
        y2 = area.y2 + self.start_y
        w = x2 - x1 + 1
        wb = (w + 7) >> 3 #bytes size of row, round up
        h = y2 - y1 + 1
        if h > 0 and w > 0:
            print('flush {}, {}, {}, {}, wb {}'.format(x1,y1,w,h, wb))
</pre>

I'v got these log output:
<pre>
flush 0, 0, 129, 7, wb 17
flush 0, 7, 129, 7, wb 17
flush 0, 14, 129, 7, wb 17
flush 0, 21, 129, 7, wb 17
flush 0, 28, 129, 7, wb 17
flush 0, 35, 129, 7, wb 17
flush 0, 42, 129, 7, wb 17
flush 0, 49, 129, 7, wb 17
flush 0, 56, 129, 7, wb 17
flush 0, 63, 129, 1, wb 17
</pre>

kisvegabor, 2024-08-23:
Thanks you @xlla. Both the rounding and `lv_draw_buf_clear` should be fixed in #6714

xlla, 2024-08-25:
> another strange problem is the callback function flush_cb was called with wrong parameter area.x2, it is 1 more
> than actual

I'v pull the latest PR-6714,  this error was fixed,  with factor 8,  the flush_cb was called 8 times instead of 10 times, and the parameter area is correct.

<pre>
flush 0, 0, 128, 8, wb 16
flush 0, 8, 128, 8, wb 16
flush 0, 16, 128, 8, wb 16
flush 0, 24, 128, 8, wb 16
flush 0, 32, 128, 8, wb 16
flush 0, 40, 128, 8, wb 16
flush 0, 48, 128, 8, wb 16
flush 0, 56, 128, 8, wb 16
oled refresh area 0, 0, 128, 64
</pre>

closed 2024-08-25 by xlla