lvgl/lvgl #6709: FPS degradation from v8.3 to v9.1

vedantifx, 2024-08-20:
### LVGL version

v9.1.0

### What happened?

I am trying out **lvgl music demo application** from lvgl directory. 
I am using default SW renderer, RGB565 dual buffer configuration.
with auto play and FPS moniter enabled las screen shows avg fps.

in v8.3 i used to get around **40 fps**
in v9.1 i am getting around **15fps**

I have attached lv_conf files for both versions please let me know if i am missing anything:
[lv_conf_v9.1.zip](https://github.com/user-attachments/files/16674926/lv_conf_v9.1.zip)

### How to reproduce?

for v9.1:
void lv_port_disp_init(void)
{
    lv_display_t * disp = lv_display_create(MY_DISP_HOR_RES, MY_DISP_VER_RES);
    lv_display_set_flush_cb(disp, disp_flush);
    lv_display_set_buffers(disp, disp_buf1, disp_buf2, sizeof(disp_buf1), LV_DISPLAY_RENDER_MODE_FULL);
}
for v8.3:
void lv_port_disp_init(void)
{
    static lv_disp_draw_buf_t buf;
    static lv_disp_drv_t disp_drv;

    lv_disp_draw_buf_init(&buf, disp_buf1, disp_buf2, (MY_DISP_HOR_RES *
            MY_DISP_VER_RES));

    lv_disp_drv_init(&disp_drv);

    disp_drv.draw_buf = &buf;
    disp_drv.flush_cb = disp_flush;
    disp_drv.hor_res = MY_DISP_HOR_RES;
    disp_drv.ver_res = MY_DISP_VER_RES;
    disp_drv.full_refresh = 1;

    lv_disp_drv_register(&disp_drv);
}

flsuh_cb, timmer_handler, tick_handler has same implementation in both cases

kisvegabor, 2024-08-21:
A few questions:
- What is the target hardware?
- Please enable `LV_USE_SYSMON` and `LV_USE_PERF_MONITOR`, run `lv_demo_benchmark()` and send a screenshot about the end screen.

vedantifx, 2024-08-21:
I am using infineon psoc edge e84 hardware , code is running on CM55.

Here are the benchmarking logs:
Benchmark Summary (9.1.0 )
Name, Avg. CPU, Avg. FPS, Avg. time, render time, flush time
Empty screen,  			71%, 29, 17, 4, 13
Moving wallpaper, 			79%, 19, 40, 25, 15
Single rectangle, 			66%, 32, 20, 5, 15
Multiple rectangles, 		67%, 30, 22, 7, 15
Multiple RGB images, 		79%, 17, 43, 28, 15
Multiple ARGB images, 		86%, 11, 71, 56, 15
Rotated ARGB images, 		95%, 3, 289, 274, 15
Multiple labels, 			79%, 19, 39, 24, 15
Screen sized text, 			27%, 28, 71, 56, 15
Multiple arcs, 				75%, 18, 38, 23, 15
Containers, 				87%, 12, 64, 49, 15
Containers with overlay, 	        92%, 6, 130, 115, 15
Containers with opa,		93%, 8, 107, 92, 15
Containers with opa_layer, 	94%, 4, 200, 185, 15
Containers with scrolling, 	89%, 9, 89, 74, 15
Widgets demo, 			88%, 9, 72, 57, 15
All scenes avg., 			79%, 15, 81, 67, 14

GorgonMeducer, 2024-08-26:
> code is running on CM55.

Since you are using CM55, if and only if you are suing Arm Compiler 6 or LLVM, you can try to enable arm-2d in both v8.3.x and v9.1.0
This will help to improve the performance.

kisvegabor, 2024-08-26:
From the data, it seems the memory bandwidth of RAM, flash or both is very slow. Is it really the case?

tdjastrzebski, 2024-09-13:
@vedantifx Were you able to locate the culprit? It seems like I may be facing a similar issue. See [here](https://github.com/lvgl/lvgl/pull/6691#issuecomment-2347046707) (https://github.com/lvgl/lvgl/pull/6691)

vedantifx, 2024-09-13:
@tdjastrzebski I found two issue in FPS calculation:
1. v8.3.5 considers only time to render the frame inside lv_timer_handler(). any delay we put in for loop of lv_timmer_handler() is not considered. where as v9.1 do considers this delay as well.

2. avarage FPS variable in v9.1 **fps_avg_total** is declared as uint32 instead of float which creates lot of precision error.

After correcting for these two things i am getting comparable performance in v8.3.5 and v9.1.0

tdjastrzebski, 2024-09-13:
@kisvegabor Do you think this may be the source of the of the problem I described in https://github.com/lvgl/lvgl/pull/6691?
This would mean actually there is no performance problem but measurement issue.

kisvegabor, 2024-09-13:
The [v8.3-with-v9-benchmark](https://github.com/lvgl/lvgl/tree/v8.3-with-v9-benchmark) branch and v9 should measure the fps the same way, so it shouldn't be the problem. Did you use tis branch?

@tdjastrzebski, in your case the difference was visible by eye too, right?

tdjastrzebski, 2024-09-13:
Yes, for 8.3 perf I used [v8.3-with-v9-benchmark](https://github.com/lvgl/lvgl/tree/v8.3-with-v9-benchmark) branch.
If it is visible? Well, good question. To make sure, I made two recordings with my cellphone and played side-by-side.
8.3 demo does not work faster per se compared to 9.3 but almost everything works way more smoothly.
https://drive.google.com/file/d/1RMzkAoqWDLCS9bkxtRCWKYlvV4DSH1cc/view?usp=sharing
https://drive.google.com/file/d/1h9JJMOwkQlBcUJ8O335-y5W6OHDLzvV3/view?usp=sharing

kisvegabor, 2024-09-16:
v9 is clearly slower on the video. See the moving wallpaper case for example. 
- In case of v9 [this part](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/blend/lv_draw_sw_blend_to_rgb565.c#L775-L780) copies the images. (I assume that you have RGB565 color format)
- In v8 [this one](https://github.com/lvgl/lvgl/blob/v8.3-with-v9-benchmark/src/draw/sw/lv_draw_sw_blend.c#L660-L665)

Could you measure the time of these `memcpy`s.)

tdjastrzebski, 2024-09-17:
@kisvegabor I use (X)RGB888 format both for LVGL and external display buffer.
After enabling `LV_USE_REFR_DEBUG` I can say that stripes are much wider for 8.3. This may provide some hint.
In both cases I use the same LVGL buffer size: `SCREEN_WIDTH * SCREEN_HEIGHT / 10`
[8.3 benchmark with stripes](https://drive.google.com/file/d/1QPoZvyYFQixnlH2whDf5ztVDUBTSWHxA/view?usp=sharing)
[9.3 benchmark with stripes](https://drive.google.com/file/d/1np3qGkdRRzVjbSRtzD4oFGZVINwv_ET3/view?usp=sharing)

kisvegabor, 2024-09-17:
Ahhhhh, In v9 the buffer size is in bytes, in v8 it was in pixels. 

So in v8 I guess you have something like:
```c
lv_color_t buf[SCREEN_WIDTH * SCREEN_HEIGHT / 10]; //lv_color_t is 32 bit
```

In v9 you should have:
```c
//lv_color_t is fixed to RGB888 in v9, so use uint8_t directly
uint8_t buf[SCREEN_WIDTH * SCREEN_HEIGHT / 10 * 4]; 
lv_display_set_buffers(disp, buf, NULL, sizeof(buf), LV_DISPLAY_RENDER_MODE_PARTIAL);
```

tdjastrzebski, 2024-09-17:
Yeah, thanks, that would explain the problem.
Below are the new results. 9.3 with DMA2D is still substantially slower but the difference is no longer so dramatic and it can be explained by 1/ DMA2D limited support, 2/ widgets being more refined.

Benchmark Summary (9.3.0 dev) |   |   |   |  
-- | -- | -- | -- | --
Name | Avg. CPU | Avg. FPS | Avg. time | render time | flush time
Empty screen | 85% | 28 | 28 | 19 | 9
Moving wallpaper | 98% | 2 | 360 | 350 | 10
Single rectangle | 29% | 28 | 5 | 1 | 4
Multiple rectangles | 44% | 28 | 14 | 13 | 1
Multiple RGB images | 94% | 13 | 58 | 48 | 10
Multiple ARGB images | 100% | 13 | 59 | 49 | 10
Rotated ARGB images | 100% | 0 | 1190 | 1170 | 20
Multiple labels | 86% | 12 | 89 | 81 | 8
Screen sized text | 12% | 28 | 9 | 9 | 0
Multiple arcs | 93% | 5 | 157 | 150 | 7
Containers | 100% | 5 | 159 | 155 | 4
Containers with overlay | 100% | 4 | 173 | 167 | 6
Containers with opa | 100% | 1 | 389 | 382 | 7
Containers with opa_layer | 100% | 1 | 516 | 507 | 9
Containers with scrolling | 100% | 3 | 315 | 306 | 9
Widgets demo | 100% | 2 | 331 | 323 | 8
All scenes avg. | 83% | 10 | 240 | 233 | 7
  |   |   |   |   |  
Benchmark   Summary (8.3.11 ) |   |   |   |  
Name | Avg. CPU | Avg. FPS | Avg. time | render time | flush time
Empty screen | 77% | 33 | 24 | 16 | 8
Moving wallpaper | 98% | 20 | 55 | 49 | 6
Single rectangle | 11% | 33 | 0 | 0 | 0
Multiple rectangles | 48% | 33 | 15 | 6 | 9
Multiple RGB images | 97% | 16 | 51 | 43 | 8
Multiple ARGB images | 100% | 16 | 51 | 42 | 9
Rotated ARGB images | 100% | 3 | 1097 | 1095 | 2
Multiple labels | 84% | 17 | 92 | 83 | 9
Screen sized text | 11% | 33 | 7 | 7 | 0
Multiple arcs | 96% | 7 | 132 | 123 | 9
Containers | 100% | 10 | 85 | 80 | 5
Containers with overlay | 100% | 10 | 93 | 88 | 5
Containers with opa | 100% | 10 | 93 | 89 | 4
Containers with opa_layer | 51% | 34 | 4 | 3 | 1
Containers with scrolling | 98% | 6 | 171 | 159 | 12
Widgets demo | 96% | 7 | 156 | 150 | 6
All scenes avg. | 79% | 18 | 132 | 127 | 5

kisvegabor, 2024-09-17:
Great! 
Moving wallpaper is still 7 times slower which is suspicious. Can you run the same without DMA2D to eliminate other config issues?

tdjastrzebski, 2024-09-17:
[image]

kisvegabor, 2024-09-17:
It's also interesting why the flush time is different.

tdjastrzebski, 2024-09-17:
Here are my results without DMA2D. There are some improvements in 'multiple labels'

Benchmark   Summary (9.3.0 dev) |   |   |   |   |  
-- | -- | -- | -- | -- | --
Name | Avg. CPU | Avg. FPS | Avg. time | render time | flush time
Empty screen | 99% | 14 | 69 | 60 | 9
Moving wallpaper | 100% | 2 | 403 | 394 | 9
Single rectangle | 42% | 28 | 12 | 12 | 0
Multiple rectangles | 93% | 16 | 56 | 52 | 4
Multiple RGB images | 100% | 3 | 274 | 262 | 12
Multiple ARGB images | 100% | 3 | 263 | 257 | 6
Rotated ARGB images | 100% | 0 | 1254 | 1242 | 12
Multiple labels | 94% | 8 | 110 | 103 | 7
Screen sized text | 21% | 26 | 53 | 52 | 1
Multiple arcs | 93% | 4 | 199 | 190 | 9
Containers | 100% | 3 | 224 | 217 | 7
Containers with overlay | 100% | 1 | 628 | 620 | 8
Containers with opa | 100% | 1 | 474 | 467 | 7
Containers with opa_layer | 100% | 1 | 581 | 578 | 3
Containers with scrolling | 100% | 2 | 403 | 392 | 11
Widgets demo | 100% | 2 | 385 | 378 | 7
All scenes avg. | 90% | 7 | 336 | 329 | 7
  |   |   |   |   |  
  |   |   |   |   |  
Benchmark Summary (8.3.11 ) |   |   |   |   |  
Name | Avg. CPU | Avg. FPS | Avg. time | render time | flush time
Empty screen | 99% | 20 | 55 | 46 | 9
Moving wallpaper | 100% | 9 | 145 | 137 | 8
Single rectangle | 27% | 33 | 6 | 6 | 0
Multiple rectangles | 92% | 21 | 46 | 42 | 4
Multiple RGB images | 100% | 10 | 112 | 104 | 8
Multiple ARGB images | 100% | 6 | 240 | 231 | 9
Rotated ARGB images | 100% | 3 | 1137 | 1122 | 15
Multiple labels | 89% | 13 | 158 | 153 | 5
Screen sized text | 15% | 33 | 18 | 18 | 0
Multiple arcs | 99% | 6 | 175 | 167 | 8
Containers | 100% | 6 | 170 | 167 | 3
Containers with overlay | 100% | 3 | 308 | 301 | 7
Containers with opa | 100% | 3 | 294 | 288 | 6
Containers with opa_layer | 75% | 32 | 17 | 15 | 2
Containers with scrolling | 99% | 3 | 302 | 294 | 8
Widgets demo | 97% | 4 | 276 | 269 | 7
All scenes avg. | 87% | 12 | 216 | 210 | 6

[image]

kisvegabor, 2024-09-17:
Pfff, still not good somehow. Moving wallpaper is still 3 times slower. Just to confirm, `LV_COLOR_DEPTH` is 32 in both cases right?  (`LV_COLOR_DEPTH 24` is supported in v9 but it's slower as it's harder to deal with 3 bytes)

---

I just noticed something:
- In v9 the cogwheel used in Moving wallpaper is fixed RGB565 so LVGL needs to convert the colors on the fly.
- In v8 it's set according to the `LV_COLOR_DEPTH` so it can perform a memcpy

Please try using this file in v9.
[img_benchmark_cogwheel_rgb.c.zip](https://github.com/user-attachments/files/17029792/img_benchmark_cogwheel_rgb.c.zip)

kisvegabor, 2024-09-18:
Huh, great! :slightly_smiling_face: 

```
Containers with opa_layer	75%	32	17	15	2
```
Is clearly a measurement issue in case of v8. It should take more time than "Containers with opa". How does it look like by eye? Especially "Containers with overlay" where v8 has 3 FPS but v9 has only 1.5 FPS (according to the benchmark).

Let's focus on SW render only for now.

kisvegabor, 2024-09-23:
By eye the "containers with scrolling" looks the same to me. I see the same "scanline" passing with roughly the same speed. 

I did my best to port the benchmark and render time measurement to to v8, but it can happen that there are some differences between v8 and v9 in this regard. "Fortunately" the container related tests are slow enough to count the FPS on your fingers, so we can rely on that too. :smile: 

I think opa layer is not working in v8, because you need to enable `https://github.com/lvgl/lvgl/blob/release/v8.4/lv_conf_template.h#L35`

tdjastrzebski, 2024-09-23:
Enabling [LV_COLOR_SCREEN_TRANSP](https://github.com/lvgl/lvgl/blob/release/v8.4/lv_conf_template.h#L35) helped.
As far as 'containers with overlay' and 'containers with opa' I dare to disagree.
Yes, speed is the same but in case of 8.3 containers move more smoothly, reported FPS confirms that.
Below are my current 9.3 results compared to 8.3
[image]

tdjastrzebski, 2024-09-23:
To be fair I have also compared RELEASE build performance; without DMA2D and with DMA2D enabled.
In such comparison rev 9.3 looks a bit better, no surprise, unless DMA2D is enabled (2nd table).

[image]

[image]

kisvegabor, 2024-09-25:
> ST GPU2D support would probably make a substantial difference. Is it in works?

Yes, I could just make it work on my end yesterday. @liamHowatt will finalize it this week/early next week.

---

So weren't your trying it  with `-O3` optimization so far? Compared to `-O0` `-O3` is a 2-3x boost. :slightly_smiling_face: 

> One result which repeatedly stands out is 'screen sized text' where 8.3 is much faster.

It's interesting because it's 2-3 times faster in v9 for sure.  It's confirmed from many sources. With `LV_DEBUG_REFR` do you see the same area is changing? Does both scroll the text? If not please make the text loner manually in the benchmark.

kisvegabor, 2024-10-07:
Probably it's because the rendering pipeline of v9 is more complex and the optimizer can eliminate most of the overheads.

kissa96, 2024-10-24:
Hi, is this issue still ongoing?

kisvegabor, 2024-10-30:
Closing it in lack of feedback. Please let us know if the optimization didn't solve the problem.

closed 2024-10-30 by kisvegabor