lvgl/lvgl #6711: Font license files in LVGL

yoshi-ogawa, 2024-08-20:
### Introduce the problem

LVGL has some fonts such as Montserrat and FontAwesome, but there is no license file in the repository. Developers who create an application program using LVGL with the default fonts need to check licenses of the fonts by themselves.

For example, Montserrat is licensed under the [SIL Open Font License, Version 1.1](https://github.com/JulietaUla/Montserrat/blob/master/OFL.txt). According to the license terms, the font may be bundled, redistributed and/or sold with any software, provided that each copy contains the above copyright notice and this license. So, developers need to introduce license files or something to comply with the license terms. If the license file of Montserrat is contained in LVGL source code, I think it would be helpful for developers.

### Proposal

I would like to suggest store license files for DejaVuSans, FontAwesome, Montserrat, SimSun, and unscii-8 in the LVGL repository.
It might be a good idea to store the license files in src/font/ or scripts/built_in_font/.

kisvegabor, 2024-08-21:
Thank you for pointing it out. Very true! 

I think `scripts/built_in_font/` is a better place. 

Do you have time to send a PR to add the licenses?

yoshi-ogawa, 2024-08-22:
> Do you have time to send a PR to add the licenses?

Yes, but I have to identify licenses for the fonts before that.
Although I searched on the internet, I have not found information about FontAwesome and SimSun so far. Do you have information about that?

- FontAwesome
According to the [website of FontAwesome](https://fontawesome.com/), free version and pro version of the font are available. But I could not identify whether 'FontAwesome5-Solid+Brands+Regular.woff' in LVGL is free version or pro version. If it is free version, the font would be licensed under [CC BY 4.0 License, SIL OFL 1.1 License, and MIT License](https://github.com/FortAwesome/Font-Awesome).
- SimSun
This font seems to be released with Windows OS, and permission may be necessary to redistribute the font. Unfortunately, I do not have information about that.

Regarding other three fonts, I found information as follows:
- DejaVuSans
License: it consists of multiple type of licenses.
License file: https://github.com/dejavu-fonts/dejavu-fonts/blob/version_2_35/dejavu-fonts/LICENSE
- Montserrat
License: SIL Open Font License 1.1
License file: https://github.com/JulietaUla/Montserrat/blob/v7.200/OFL.txt
- unscii-8
License: Public Domain
For information about this font, see http://viznut.fi/unscii/ .

kisvegabor, 2024-08-23:
`FontAwesome5-Solid+Brands+Regular.woff` is put together by me to have a single file for all the required icons. 

Probably we need to replace SimSun in this case. :slightly_frowning_face:

yoshi-ogawa, 2024-08-27:
Hi @kisvegabor ,
Do you remember the original font files included in FontAwesome5-Solid+Brands+Regular.woff?
I would like to know whether they are distributed under free license or not.

Regarding SimSun font, I received advice via email as follows.
> SimSun can be replaced with Source Han Serif. Both are Song style fonts, but while commercial use of SimSun likely requires licensing from Microsoft, Source Han Serif can be used for free.
> https://github.com/adobe-fonts/source-han-serif

Since the SimSun font is used in various samples and documents, it needs changes in many files.
I guess it should be fixed in another PR.

kisvegabor, 2024-08-30:
> Do you remember the original font files included in FontAwesome5-Solid+Brands+Regular.woff?
I would like to know whether they are distributed under free license or not.

It was a long time ago, but IIRC I've just downloaded the Free pack from [here](https://fontawesome.com/download) and used the fonts from the `webfonts` folder. 

> Regarding SimSun font, I received advice via email as follows.

Ah, that's fantastic! Who gave this advice?

> I guess it should be fixed in another PR.

Yes, definitely. Let's proceed step-by-step. 

---

Thank you so much for looking into this: :pray:

yoshi-ogawa, 2024-09-20:
I'm sorry for my slow response.

> It was a long time ago, but IIRC I've just downloaded the Free pack from [here](https://fontawesome.com/download) and used the fonts from the webfonts folder.

Thank you for the information.
I checked old font packages and found that FontAwesome5-Solid+Brands+Regular.woff in LVGL was generated by Font Awesome v5.9.0. The number of fontversion in FontAwesome5-Solid+Brands+Regular.woff, which can be checked by fc-query command, matched with the numbers in font files of Font Awesome v5.9.0.
And I found LICENSE.txt in the v5.9.0 font package was not enough to comply with the font licenses (see https://github.com/FortAwesome/Font-Awesome/issues/17048).
Fortunately, this issue has already been fixed in version v6.1.0 or later. So, I will introduce the license file from v6.1.0 with minor changes.

>> Regarding SimSun font, I received advice via email as follows.

> Ah, that's fantastic! Who gave this advice?

I got the advice from "Liu Yi" via notifications@github.com, but the comment was not left here for some reason.

Now, my questions have been cleared up. I'll create a PR for this.

yoshi-ogawa, 2024-10-03:
I investigated SimSun font and found that we should not use it in LVGL.

- SimSun & NSimSun font family
https://learn.microsoft.com/en-us/typography/font-list/simsun
- Font redistribution FAQ
https://learn.microsoft.com/en-us/typography/fonts/font-faq

The FAQ page says, "Apart from the document embedding rights described previously, you may not redistribute the Windows fonts. You may not copy them to other computers or servers, and you may not convert them to other formats, including bitmap formats, or modify them." So, we need to replace the font.

And also, I checked whether we could replace SimSun font with Source Han Serif font. 
At the moment, there seems to be no problem in the replacement. I will create another PR for the replacement.

kisvegabor, 2024-10-14:
Fantastic, thank you in advance!

closed 2024-11-09 by lvgl-bot