lvgl/lvgl #6715: how to use lv_vector_path_line_to and lv_draw_label together

xiangxud, 2024-08-21:
### Introduce the problem

in my test code 
static void DrawLineTest() {
    IGUIHANDLER* handler = pGlobalGUIHandler;
    if (handler == NULL) return;
    lv_fpoint_t pts1[] = { {200, 200}, {400, 200}, {450, 350}, {350, 150} };
    lv_fpoint_t pts[] = { {200, 200}, {300, 100}, {350, 350}, {400, 150}, {450, 250}, {500, 300} ,{550, 250} ,{350, 550} };
    if (handler->drawTextStyle) {
        handler->drawTextStyle(gGisFont, 0xFF0000, 0, LV_OPA_30);
    }

    if (handler->drawText) { //only this is not displaying
        handler->drawText("GIS地图引擎测试 for LVGL ver2.0", 20, 20);
    }
#if 1 
    if (handler->setVectorStrokeStyle) {
        handler->setVectorStrokeStyle(2, 0xff000, 1, LV_OPA_80);
    }
    if (handler->setVectorFillStyle) {
        handler->setVectorFillStyle( 2,0x0000ff, LV_OPA_80);
    }
    if (handler->fillPolygon) {
        handler->fillPolygon(pts, 8, 0, 0);
    }
#endif
#if 0
    if (handler->drawLineStyle) {
        handler->drawLineStyle(10,2,0xff00ff,LV_OPA_70);
    }
    if (handler->drawLineTo) {
        handler->drawLineTo(30, 20, 800, 600 );
    }
    GIS_RECTXY rect = { 50,200,50,200 };
    if (handler->setVectorFillStyle) {
        handler->setVectorFillStyle(2, 0x00ff00, LV_OPA_80);
    }
    if (handler->fillRectEx) {
       handler->fillRectEx(&rect);
    }
    GIS_RECTXY rect1 = { 150,550,200,600 };
    if (handler->drawRectEx) {
        handler->drawRectEx(&rect1);
    }
#endif
#if 0
    BitmapType* bmp;
    if (handler->getBitmap) {
        bmp = handler->getBitmap(1);
        if (handler->drawBitmap && bmp) {
            handler->drawBitmap(bmp, 200, 400, LV_OPA_50);
        }
    }

#endif

    lv_draw_vector(handler->vector_ctx);
#if 0 //open this the label is displaying ok
    if (handler->drawText) {
        handler->drawText("GIS地图引擎测试 for LVGL ver2.0", 20, 20);
    }
#endif
    lv_canvas_finish_layer(handler->canvas, &handler->layer);
}
after     lv_draw_vector(handler->vector_ctx); call  handler->drawText("GIS地图引擎测试 for LVGL ver2.0", 20, 20); the label display
 

```
#include "IGUIHandler.h"
#include "lvgl/lvgl.h"
[... 818 lines cut ...]
    }
    if (handler->canvas, &handler->layer) {
        lv_canvas_finish_layer(handler->canvas, &handler->layer);
    }
    if (handler->vector_ctx) {
        lv_vector_dsc_delete(handler->vector_ctx);
        handler->vector_ctx = NULL;
    }
    if (handler->draw_buf) {
       // lv_draw_buf_t* draw_buf = (lv_draw_buf_t*)handler->draw_buf;
       // lv_draw_buf_destroy(draw_buf);
        // 已经在回调中销毁了，这里直接给指针清掉即可
        handler->draw_buf = NULL;
    }
    if (handler->canvas) {
        lv_obj_del((lv_obj_t*)handler->canvas);
        handler->canvas = NULL;
    }
    if (handler->priv) {
        free(handler->priv);
        handler->priv = NULL;
    }
}

```

kisvegabor, 2024-08-30:
cc @FASTSHIFT

FASTSHIFT, 2024-08-31:
This is my test case and it works fine, based on the latest master branch code:
```c
void vector_label_demo(void)
{
    lv_draw_buf_t* draw_buf = lv_draw_buf_create(320, 240, LV_COLOR_FORMAT_ARGB8888, LV_STRIDE_AUTO);
    lv_draw_buf_clear(draw_buf, NULL);

    lv_obj_t* canvas = lv_canvas_create(lv_screen_active());
    lv_canvas_set_draw_buf(canvas, draw_buf);

    lv_layer_t layer;
    lv_canvas_init_layer(canvas, &layer);

    /* draw label1 */
    {
        lv_draw_label_dsc_t label_dsc;
        lv_draw_label_dsc_init(&label_dsc);
        label_dsc.color = lv_palette_main(LV_PALETTE_BLUE);
        label_dsc.font = LV_FONT_DEFAULT;
        label_dsc.text = "Hello label1";
        label_dsc.text_local = true;

        lv_area_t label_area;
        lv_area_set(&label_area, 50, 50, 100, 100);
        lv_draw_label(&layer, &label_dsc, &label_area);
    }

    lv_vector_dsc_t* ctx = lv_vector_dsc_create(&layer);
    lv_vector_path_t* path = lv_vector_path_create(LV_VECTOR_PATH_QUALITY_MEDIUM);

    lv_vector_path_clear(path);
    lv_vector_dsc_identity(ctx);

    lv_fpoint_t pts[] = { { 50, 50 }, { 200, 200 }, { 50, 200 } };
    lv_vector_path_move_to(path, &pts[0]);
    lv_vector_path_line_to(path, &pts[1]);
    lv_vector_path_line_to(path, &pts[2]);
    lv_vector_path_close(path);
    lv_vector_dsc_set_fill_color(ctx, lv_color_make(0xFF, 0x00, 0x00));
    lv_vector_dsc_scale(ctx, 0.5, 0.5);
    lv_vector_dsc_add_path(ctx, path);

    lv_draw_vector(ctx);
    lv_vector_path_delete(path);
    lv_vector_dsc_delete(ctx);

    /* draw label2 */
    {
        lv_draw_label_dsc_t label_dsc;
        lv_draw_label_dsc_init(&label_dsc);
        label_dsc.color = lv_palette_main(LV_PALETTE_GREEN);
        label_dsc.font = LV_FONT_DEFAULT;
        label_dsc.text = "Hello label2";
        label_dsc.text_local = true;

        lv_area_t label_area;
        lv_area_set(&label_area, 100, 100, 200, 200);
        lv_draw_label(&layer, &label_dsc, &label_area);
    }

    lv_canvas_finish_layer(canvas, &layer);
}
```

[image]

@xiangxud 
You can try this with the latest master, or switch the order of the draw labels (draw vectors before labels), because the previous thorvg renderer would empty the buffer before drawing, possibly causing labels already drawn to be erased.

It is also best to provide a minimal case (no more than 200 lines) of recurrence problems that rely only on the LVGL API, because my environment does not have your data structure and cannot run your code directly.

xiangxud, 2024-09-16:
thanks for your great job

closed 2024-09-16 by xiangxud