lvgl/lvgl #6718: LV_DRAW_BUF_INIT_STATIC broken for LV_COLOR_FORMAT_L8

capricorn-one, 2024-08-22:
### LVGL version

v9.1.1.dev

### What happened?

Error: argument of type "uint32_t" (aka "unsigned int") is incompatible with parameter of type "lv_color_format_t"C/C++(167)

Not sure if this works for any other color formats, don't see how it could tbh, but easy fix for now is casting header->cf as a lv_color_format_t in the define.

`#define LV_DRAW_BUF_INIT_STATIC(name) \
    do { \
        lv_image_header_t * header = &name.header; \
        lv_draw_buf_init(&name, header->w, header->h, (lv_color_format_t)header->cf, header->stride, buf_##name, sizeof(buf_##name)); \
        lv_draw_buf_set_flag(&name, LV_IMAGE_FLAGS_MODIFIABLE); \
    } while(0)
`

### How to reproduce?

```
LV_DRAW_BUF_DEFINE_STATIC(buffer, RPM_CANVAS_WIDTH, RPM_CANVAS_HEIGHT, LV_COLOR_FORMAT_L8);
LV_DRAW_BUF_INIT_STATIC(buffer);
```

kisvegabor, 2024-08-23:
I see no error or warning for it with GCC but opened a PR to fix it here: #6729

capricorn-one, 2024-08-25:
yeah... I'm not surprised to hear that. I'm using platform.io Arduino framework, probably has more to do with that. But given other people will use it, and doesn't compile as is, probably worth fixing. Thanks for putting in the PR!

PR #6729 (2024-08-29): fix(draw): cast color_format in LV_DRAW_BUF_INIT_STATIC
fixes #6718

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvg [... cut]
closed 2024-08-29 by XuNeo