lvgl/lvgl #6721: Touch device not working using indev_create since v9.1

IAmCillex, 2024-08-22:
### LVGL version

v9.1.0

### What happened?

creating touchpad input device using indev worked in v9.0 but not in v9.1.
Code below can be implemented in lvgl/linux_port main.c

snipped works out of the box when changing lvgl import back to v9.0

### How to reproduce?

@main.c

void evdev_read(lv_indev_t * indev, lv_indev_data_t * data){}

static void lv_linux_touch_init() {
    lv_indev_t * indev = lv_indev_create();
    lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER); 
    lv_indev_set_read_cb(indev, evdev_read);
    lv_evdev_create(LV_INDEV_TYPE_POINTER, "/dev/input/event2");
}

static void lv_linux_disp_init(void)
{
    const char *device = getenv_default("LV_LINUX_FBDEV_DEVICE", "/dev/fb0");
    lv_display_t * disp = lv_linux_fbdev_create();
    lv_linux_fbdev_set_file(disp, device);
    lv_linux_touch_init();
}

int main(void)
{
    lv_init();
    lv_linux_disp_init();

...

kisvegabor, 2024-08-23:
You should simply call `lv_evdev_create(LV_INDEV_TYPE_POINTER, "/dev/input/event2");` instead of 
```c
lv_indev_t * indev = lv_indev_create();
lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER);
lv_indev_set_read_cb(indev, evdev_read);
```

closed 2024-09-15 by lvgl-bot

druppy, 2025-01-16:
I have made a small test example in 9.2 using a small framebuffer and evdev (using a hynitron touch driver).

The touch device provides the right coordinars on touch (I have another application that uses cairo for drawing), but when using lvgl, and enabling logs, I get this

```
[Warn]	(142027.940, +33)	 indev_pointer_proc: X is 406 which is greater than hor. res (240) lv_indev.c:692
[Warn]	(142027.940, +0)	 indev_pointer_proc: Y is -19 which is smaller than zero lv_indev.c:695
``` 

When touching the display. Could some of the point processing be doing things wrong here ? 

I will try to be able to debug this, to see I can get the some head and tail in this.

kisvegabor, 2025-01-30:
@druppy Please open a new issue and descibie which indev driver do you use and what are the related settings in `lv_conf.h` (if any).

IAmCillex, 2025-02-19:
Hi, please excuse the late response.
I tried it again (now with v9.2) but I still encounter the same issue.

checking the device with evtest returns: 
/dev/input/event2:	Hycon Capacitive Touch
Select the device event number [0-2]: 2
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "Hycon Capacitive Touch"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    581
      Min        0
      Max       -1
    Event code 1 (ABS_Y)
      Value    171
      Min        0
      Max       -1
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max       10
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max       -1
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max       -1
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)

Event: time 1739965703.326999, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 284
Event: time 1739965703.326999, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 688
Event: time 1739965703.326999, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 318
Event: time 1739965703.326999, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1739965703.326999, type 3 (EV_ABS), code 0 (ABS_X), value 688
Event: time 1739965703.326999, type 3 (EV_ABS), code 1 (ABS_Y), value 318
Event: time 1739965703.326999, -------------- SYN_REPORT ------------
Event: time 1739965703.414357, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1739965703.414357, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1739965703.414357, -------------- SYN_REPORT ------------

I changed evdev loading to: 
static void lv_linux_disp_init(void)
{
    const char *device = getenv_default("LV_LINUX_FBDEV_DEVICE", "/dev/fb0");
    lv_display_t * disp = lv_linux_fbdev_create();

    lv_linux_fbdev_set_file(disp, device);

    lv_evdev_create(LV_INDEV_TYPE_POINTER, "/dev/input/event2");
}

and enabled LOGS.
On pressing the touch the logs return:
[Trace]	(77893.901, +0)	 indev_read_core: calling indev_read_cb lv_indev.c:190
[Info]	(77893.901, +0)	 indev_proc_press: pressed at x:0 y:0 lv_indev.c:1150
[Trace]	(77893.901, +0)	 event_send_core: Sending event 2 to 0x4f3b60 with 0x4f39f8 param lv_obj_event.c:348
[Trace]	(77893.901, +0)	 lv_indev_read: finished lv_indev.c:268
[Trace]	(77893.936, +35)	 lv_indev_read: begin lv_indev.c:208
[Trace]	(77893.936, +0)	 indev_read_core: calling indev_read_cb lv_indev.c:190
[Info]	(77893.936, +0)	 indev_proc_press: pressed at x:0 y:0 lv_indev.c:1150
[Trace]	(77893.936, +0)	 event_send_core: Sending event 2 to 0x4f3b60 with 0x4f39f8 param lv_obj_event.c:348
[Trace]	(77893.936, +0)	 lv_indev_read: finished lv_indev.c:268

when pressing a similar position on the touch screen.

Do you have an idea why the device coordinates are always set to 0?
Could it be an issue with x/y min/max not set correctly?

IAmCillex, 2025-02-19:
nevermind, 
the issue was indeed min/max. Issue solved by using set_calibration

lv_indev_t * touch = lv_evdev_create(LV_INDEV_TYPE_POINTER, "/dev/input/event2");
    if (touch < 0) {
        printf("ERROR: Failed to open /dev/input/event2\n");
    }
    
    lv_evdev_set_calibration(touch, 0, 0, 800, 480);