lvgl/lvgl #6724: Scroll bar does not show using multiple containers and scroll one

Jvsjvs, 2024-08-22:
### LVGL version

v9.1 and main line

### What happened?

Scroll bars are not shown in container when a child container is scrolled.

### How to reproduce?

1) Create an empty container (A) that is positioned on the left half of the screen (50% width, 100% height).
2) Create an empty container (B) that is positioned on the right half of the screen (50% width, 100% height).
3) Create a third container (C) as a child of container (A) with a width of 100% and a height in pixels such that it is taller then the screen height (it overflows container (A)).

4) Put some arbitrary widgets as children in in container C and some arbitrary widgets in container B.

Now you are able to vertically scroll the widgets in container C while keeping the widgets in container B stationary.
When when scrolling the widgets in container C (you can make it clickable to do so) **no scroll bar is shown**.

I would expect a scroll bar to show up in container A.

Note that I use this setup to have a numerical keyboard on the right half of the screen and on the left half I have different input boxes that are targeted with the keyboard.

kisvegabor, 2024-08-23:
Hi,

Could you send a code example to reproduce it?

Jvsjvs, 2024-08-23:
Hi Gabor,

Yes I could, see below the code I created to demo the problem. However, it works as expected, so it is apparently a problem somewhere in the generated code by Squareline Studio :-(

Sorry to disturb you.

```
    lv_obj_t * leftContainer = lv_obj_create(screenContainer);
    lv_obj_set_size(leftContainer, lv_pct(50), lv_pct(100));
    lv_obj_set_pos(leftContainer, 0, 0);
    lv_obj_remove_flag(leftContainer, LV_OBJ_FLAG_SCROLLABLE);

    // right container that will have non-scrollable content (e.g. the keyboard)
    lv_obj_t * rightContainer = lv_obj_create(screenContainer);
    lv_obj_set_size(rightContainer, lv_pct(50), lv_pct(100));
    lv_obj_set_pos(rightContainer, lv_pct(50), 0);

    // Create a scrollable container
    lv_obj_t * scrollableContainer = lv_obj_create(leftContainer);
    lv_obj_set_size(scrollableContainer, lv_pct(100), lv_pct(100));
    lv_obj_set_pos(scrollableContainer, 0, 0);
    lv_obj_add_flag(scrollableContainer, LV_OBJ_FLAG_CLICKABLE);

    // Put some widgets in the scrollable container
    lv_obj_t * someText1 = lv_label_create(scrollableContainer);
    lv_label_set_text(someText1, "Hello World!");
    lv_obj_set_pos(someText1, lv_pct(30), lv_pct(50));
    lv_obj_t * someText2 = lv_label_create(scrollableContainer);
    lv_label_set_text(someText2, "Do you like scrolling!");
    lv_obj_set_pos(someText2, lv_pct(25), lv_pct(150));

    lv_unlock();

```

kisvegabor, 2024-08-23:
In this case I'd look for `LV_OBJ_FLAG`s added/removed in the SquareLine code. I remember that the SquareLine default are slightly different than the LVGL defaults.

Jvsjvs, 2024-08-24:
Thanks, I will check.

I did add the flag scrollable, and I checked the generated code which indeed adds the flag.

Anyway, I will go through the generated code once more.

Regards,

John.

Jvsjvs, 2024-08-25:
Just created an "empty" project in Squareline Studio with 2 containers and some widgets, and it does not show a scrollbar while it is scrolling. I don't see a problemetic flag being set or cleared, any idea?

This is the simple 2 container setup:
[image]

This is the code it generates:
`
   ui_Screen1 = lv_obj_create(NULL);

    ui_Left = lv_obj_create(ui_Screen1);
    lv_obj_remove_style_all(ui_Left);
    lv_obj_set_width(ui_Left, lv_pct(50));
    lv_obj_set_height(ui_Left, lv_pct(100));
    lv_obj_set_style_border_color(ui_Left, lv_color_hex(0xFF0000), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_opa(ui_Left, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(ui_Left, 1, LV_PART_MAIN | LV_STATE_DEFAULT);

    ui_Label1 = lv_label_create(ui_Left);
    lv_obj_set_width(ui_Label1, LV_SIZE_CONTENT);   /// 1
    lv_obj_set_height(ui_Label1, LV_SIZE_CONTENT);    /// 1
    lv_obj_set_x(ui_Label1, 0);
    lv_obj_set_y(ui_Label1, 50);
    lv_obj_set_align(ui_Label1, LV_ALIGN_TOP_MID);
    lv_label_set_text(ui_Label1, "Scroll down");

    ui_TextArea1 = lv_textarea_create(ui_Left);
    lv_obj_set_width(ui_TextArea1, 300);
    lv_obj_set_height(ui_TextArea1, 400);
    lv_obj_set_x(ui_TextArea1, 0);
    lv_obj_set_y(ui_TextArea1, 100);
    lv_obj_set_align(ui_TextArea1, LV_ALIGN_TOP_MID);
    lv_textarea_set_placeholder_text(ui_TextArea1, "This textarea is too big to fit on this screen.");

    ui_Label2 = lv_label_create(ui_Left);
    lv_obj_set_width(ui_Label2, LV_SIZE_CONTENT);   /// 1
    lv_obj_set_height(ui_Label2, LV_SIZE_CONTENT);    /// 1
    lv_obj_set_x(ui_Label2, 0);
    lv_obj_set_y(ui_Label2, 550);
    lv_obj_set_align(ui_Label2, LV_ALIGN_TOP_MID);
    lv_label_set_text(ui_Label2, "Text scrolling in");

    ui_Right = lv_obj_create(ui_Screen1);
    lv_obj_remove_style_all(ui_Right);
    lv_obj_set_width(ui_Right, lv_pct(50));
    lv_obj_set_height(ui_Right, lv_pct(100));
    lv_obj_set_align(ui_Right, LV_ALIGN_TOP_RIGHT);
    lv_obj_remove_flag(ui_Right, LV_OBJ_FLAG_CLICKABLE | LV_OBJ_FLAG_SCROLLABLE);      /// Flags
    lv_obj_set_style_border_color(ui_Right, lv_color_hex(0xFF0000), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_opa(ui_Right, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(ui_Right, 1, LV_PART_MAIN | LV_STATE_DEFAULT);

    ui_Label3 = lv_label_create(ui_Right);
    lv_obj_set_width(ui_Label3, LV_SIZE_CONTENT);   /// 1
    lv_obj_set_height(ui_Label3, LV_SIZE_CONTENT);    /// 1
    lv_obj_set_align(ui_Label3, LV_ALIGN_CENTER);
    lv_label_set_text(ui_Label3, "text");
`

Jvsjvs, 2024-08-25:
If I use a "Panel" instead of a "Container" then it does show scrollbars.

kisvegabor, 2024-08-25:
`lv_obj_remove_style_all(ui_Left)` removes the styles of the  scrollbars too, which makes the scrollbars transparent and zero width.

closed 2024-09-18 by lvgl-bot