lvgl/lvgl #6728: Format specifier macro mistake in lv_tiny_ttf.c

SecretaryStarlight, 2024-08-23:
### LVGL version

v9.1.0

### What happened?

The log output in function `lv_tiny_ttf_set_size()` use format specifier macro `PRIx32` defined in **inttypes.h** instead of `LV_PRIx32` defined in **lv_sprintf.h**. This may cause compile error in some case.
This may also happened in function `lv_vg_lite_buffer_format_bytes()`, but I'm not test it yet.
```c
void lv_tiny_ttf_set_size(lv_font_t * font, int32_t font_size)
{
    if(font_size <= 0) {
        LV_LOG_ERROR("invalid font size: %"PRIx32, font_size);
        return;
    }
    
    //...
}
```
The following is an excerpt from the Keil AC5 compilation log:
```
compiling lv_tiny_ttf.c...
..\..\Middlewares\lvgl\src\libs\tiny_ttf\lv_tiny_ttf.c(105): error:  #18: expected a ")"
          LV_LOG_ERROR("invalid font size: %"PRIx32, font_size);
..\..\Middlewares\lvgl\src\libs\tiny_ttf\lv_tiny_ttf.c: 0 warnings, 1 error
```

kisvegabor, 2024-08-24:
Thanks for reporting! Could you send a PR to fix it? If so we can still add the fix to v9.2 (will be released on Monday)

cc @FASTSHIFT

MacDragon, 2024-08-24:
As I already had setup for this, I quickly threw a PR for it in tiny ttf at least,

closed 2024-09-09 by SecretaryStarlight