lvgl/lvgl #6734: Misfomatted docs

kisvegabor, 2024-08-26:
### LVGL version

v9.1.1

### What happened?

There is a strange formatting issue in the label widget's docs [here](https://docs.lvgl.io/master/widgets/label.html#parts-and-styles). The raw `.rst` doesn't tall anything about making the first line bold. 

### How to reproduce?

@kdschlosser @vwheeler63 do you have any idea?

vwheeler63, 2024-08-26:
Good morning, Gabor! ( @kisvegabor )

You're up early!  Forgive me if I am saying something you already know, but it appears to be a bug in whatever is processing the ":cpp:enumerator:" ReST syntax in the `label.rst` file, trying to make the first line into a heading....  Do we know a time in the past when it did not do that?

vwheeler63, 2024-08-26:
@kisvegabor I am submitting a quick PR to remedy.

vwheeler63, 2024-08-26:
Done:  PR #6735

closed 2024-08-26 by kisvegabor