lvgl/lvgl #6740: lv_draw_task_get_draw_dsc' was not declared

nielseng2017, 2024-08-26:
### LVGL version

9.1

### What happened?

The environment is Arduino IDE 2.3.2 using a Cheap Yellow Display (CYD) ESP32.
Successfully implement a screen with multiple lv_buttonmatrix, then tried to utilize the code from the "Custom Buttons" example in the lvgl documentation. I replicated the code for the second button to change the color of four buttons in my button matrix.
The compilation errors our with:
C:\Users\George\Documents\Arduino\Cheap_Yellow_Display_LVGL_FromScratch\Cheap_Yellow_Display_LVGL_FromScratch.ino: In function 'void event_handler_statematrix2(lv_event_t*)':
C:\Users\George\Documents\Arduino\Cheap_Yellow_Display_LVGL_FromScratch\Cheap_Yellow_Display_LVGL_FromScratch.ino:147:37: error: 'lv_draw_task_get_draw_dsc' was not declared in this scope; did you mean 'lv_draw_task_get_arc_dsc'?
  147 |     lv_draw_dsc_base_t * base_dsc = lv_draw_task_get_draw_dsc(draw_task); //draw_task
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                                     lv_draw_task_get_arc_dsc

exit status 1

Compilation error: 'lv_draw_task_get_draw_dsc' was not declared in this scope; did you mean 'lv_draw_task_get_arc_dsc'?

### How to reproduce?

static void event_handler_statematrix2(lv_event_t * e)
{
    lv_obj_t * obj = (lv_obj_t*)lv_event_get_target(e);  // would not compile without (lv_obj_t*) <<<<<<<<<<
    lv_draw_task_t * draw_task = lv_event_get_draw_task(e);
    lv_draw_dsc_base_t * base_dsc = lv_draw_task_get_draw_dsc(draw_task); //draw_task
    /*When the button matrix draws the buttons...*/
    if(base_dsc->part == LV_PART_ITEMS) {
        bool pressed = false;
        if(lv_buttonmatrix_get_selected_button(obj) == base_dsc->id1 && lv_obj_has_state(obj, LV_STATE_PRESSED)) {
            pressed = true;
        }

        /*Change the draw descriptor of the 1st button*/
        if(base_dsc->id1 == 0) {
            lv_draw_fill_dsc_t * fill_draw_dsc = lv_draw_task_get_fill_dsc(draw_task);
            if(fill_draw_dsc) {
                fill_draw_dsc->radius = 0;
                if(pressed) fill_draw_dsc->color = lv_palette_darken(LV_PALETTE_GREEN, 3);
                else fill_draw_dsc->color = lv_palette_main(LV_PALETTE_GREEN);
            }
            lv_draw_box_shadow_dsc_t * box_shadow_draw_dsc = lv_draw_task_get_box_shadow_dsc(draw_task);
            if(box_shadow_draw_dsc) {
                box_shadow_draw_dsc->width = 6;
                box_shadow_draw_dsc->ofs_x = 3;
                box_shadow_draw_dsc->ofs_y = 3;
            }
            lv_draw_label_dsc_t * label_draw_dsc = lv_draw_task_get_label_dsc(draw_task);
            if(label_draw_dsc) {
                label_draw_dsc->color = lv_color_white();
            }

        }
		/*Change the draw descriptor of the 2nd button*/
        else if(base_dsc->id1 == 1) {
             lv_draw_fill_dsc_t * fill_draw_dsc = lv_draw_task_get_fill_dsc(draw_task);
            if(fill_draw_dsc) {
                fill_draw_dsc->radius = 0;
                if(pressed) fill_draw_dsc->color = lv_palette_darken(LV_PALETTE_YELLOW, 3);
                else fill_draw_dsc->color = lv_palette_main(LV_PALETTE_YELLOW);
            }
            lv_draw_box_shadow_dsc_t * box_shadow_draw_dsc = lv_draw_task_get_box_shadow_dsc(draw_task);
            if(box_shadow_draw_dsc) {
                box_shadow_draw_dsc->width = 6;
                box_shadow_draw_dsc->ofs_x = 3;
                box_shadow_draw_dsc->ofs_y = 3;
            }
            lv_draw_label_dsc_t * label_draw_dsc = lv_draw_task_get_label_dsc(draw_task);
            if(label_draw_dsc) {
                label_draw_dsc->color = lv_color_white();
            }
        }
		/*Change the draw descriptor of the 3rd button*/
        else if(base_dsc->id1 == 2) {
             lv_draw_fill_dsc_t * fill_draw_dsc = lv_draw_task_get_fill_dsc(draw_task);
            if(fill_draw_dsc) {
                fill_draw_dsc->radius = 0;
                if(pressed) fill_draw_dsc->color = lv_palette_darken(LV_PALETTE_RED, 3);
                else fill_draw_dsc->color = lv_palette_main(LV_PALETTE_RED);
            }
            lv_draw_box_shadow_dsc_t * box_shadow_draw_dsc = lv_draw_task_get_box_shadow_dsc(draw_task);
            if(box_shadow_draw_dsc) {
                box_shadow_draw_dsc->width = 6;
                box_shadow_draw_dsc->ofs_x = 3;
                box_shadow_draw_dsc->ofs_y = 3;
            }
            lv_draw_label_dsc_t * label_draw_dsc = lv_draw_task_get_label_dsc(draw_task);
            if(label_draw_dsc) {
                label_draw_dsc->color = lv_color_white();
            }
        }
		/*Change the draw descriptor of the 4th button*/
        else if(base_dsc->id1 == 3) {
             lv_draw_fill_dsc_t * fill_draw_dsc = lv_draw_task_get_fill_dsc(draw_task);
            if(fill_draw_dsc) {
                fill_draw_dsc->radius = 0;
                if(pressed) fill_draw_dsc->color = lv_palette_darken(LV_PALETTE_BLUE, 3);
                else fill_draw_dsc->color = lv_palette_main(LV_PALETTE_BLUE);
            }
            lv_draw_box_shadow_dsc_t * box_shadow_draw_dsc = lv_draw_task_get_box_shadow_dsc(draw_task);
            if(box_shadow_draw_dsc) {
                box_shadow_draw_dsc->width = 6;
                box_shadow_draw_dsc->ofs_x = 3;
                box_shadow_draw_dsc->ofs_y = 3;
            }
            lv_draw_label_dsc_t * label_draw_dsc = lv_draw_task_get_label_dsc(draw_task);
            if(label_draw_dsc) {
                label_draw_dsc->color = lv_color_white();
            }
        }
    }
}

lhdjply, 2024-08-27:
#6742  The 9.1 documentation is incorrect; you can update to version 9.2 or wait for the 9.1 documentation to be fixed.

kisvegabor, 2024-08-29:
Should be fixed when https://github.com/lvgl/lvgl/pull/6760 is merged

kisvegabor, 2024-08-29:
The pr is merged and the v9.1 docs is correct now.

closed 2024-08-29 by kisvegabor