lvgl/lvgl #6751: Multi-layer doesnt work for v9.2

fatiihyildizz, 2024-08-28:
### LVGL version

v9.2

### What happened?

Hello, 
The multi-layer mechanism I worked on for v9.1 does not work in v9.2. It goes into flush callback for the first buffer but does not switch to the second buffer.

kisvegabor, 2024-08-29:
Could you be more specific about he issue? What is your exact setup for the buffers/layers?

fatiihyildizz, 2024-08-29:
Hello,
I am using NXP i.MX RT1176. There are 3 layers and 3 buffers for the display.
[image]
To flush the displays, I perform the operation by controlling the layers in a common flush callback.
[image]
For the v9.1, when I filled the displays separately, they were coming in the callback below in order. However, in v9.2, only the display I named "ps_disp_bg" comes from these displays. The same code works as it should in v9.1.

kisvegabor, 2024-08-29:
A few notes:
- In DIRECT_MODE if multiple areas are updated on the screen the flush_cb will be called multiple times for the same screen. Therefore you need to switch frame buffers only on the last area of the the given display. You can use `lv_display_flush_is_last(display)` for this.
- Where do you call `lv_display_flush_ready`?
- With a debugger do you see if it's spinning in a loop somewhere when it stuck?

kisvegabor, 2024-09-05:
Is it also resolved similarly to https://github.com/lvgl/lvgl/issues/6750 ?

fatiihyildizz, 2024-09-09:
Hi @kisvegabor ,
The reason we used Multilayer was that flicker occurred on the screen during visual animations. We thought we could increase the speed by shrinking the areas with Multilayer. However, we saw that the flicker problem disappeared when the `lv_display_flush_is_last` function you mentioned above was checked in the callback. So we no longer need Multilayer. **_I recommend that those who experience flicker pay attention to this._**

kisvegabor, 2024-09-10:
Thank you for the feedback! 

I close this issue.

closed 2024-09-10 by kisvegabor