lvgl/lvgl #6754: btnmatrix: const-qualify the 'map' input

julianfoad, 2024-08-28:
### LVGL version

observed around v8.1, looks similar in v9.2.0

### What happened?

In essence:

```diff
-void lv_btnmatrix_set_map(lv_obj_t * btnm, const char * map[])
+void lv_btnmatrix_set_map(lv_obj_t * btnm, const char *const map[])
```

**Actual:** An [Infinitime Calculator app](https://github.com/InfiniTimeOrg/InfiniTime/pull/375) was unable to declare its button-matrix map inputs fully 'const' and these maps then occupied 160 bytes of non-stack "static" RAM.

**Expected:** These map inputs should be able to be fully 'const', so they can be placed in ROM.

**A patch:** The following patch applies to the old version of LVGL (around v8.1) that Infinitime is using: https://lab.trax.im/gentle/lvgl/-/commit/6f310848a55a4e7203d64ff8086a10d4242c63cd . I verified that the surrounding LVGL code (old version around v8.1) does treat these inputs as 'const' in effect, it was just missing the 'const' qualifier in a few places.

It looks like the current LVGL code lacks 'const' in the same way, but has some other changes.

**A work-around:** by using `const_cast`s in the caller (Infinitime Calculator app in my case) a fully constant map can be passed in. Implemented in [this patch to Calculator](https://lab.trax.im/gentle/infinitime/-/commit/eebd027eecb2a4c756d02efecd11f96fc914d95a) .

kisvegabor, 2024-08-29:
I don't understand the problem. Passing non-const as const should work and const is the stronger constraint. Treating non-const as const shouldn't case any issues. Or is it a cpp specific issue?

julianfoad, 2024-08-29:
@liamHowatt OK, here it is, now rebased on your master/HEAD, in my `buttonmatrix-const` branch: https://lab.trax.im/gentle/lvgl/-/tree/buttonmatrix-const

Unfortunately Infinitime isn't updated to use your latest and I don't have a handy way to test a build with this change, so I just made this patch by hand. I did it carefully (I am experienced in C) but as always it's possible I could have missed something.

If you wish to try it, you can pull/merge my `buttonmatrix-const` branch into your `master` branch like this, the standard Git way (as I choose not to do my work in Microsoft Github):

```sh
git pull https://lab.trax.im/gentle/lvgl.git buttonmatrix-const
```

julianfoad, 2024-08-29:
@kisvegabor Did you maybe read it the wrong way round? As it is, the function signature says "I want to be able to write through this pointer."

julianfoad, 2024-08-29:
Re. "I don't have a handy way to test"... it's just occurred to me that as a mature piece of software you've probably got a test suite. Sorry, I hadn't even looked, I wasn't planning to come here and work on this, and I haven't really got time for it, but now I've got this far maybe if it's quick and easy I'll go the extra mile and see if I can find and run a test suite. OK, I see it now. Let me give it a go.

julianfoad, 2024-08-29:
And as soon as I look at `tests/src/test_cases/widgets/test_btnmatrix.c` ... I put 'const' on the return value of the getter as well, which may break code like in the test suite:

```c
void test_button_matrix_creation(void)
{
    const char ** map;
    map = lv_buttonmatrix_get_map(btnm);
```

At this point I'm going to have to ask you to make the decisions about what to do there (such as leave it as returning non-const or making two different getters), and let me know if you're willing to take over at this point, and let me know if there are any quick ways I can help you further with it, as I'm afraid I can't get stuck into this properly myself. I just hope this much is already of some help.

julianfoad, 2024-08-29:
@liamHowatt Thanks for taking this to completion, and catching that place where I pasted "const" in the wrong place.

Nice to see how, in the test file, if I'm reading it correctly, the caller uses constant maps like in my use case, and this patch gets rid of the casts that were previously needed there.

kisvegabor, 2024-08-30:
Ahh, I see now, thank you! I'm looking into the PR! :slightly_smiling_face:

PR #6765 (2024-09-02): fix(buttonmatrix): use const arrays
Fixes #6754

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3.4.12` needs to i [... cut]