lvgl/lvgl #6755: Is there a plan to make labels support color font？

ckoh01, 2024-08-29:
### Introduce the problem

Currently, due to the large size of bin format files convented with raw images, loading them from external flash takes a long time, which resulting in slow rendering speed.
If lv_font_comv can be used to convert color fonts, the resulting font bin file size will be much smaller, which result in much faster loading speed.

### Proposal

https://github.com/googlefonts/noto-emoji
[image]

I want to convert the image to SVG first, then convent it to color fonts on line(https://icomoon.io/app/#/select), and finally convent *.ttf to font bin files.  Finally, use Lvgl labels that support color font to display the image.

However, the current labels may not support color fonts. Is it possible for labels to have this feature in the future?

kisvegabor, 2024-08-29:
I think you are looking for [image font](https://docs.lvgl.io/master/others/imgfont.html). You can store the images as indexed 8 bit too.

ckoh01, 2024-08-29:
I thought [image font](https://docs.lvgl.io/master/others/imgfont.html) still use image array. which size is still a little big for loading from extern flash.

[image]

My goal is for the label to support color font. Is this feasible？

kisvegabor, 2024-08-30:
Sorry, I don't understand the problem. So you can display the images from the Noto font but the color is incorrect?

ckoh01, 2024-08-30:
sorry, i mean
[[LV_SYMBOL_](https://docs.lvgl.io/master/overview/font.html#fonts)](https://docs.lvgl.io/master/overview/font.html#special-fonts)
FontAwesome font can be showed in label widgets, but it alway show mono color(like red or else).

Noto font is a color font, can the label display Noto font's characters in color?

liamHowatt, 2024-09-04:
This new feature might do exactly what you want :slightly_smiling_face: #6686

kisvegabor, 2024-09-04:
Note that even FreeType support only bitmaps and not vectors. See [here](https://github.com/lvgl/lvgl/pull/6686/files#diff-5c88f9b0a37d8d678ec24b9eae8b59a25e1423543c7d9f12ff2c15d2c0bcd1bbR92-R94). 

For embedded I think the combination of SVG `lv_imagefont` would be nice. (SVG is on the roadmap of LVGL v9.3)

closed 2024-09-28 by lvgl-bot