lvgl/lvgl #6756: ime_pinyin array overwrite crash

squi2rel, 2024-08-29:
### LVGL version

v8.4.0

### What happened?

just repeat type one letter, it will crash
[image]

### How to reproduce?

In this example, I pressed 'a'
[image]

kisvegabor, 2024-08-29:
Does it crash in v9.2 too?

kisvegabor, 2024-08-29:
It seems working [here](https://docs.lvgl.io/8.3/others/ime_pinyin.html?highlight=pinyin) and [here](https://docs.lvgl.io/master/others/ime_pinyin.html#pinyin-ime) too.

squi2rel, 2024-08-29:
The [example](https://docs.lvgl.io/master/others/ime_pinyin.html#example) also crash too
[image]

kisvegabor, 2024-08-29:
Please send a video. I've added like 50 `f`s and it's still working :smile:

squi2rel, 2024-08-29:
https://github.com/user-attachments/assets/35013547-8b93-42c3-b6cd-e2dc4c442db3

maybe 70 letters?

kisvegabor, 2024-08-30:
Okay, I crashed in v8 but worked in v9. It seems this PR fixed the issue: https://github.com/lvgl/lvgl/pull/6501/files

lhdjply, 2024-08-30:
> Okay, it crashed in v8 but worked in v9. It seems this PR fixed the issue: https://github.com/lvgl/lvgl/pull/6501/files

It is crash in V9 too. see pr #6767

PR #6767 (2024-09-02): fix(ime): fix ime crash when input_char is too long
A clear and concise description of what the bug or new feature is.

fixes #6756

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lv [... cut]
closed 2024-09-02 by FASTSHIFT