lvgl/lvgl #6759: keypad LONG_PRESSED_REPEAT event can't work

liriyao, 2024-08-29:
### LVGL version

v9.0.0

### What happened?

in indev_keypad_proc function,uint32_t prev_key = i->keypad.last_key; prev_key can't  always be LV_INDEV_STATE_PRESSED altough i keep pressing "Enter" 。So i won't enter the pressing proc " else if(is_enabled && data->state == LV_INDEV_STATE_PRESSED && prev_state == LV_INDEV_STATE_PRESSED) " can't be ture .

### How to reproduce?

in windows simulator .add an pressing event to any obj then focus to  that obj and keep pressing "Enter". it won't trigger the event

kisvegabor, 2024-08-30:
I can look into it only a little bit later. If you have time please send a pull request with a proposed fix.

liriyao, 2024-08-30:
This problem may be caused by a bad keyboard return rate。windows message thread is too fast,lv_windows_keypad_device_window_message_handler too fast .windows keypad message list generated by lv_windows_push_key_to_keyboard_queue ,0 will come up sometimes. although i keep pressing any key on keyboard. I am a rookie, just start wokring really don't know how to fix this issue .sorry can't help you 

> I can look into it only a little bit later. If you have time please send a pull request with a proposed fix.

kisvegabor, 2024-09-04:
Note that, this issue might be specific to the Windows driver. cc @MouriNaruto do you have any ideas why the LONG_PRESSED_REPEAT is not working with keys on Windows?

In case of SDL for example SDL handles key presses/releases. That is if I keep pressing key `f` SDL will repeat it so LVGL won't see the long press.

liriyao, 2024-09-05:
By the way,i used deley to lower the speed of windows message thread,the situation will be better a lot.

MouriNaruto, 2024-09-05:
In LVGL Windows backend, when we get the WM_KEYUP Windows window message, we will push the LV_INDEV_STATE_PRESSED status to the keyboard queue in the window thread. Then, LVGL main thread process the keyboard queue in the keypad callback.

Maybe we need a way to notify the LVGL when message is here.

Kenji Mouri

MouriNaruto, 2024-09-05:
Another solution is introduce a signal, makes window thread waiting for the LVGL main thread execute the keypad callback.

But it should affect the keyboard input performance. (For example, lost the keys user input when waiting.)

Kenji Mouri

kisvegabor, 2024-09-05:
If you record the PRESSE state let's say at t=100 and the RELEASED state at t=600 LVGL should process it correctly. 

Could you print what is the actual input that LVGL gets (with a timestamp coming from `lv_tick_get()`)?

closed 2024-09-28 by lvgl-bot