lvgl/lvgl #6763: v9.3 planning

kisvegabor, 2024-08-29:
### Introduce the problem

**Goals**
Plan the core feature and directors of v9.3. I have 2 goals for 2024 Q3 and Q4:
1. Make LVGL more appealing for MCUs: ready-to-use projects, Linux drivers, more docs, heavier features 
2. Improve the developer experience: docs and CI review, more examples, etc

**Schedule**
Targeting 2025 Q1

### Proposal

In light of that, I would like to focus on these features and updates:
**Features**
- [ ] Blurring https://github.com/lvgl/lvgl/issues/6018
- [ ] Tiled rendering https://github.com/lvgl/lvgl/pull/6761
- [ ] OpenGL rendering https://github.com/lvgl/lvgl/issues/6017
- [ ] Multi-touch (scroll, zoom, rotate) and mouse wheel scroll https://github.com/lvgl/lvgl/issues/6014
- [ ] Widget caching
- [ ] SVG https://github.com/lvgl/lvgl/issues/6022
- [ ] Improve the span widget https://github.com/lvgl/lvgl/issues/6019
- [ ] Double and triple click: https://github.com/lvgl/lvgl/pull/6187
- [ ] Font manager https://github.com/lvgl/lvgl/pull/6482

**Docs**
- [ ] Review and complete the docs https://github.com/lvgl/lvgl/issues/6671
- [ ] Yocto and Build root docs
- [ ] Add the docs page for all chip vendor partners
- [ ] Write detailed tutorials about the important features
- [ ] Add more examples where needed
 
**Developer experience**
- [ ] Have 5 repos for MPU-based dev boards
- [ ] Have repos for all the officially supported boards 
- [ ] Control the number of open issues and PRs
- [ ] API unification (for the editor and binding generation)
- [ ] Clean up the CI https://github.com/lvgl/lvgl/issues/5396
- [ ] Separate lv_conf.h to platform and UI-specific parts https://github.com/lvgl/lvgl/issues/6010
- [ ] Have a demo for MPUs (Full HD)

---

Leave a comment if you would like to see other features too in v9.3. 
cc @liamHowatt @XuNeo @FASTSHIFT @etag4048

XuNeo, 2024-09-06:
Should we also consider optimizing flash usage (maybe RAM usage too)?

kisvegabor, 2024-09-10:
What do you mean by that in more detail? Disabling modules? Or a general review?

XuNeo, 2024-09-11:
> What do you mean by that in more detail? Disabling modules? Or a general review?

Last time I tried to disable widgets to reduce flash usage on my `luavgl` project, it fails to compile. Widgets should be disabled automatically when the one it depends on is disabled. 

Disable software renderers not used also saves flash according to https://github.com/lvgl/lvgl/pull/6313. 

We may need to investigate what other modules can be disabled. A white paper to conclude the methods would be great.

kisvegabor, 2024-09-11:
> Last time I tried to disable widgets to reduce flash usage on my luavgl project, it fails to compile.

Bindings are really a special case as the linker can't eliminate the unused functions. 

> Widgets should be disabled automatically when the one it depends on is disabled.

I wouldn't disable them automatically because it might result in hard to debug config issues. Instead I'd add `#error`s about some other module is required for this one.

xlbao123, 2024-09-13:
I hope LVGL can become more modular by providing a plugin mechanism, placing all non-core plugins in a unified directory. The advantages of this approach are: 
- it would make module updates and development easier
- allow for easier removal of unnecessary modules
- reduce the need for macros
- simplify the code in lv_init

Some modules that come to mind are:
- File I/O Manager: Similar to the current lv_fs_drv.
- Resource Manager: Supporting such as resource_file, resource_image, resource_font, etc., and managing all resources uniformly with support for cache management algorithms.
- Font Manager.
- Screen Transition Plugin.
- Various image formats/decoders, SVG, Lottie, etc., as independent plugins.
- Gesture Plugin for touch.

kisvegabor, 2024-09-13:
@xlbao123 Now the `others` (optional extra modules) and `libs` (3rd party lib support) has their own folder. How would a better folder structure look like in your opinion?

xlbao123, 2024-09-13:
> @xlbao123 Now the `others` (optional extra modules) and `libs` (3rd party lib support) has their own folder. How would a better folder structure look like in your opinion?

My idea is to compile the entire core into` liblvgl.a`, and all other modules as independent libraries in the form of plugins, such as 
- gesture plugin:` liblvgl-plugin-gesture.a`
- fio manager plugin: ` liblvgl-plugin-io-posix.a`, ` liblvgl-plugin-image-memfs.a`, ` liblvgl-plugin-image-vfs.a` (FS like fatfs/littlefs are integrated via vfs)
- image plugin:` liblvgl-plugin-image-lodepng.a`, `liblvgl-plugin-image-svg.a`, ...
- font manager plugin: `liblvgl-plugin-font-freetype.a`, `liblvgl-plugin-font-binfont.a`, ...
- render plugin: `liblvgl-plugin-render-swrender.a`, `liblvgl-plugin-render-vglite.a`, `liblvgl-plugin-render-dma2d.a`. (Since some renderers might partially use software rendering for certain apis, it might be necessary to separate out `liblvgl-swrender.a` as well.)

**Each plugin would have a plugin_name, version, unified init/deinit interfaces, and plugin-specific options.**

This approach would eliminate a large number of macros,  each platform would simply link to different libraries, and making it easy to develop new plugins as well.

xlbao123, 2024-09-13:
I have a question, and I'm not sure if I understand it correctly. I see that lvgl uses a refr timer to periodically check for dirty rectangles. For optimal performance, I set this timer to 60Hz.

- On certain pages, such as one where the content only refreshes once per second, the timer will frequently wake the MCU from sleep to check for dirty rectangles, and then it will go back to sleep if there's no update. This can waste a lot of power. How can I make sure that the redraw is triggered only when there is a change in the page data?

- Once a redraw is complete, the MCU waits for the next refresh timer period to start rendering the next frame. How can I save this time (so that as soon as one frame is rendered, and the next frame's buffer is ready and there is a redraw event, the next frame starts rendering immediately)?

kisvegabor, 2024-09-19:
> Each plugin would have a plugin_name, version, unified init/deinit interfaces, and plugin-specific options.
This approach would eliminate a large number of macros, each platform would simply link to different libraries, and making it easy to develop new plugins as well.

How would the code look like? Would you put all the plugins into the same `lvgl` monorepo?

> On certain pages, such as one where the content only refreshes once per second, the timer will frequently wake the MCU from sleep to check for dirty rectangles, and then it will go back to sleep if [...]

It's already fixed (https://github.com/lvgl/lvgl/commit/d824618366e474eef57081d647031965e6edc136), please try with the latest master.

> Once a redraw is complete, the MCU waits for the next refresh timer period to start rendering the next frame. How can I save this time (so that as soon as one frame is rendered, and the next frame's buffer is ready and there is a redraw event, the next frame starts rendering immediately)?

Can you just simple select 120Hz instead of 60? :slightly_smiling_face: If there is nothing to do the refresh timer will be paused (with the fix mentioned above)

xlbao123, 2024-09-20:
> How would the code look like? Would you put all the plugins into the same `lvgl` monorepo?

I will try to describe it using some code (in the code below, I’ve added support for the dynamic library, but I think this may not be necessary).

`lv_plugin.h`
```
#ifndef LV_PLUGIN_H
#define LV_PLUGIN_H

struct lv_plugin_state;

typedef struct
{
    char *filename;
    void *dll_handle;
    int valid;
    struct lv_plugin_state info;
} lv_plugin_handle_t;

#define FMT_PLUGIN_VERSION(major, minor, build) (major << 24 | minor << 16 | build)
#define GET_MAJOR(v) (((v) >> 24) & 0xff)
#define GET_MINOR(v) (((v) >> 16) & 0xff)
#define GET_BUILD(v) ((v) & 0xffff)

typedef enum
{
    SERVICE_NUMBER_EARLY = -100,
    SERVICE_NUMBER_NORMAL = 0,
    SERVICE_NUMBER_LATE = 100,
} lv_service_number_t;

typedef uint32_t lv_version_t;

typedef struct lv_plugin_state
{
    const char *name;
    void *plugin_data;
    int (*init)(char *options, void *pdata);
    void (*destory)(void *pdata);
    lv_version_t version;
    int service_number;
} lv_plugin_state_t;

typedef int (*lv_plugin_create_func_t)(lv_plugin_state_t *state);

#if defined(LV_ENABLE_DYNAMIC_PLUGINS)

#define LV_PLUGIN_CREATE_FUNC_STR "lv_plugin_create"

#if defined(_WIN32)
#define LV_PLUGIN_CREATE_FUNC DLLExport lv_plugin_create
#else
#define LV_PLUGIN_CREATE_FUNC lv_plugin_create
#endif /* _WIN32 */

#else

// For static plugin deployments, define a single entry point for the plugin list to be provided.
const lv_plugin_create_func_t *lv_get_plugins();

#endif /* LV_ENABLE_DYNAMIC_PLUGINS */

#endif /* LV_PLUGIN_H */
```

`lv_plugin.c`
```
#include <lv_plugin.h>

typedef struct
[... 146 lines cut ...]
    char *options[OPTIONS_NUM];
    int option_count = 0;

    options[option_count++] = "lodepng";   ///> plugin name
    options[option_count++] = "opt0=val0"; ///> plugin option

    options[option_count++] = "lodepng";
    options[option_count++] = "opt1=val1";

    assert(option_count <= OPTIONS_NUM);
    lv_init(options, option_count);

    // ...
}
```

1. `lv_plugins` may makes the code look ugly. However, there are ways to optimize this, we can link plugin entry points into a specific section, just like modules in linux kernel.

2. As the features of LVGL continue to grow, one of the advantages of doing this is that we can gradually eliminate more macros: 
    - Feature toggles can be replaced by whether the corresponding plugin is linked or not.
    - For the configuration of specific plugin, if it's a compile-time option, it can be controlled internally by macros within the plugin; if it's a runtime option, it can be replaced by `plugin_option`.

3. Separating the modules is just the first step, and a further step would be interface isolation: avoiding direct calls from the core to plugin interfaces, and also preventing plugins from calling each other's interfaces. Another challenge is to gradually separate some modules from liblvgl, making the core more streamlined.

4. As for the repo management of plugins, I think some core plugins can still be placed in a unified directory within the lvgl repo, while others, especially those not closely tied to the LVGL version, can be maintained in independent repos.

kisvegabor, 2024-10-03:
Sorry for the late reply. I see the potential value in this approach, but let's wait a little before jumping into it. I'd like to see if there is a real need for such a change.

stachurr, 2024-10-15:
It would be nice to allow access to fragments on the stack by index. Currently, fragments on the stack can only be found if each fragment has a unique container via `lv_fragment_manager_find_by_container`, which is not guaranteed/enforced.

Here's code that would accomplish this:
```
lv_fragment_t * lv_fragment_manager_get_by_index(lv_fragment_manager_t * manager, size_t index)
{
    LV_ASSERT(manager);
    size_t idx = 0;
    lv_fragment_managed_states_t * states;
    LV_LL_READ(&manager->attached, states) {
        if (idx++ == index) return states->instance;
    }
    return NULL;
}
```

kisvegabor, 2024-10-16:
To be honest I haven't used the fragments in real projects yet, but your request seems reasonable. Please send a PR. :slightly_smiling_face:

stachurr, 2024-10-16:
Maybe it is my misunderstanding of the fragments concept, but wouldn't it be desirable to have only one container for all associated fragments? This is what makes them useful, no? I will make a PR in the near future. Thanks!

nicusorcitu, 2025-01-15:
@kisvegabor do you have a much clear date for release 9.3? I see is Q1 but a more precise date will help. Thanks.

kisvegabor, 2025-01-16:
Probably the end of Q1 (end of March) is the most realistic. Is there a deadline on your end that we should align with?

nicusorcitu, 2025-01-16:
I thought it would be released sooner in January, we have a release freeze in February, so definitely 9.3 will not be on time.

kisvegabor, 2025-01-29:
There are a few feature that we should have in v9.3 and they are not ready yet:
- OpenGL rendering: Basics are  working, but there are some issues on embedded devices. @liamHowatt and @EDGEMTech-GabrielC are tracking it
- Multi-touch (scroll, zoom, rotate) and mouse wheel scroll: The basics are there, but needs to be completed with more feature support @etag4048 and @EDGEMTech-GabrielC 
- SVG: BAsics are working the image decoder part is not ready. @onecoolx is working on it 

The others can slip to the next release. So I think realistically we can have a feature freeze at the end of February. 🙁 Is it a big issue for you? If yes, we can figure out something.

cshilwant, 2025-03-05:
> Make LVGL more appealing for MPUs: ready-to-use projects, Linux drivers, more docs, heavier features

@kisvegabor Are there any plans to add Gstreamer rendering support (similar to FFmpeg) as MPUs heavily rely on gstreamer for multimedia streaming?

kdschlosser, 2025-03-18:
Think we are going to hit the target date for this release? the end of Q1 2025 is in < 2 weeks.

kisvegabor, 2025-03-18:
> @kisvegabor Are there any plans to add Gstreamer rendering support (similar to FFmpeg) as MPUs heavily rely on gstreamer for multimedia streaming?

Yes, it just came up a few weeks ago, but it's not scheduled on our end yet. 

> Think we are going to hit the target date for this release? the end of Q1 2025 is in < 2 weeks.

I'll check the ongoing things and start wrapping it up. It might slip to beginning of April.

liamHowatt, 2025-03-18:
Reminder to revert or keep this breaking change before v9.3: https://github.com/lvgl/lvgl/pull/7310#issuecomment-2623245065

liamHowatt, 2025-03-26:
Note to self and others to ensure this is fixed before v9.3: #7797

grenhand, 2025-03-28:
Hello, may I ask if the upcoming V9.3 release will support a font manager?

kisvegabor, 2025-03-31:
By font manager, do you mean this? https://github.com/lvgl/lvgl/issues/6806

grenhand, 2025-03-31:
> 字体管理器，您指的是这个吗？[#6806](https://github.com/lvgl/lvgl/issues/6806)

Yes.

kisvegabor, 2025-04-01:
@FASTSHIFT we have discussed that the font manager will be extended beyond FreeType to support the built-in fonts too.  (https://github.com/lvgl/lvgl/issues/6806). Will have time work on it in the near future?

FASTSHIFT, 2025-04-02:
> [@FASTSHIFT](https://github.com/FASTSHIFT) we have discussed that the font manager will be extended beyond FreeType to support the built-in fonts too. ([#6806](https://github.com/lvgl/lvgl/issues/6806)). Will have time work on it in the near future?

Let's start here: https://github.com/lvgl/lvgl/pull/8038

grenhand, 2025-04-03:
By the way, for FreeType, which version do you recommend? I'm on the latest version, but I'm getting compilation errors.

[image]

FASTSHIFT, 2025-04-03:
> By the way, for FreeType, which version do you recommend? I'm on the latest version, but I'm getting compilation errors.
> 
> [image]

The compilation error is because the freetype library is not correctly linked. Please install it according to the [documentation](https://docs.lvgl.io/master/details/libs/freetype.html#for-unix) and execute:
```bash
cd build
cmake .. -DLV_USE_FREETYPE=ON
make -j
```

grenhand, 2025-04-03:
> > By the way, for FreeType, which version do you recommend? I'm on the latest version, but I'm getting compilation errors.
> > [image]
> 
> The compilation error is because the freetype library is not correctly linked. Please install it according to the [documentation](https://docs.lvgl.io/master/details/libs/freetype.html#for-unix) and execute:
> 
> cd build
> cmake .. -DLV_USE_FREETYPE=ON
> make -j

Thanks,i fixed it by adding ftsystem.c to the project.

grenhand, 2025-04-03:
I used lv_freetype_font_create to create a new font. When I applied LV_FREETYPE_FONT_STYLE_ITALIC, it worked, but LV_FREETYPE_FONT_STYLE_BOLD didn't. What am I missing?

nicusorcitu, 2025-04-04:
hi @kisvegabor 
When is the release planned?

sukesh-ak, 2025-04-05:
@kisvegabor 
What are the plans for `lv_observer`. 
Is it going to be part of lvgl core or are there plans to have a replacement of it?

I am about to start a project and was looking to build UI API for it.

FASTSHIFT, 2025-04-07:
> I used lv_freetype_font_create to create a new font. When I applied LV_FREETYPE_FONT_STYLE_ITALIC, it worked, but LV_FREETYPE_FONT_STYLE_BOLD didn't. What am I missing?

`LV_FREETYPE_FONT_STYLE_BOLD` support is also incomplete and temporarily unavailable.

cosmindanielradu19, 2025-04-10:
> Are there any open items on NXP's side?

Yes, we have changes for both VGLite an G2D.
PRs will be opened soon.

kisvegabor, 2025-04-14:
Just commented in https://github.com/lvgl/lv_font_conv/pull/134

GorgonMeducer, 2025-04-16:
@kisvegabor #8113
1. The cmsis-pack is done. Please do NOT merge as we should wait until the last minute of the cooling period. 
2. I will update the arm-2d adaptor to introduce more acceleration opportunities. (by end of this week)

kisvegabor, 2025-04-16:
Thank you Gabriel!

GorgonMeducer, 2025-04-17:
@kisvegabor 
In #7244 the LVGL intrinsics support was damaged...
We get issues like this:

```c
C:/Users/gabriel/AppData/Local/Arm/Packs/LVGL/lvgl/9.3.0/src/draw/sw/lv_draw_sw_img.c(293): error: use of undeclared identifier 'draw_unit'
  293 |                                                   draw_unit,        /* target buffer, buffer width, buffer height, buffer stride */
      |                                                   ^
C:/Users/gabriel/AppData/Local/Arm/Packs/LVGL/lvgl/9.3.0/src/draw/sw/lv_draw_sw_img.c(561): error: use of undeclared identifier 'blend_area'
  561 |         if(LV_RESULT_INVALID == LV_DRAW_SW_RGB565_RECOLOR(dest_buf, blend_area, color, mix)) {
      |                                                                     ^
C:/Users/gabriel/AppData/Local/Arm/Packs/LVGL/lvgl/9.3.0/src/draw/sw/lv_draw_sw_img.c(600): error: use of undeclared identifier 'blend_area'
  600 |         if(LV_RESULT_INVALID == LV_DRAW_SW_RGB888_RECOLOR(dest_buf, blend_area, color, mix, cf_final)) {
      |                                                                     ^
C:/Users/gabriel/AppData/Local/Arm/Packs/LVGL/lvgl/9.3.0/src/draw/sw/lv_draw_sw_img.c(600): error: use of undeclared identifier 'cf_final'
  600 |         if(LV_RESULT_INVALID == LV_DRAW_SW_RGB888_RECOLOR(dest_buf, blend_area, color, mix, cf_final)) {
```

GorgonMeducer, 2025-04-17:
@kisvegabor I have finished updating the arm2d support and the cmsis-pack is ready to review.

By the way, I have saw at least 15 last minutes PRs. I have updated the cmsis-pack to catchup with them.

If you have more, please let me know. I need to update the cmsis-pack until the last minute.

GorgonMeducer, 2025-05-04:
@kisvegabor Three weeks passed. Any plan for the release? Thank you.

kisvegabor, 2025-05-07:
Sorry, there is only one key thing left: https://github.com/lvgl/lvgl/pull/8204
Once it's merged we are good for the feature freeze week.

GorgonMeducer, 2025-05-13:
I am about to go on vacation on Friday and will not have stable access to GitHub until June.  
@kisvegabor

kisvegabor, 2025-06-04:
V9.3 was released yesterday. It was also pushed to all library/component registries. 

As no crisis was reported so far, I close this issue. 

Thank you very much for the hard of you all! ❤️