lvgl/lvgl #6770: Documentation build system, Breathe replacement

kdschlosser, 2024-08-30:
### Introduce the problem

This issue is being opened to see what things should be supported by the replacement script I am writing. The reason why Breathe needs to be replaced is because it has not been updated since 2009 and there is a slot of functionality that doesn't cross from Doxygen into the Sphinx documentation. There are also little annoyances in Breathe that can be corrected. The biggest annoyance is it's split use of the C and CPP Sphinx Domains, some objectsm get placed under the C domain and some get placed under the CPP domain and having to figure out what went where is rather annoying...

kdschlosser, 2024-08-30:
The biggest feature is the ability to use Sphinx roles and directives directly in the doxygen docstrings. An example of this is referencing other areas of the documentation. No longer is there going to be a wall between the API documentation and the rest of the documentation. Currently we are able to reference areas of the API from the rest of the documentation but not the other way around.

With the script I am writing you will be able to use Sphinx type roles and directives directly in the Doxygen docstrings.

Here are some examples
* ":ref:\`some_reference_name\`"
* ":c:macro:\`some_macro\`"
* ":c:struct:\`some_struct\`"
* ":c:union:\`some_union\`"
* ":c:type:\`some_type\`"
* ":c:member:\`some_struct::some_member\`"
* ":c:member:\`some_union::some_member\`"
* ":c:var:\`some_var\`"
* ":c:function:\`some_function\`"

kdschlosser, 2024-08-30:
There are things that do not directly translate between Doxygen and Sphinx. In those cases we can do simple things to help make it possible. The `@deprecated` directive in Doxygen is one of these cases...

Doxygen has the `@deprecated` directive and Sphinx does as well. They work differently but it can be made to work...

Doxygen's directive works as follows
```c
/**
 * @deprecated any text that is wanting to be added, maybe adding references to the replacement.
 **/
```

Sphinx is done like this.
```rst
.. deprecated:: 1.1.1

    text that is wanting to be added, maybe adding references to the replacement.
```

The thing here is the version. To make this work properly all we would need to do is add the version
```c
/**
 * @deprecated 1.1.1 text that is wanting to be added, maybe adding references to the replacement.
 **/
```

Just have the version be the first thing after the directive. A simple regex pattern to make sure it is a version that is pulled and if no version is supplied then the version gets left empty. Easy enough.. Just need to decide if the version should be the version that it was deprecated in or if it it the version when the deprecated item will be removed.

There are things like that which will become available using the replacement script for Breathe. The other thing I am going to have the script do ism actually create rst files for the API instead of doing it in ram. It will be easier to see what is going on if some kind of an error happens. we will be able to see what is actually being created so the errors and also formatting problems would be easier to correct.

kdschlosser, 2024-08-30:
Nice thing about Sphinx is it is very easy to add custom roles and directives to it. So I am able to extend the Sphinx to add the things that are in Doxygen but not in Sphinx. an example of this is the `emphasis` tag that Doxygen uses. While this can be done in Sphinx using `*text*` adding a custom role that make the connection between Doxygen and Sphinx using the tag name that Doxygen has in the XML output will make it easier to track what is happening and where. To do this I add a custom role for emphasis to the root index.rst file

```rst
.. role:: emphasis(em)
``` 

That role tells Sphinx to generate `<em>text<em/>` HTML code

Here are the custom roles that I have come up with so far.

* `.. role:: emphasis(em)`
* `.. role:: bold(b)
*.. role:: center(center)
* .. role:: ins(ins)
* .. role:: underline(u)
* .. role:: superscript(sup)
* .. role:: del(del)
* .. role:: s(s)
* .. role:: cite(cite)
* .. role:: preformatted(pre)
* .. role:: strike(del)
* .. role:: small(small)
* .. role:: subscript(sub)
* .. role:: computeroutput(samp)

Doxygen doesn't support ReST syntax. However, it does support Markdown syntax and some things are the same between ReST and Markdown. 

doxygen supported markdown..

emphasis (italics)
`*single asterisks*`
`_single underscores_`

strong emphasis (bold italics)
`**double asterisks**`
`__double underscores__`

strike through
`~~double tilde~~`

hrules

`- - -`
`______`
* hrules

verbatim block 
```
This a normal paragraph

    This is a code block

We continue with a normal paragraph again.
```

headers
```
This is a level 1 header
==============

This is a level 2 header
------------------------

# This is a level 1 header
## This is a level 2 header ##
### This is a level 3 header ###
```

block quotes
```
> This is a block quote
> spanning multiple lines
```

ordered and unordered lists
```
- Item 1

  More text for this item.

- Item 2
  + nested list item.
  + another nested item.
- Item 3

1. First item.
2. Second item.

1. item.
   a. sub list
      i. sub list
      ii. sub list
          A. sub list
          B. sub list
             I. sub list
             II. sub list
   b. sub list

2. item.
```

inline code block

Use the \`printf()\` function.

kdschlosser, 2024-09-01:
@vwheeler63 

Check out this bit of handy code...

[sphinx_table_builder.py.txt](https://github.com/user-attachments/files/16826793/sphinx_table_builder.py.txt)

I still have to get the row spanning working, the column span, vertical align and horizontal align are working without a hitch. This is to convert doxygen XML tables to ReST tables.....

vwheeler63, 2024-09-01:
> Check out this bit of handy code...

Nice.

kdschlosser, 2024-09-02:
anonymous unions and structures will be able to be added but.... when dealing with nested structures/unions they will need to have a wrapper macro...

so if we take this as an example what doxygen does is it takes the fields of the nested struct.union and it merges them with the parent struct/unions fields. While technically speaking this is what happens in compiled does it doesn't show the correct API of things specifically when dealing with a union. As we know a union shares the same memory location across all fields in the union so by merging the union's fields down into a structures fields you loose that insight.
 
```c
typedef struct {
    lv_prop_id_t id;
    union {
        int32_t num;                /**< Number integer number (opacity, enums, booleans or "normal" numbers)*/
        bool enable;                /**< booleans*/
        const void * ptr;           /**< Constant pointers  (font, cone text, etc)*/
        lv_color_t color;           /**< Colors*/
        lv_value_precise_t precise; /**< float or int for precise value*/
        lv_point_t point;           /**< Point*/
        struct {
            lv_style_value_t style; /**< Make sure it's the first element in struct. */
            uint32_t selector;      /**< Style selector, lv_part_t | lv_state_t */
        };
    };
} lv_property_t;
```

another thing that doxygen does because it merges the fields is the union vanishes for the most part. The fields that have been merged have no marker to identify that they are from a union.. apart of the "innerclass" for the compounddef for the structure it does tell you that there is a nested union.struct but it doesn't tell you where in the order of the fields the union is supposed to be placed. In order to build a proper representation I am having to load the file that contains the union and then look at the field names to determine where exactly the union is located. The whole anonymous union/struct portion makes it an even bigger mess because of the what things get named by doxygen.

The first block of code is correct C syntax and a compiler will compile it.

The following code is NOT correct C syntax. In order to name the structure/union you then have to also provide a field for the structure/union to reside in. That pretty much defeats the purpose to declaring a structure/union without a name when they are nested.

```c
typedef struct {
    lv_prop_id_t id;
    union anonymous_union {
        int32_t num;                /**< Number integer number (opacity, enums, booleans or "normal" numbers)*/
        bool enable;                /**< booleans*/
        const void * ptr;           /**< Constant pointers  (font, cone text, etc)*/
        lv_color_t color;           /**< Colors*/
        lv_value_precise_t precise; /**< float or int for precise value*/
        lv_point_t point;           /**< Point*/
        struct anonymous_struct {
            lv_style_value_t style; /**< Make sure it's the first element in struct. */
            uint32_t selector;      /**< Style selector, lv_part_t | lv_state_t */
        };
    };
} lv_property_t;
```

funny thing... doxygen reads this without any problem. and it does so without merging the fields together or creating a field for the structure/union.. YAY!!! That's actually helpful. Each field in a structure also has the line number that the field exists on in the code file. it also has the line number for where the structure/union is positioned as well. so using the line numbers I am able to determine exactly where in the structure/union the nested structure/union exists. YAY!!...

Sphinx has a big leg up compared to doxygen. It is able to properly handle anonymous structures and unions by prepending the name with an `@`. so if we write the code like this...

```c
typedef struct {
    lv_prop_id_t id;
#ifdef DOXYGEN
    union _anon {
#else
    union {
#endif /*DOXYGEN*/
        int32_t num;                /**< Number integer number (opacity, enums, booleans or "normal" numbers)*/
        bool enable;                /**< booleans*/
        const void * ptr;           /**< Constant pointers  (font, cone text, etc)*/
        lv_color_t color;           /**< Colors*/
        lv_value_precise_t precise; /**< float or int for precise value*/
        lv_point_t point;           /**< Point*/
#ifdef DOXYGEN
        struct _anon {
#else
        struct {
#endif /*DOXYGEN*/
            lv_style_value_t style; /**< Make sure it's the first element in struct. */
            uint32_t selector;      /**< Style selector, lv_part_t | lv_state_t */
        };
    };
} lv_property_t;
```

I am now able to properly locate where the nested union/struct is located in the parent union/struct and I am also able to properly mark the union/struct as being anonymous so sphinx is able to build a correct representation of it in the documentation.. YAY...

kdschlosser, 2024-09-02:
another thing I tested and found that it works is for function like macros. A function like macro is a macro that takes parameters. The issue with documenting these is the lack of typing and what it's sort of return value is. 

This code does work..
```c
/**
 * Test Macro
 *
 * @param x               x parameter
 * @param b              b parameter
 *  
 * @retval uint32_t    description of what is being returned
 **/
#define TEST_MACRO(x, b)   ((uint32_t)x + (uint32_t)b)
```

OK sweet that works. Since that does properly populate into a macro's docstring I can use some string kung fu to properly read the following..

```c
/**
 * Test Macro
 *
 * @param x               ``lv_coord_t`` x parameter
 * @param b              ``lv_coord_t`` b parameter
 *
 * @retval uint32_t    description of what is being returned
 **/
#define TEST_MACRO(x, b)   ((uint32_t)x + (uint32_t)b)
```

Now that is handy isn't it..

kdschlosser, 2024-09-02:
the `@retval` doxygen directive is mnot what I would call a "complete" directive. Here is the description of what it is...

> Starts a description for a function's return value with name <return value>, followed by a description of the return value. The text of the paragraph that forms the description has no special internal structure. All visual enhancement commands may be used inside the paragraph. Multiple adjacent \retval commands will be joined into a single paragraph. Each return value description will start on a new line. The \retval description ends when a blank line or some other sectioning command is encountered.

This is really vague isn't it? There is a clue as to what it's purpose is... `Each return value description`.. meaning you can have more than one of these... But a function is only able to return a single thing so why would there be a need to use more than one of them? It's intended use is for dealing with pointers that get passed as parameters so they can be filled. Those are also return values and you can have more then one of them that gets filled..

Nice thing about this directive is there is no check that gets done to see if the first arg passed to the directive actually exists as a parameter of the function (unlike @param) because of this we have the ability to set the first argument to anything that we want and it will not cause any problems with doxygen, no warnings or errors... so we can use this to pass the type as well as a description of an actual return value from a function... While it is able to be used with `"out"` parameters at the same time I do not feel that we need to use the directive for that purpose. That information can be relayed to the user in the parameter description.

The reason why this is nice is because of how the Sphinx ReST syntax is...

so lets take a look at this function...

```c
/**
 * @brief Test function 2
 *
 * Detailed Description of test function 2

 * @param param1       Parameter 1
 * @param param2       Parameter 2
 *
 * @retval uint32_t    The return value is a 32 bit unsigned integer
 *
 * @exception assert   Test Exception
 */
uint32_t function_test2(uint32_t param1, uint32_t param2);
```

this is the doxygen output...
```xml
<memberdef kind="function" id="test_8h_1a8d9604caaefc9b0d0e12171981570f00" prot="public" static="no" const="no" explicit="no" inline="no" virt="non-virtual">
        <type>uint32_t</type>
        <definition>uint32_t function_test2</definition>
        <argsstring>(uint32_t param1, uint32_t param2)</argsstring>
        <name>function_test2</name>
        <param>
          <type>uint32_t</type>
          <declname>param1</declname>
        </param>
        <param>
          <type>uint32_t</type>
          <declname>param2</declname>
        </param>
        <briefdescription>
<para>Test function 2. </para>
        </briefdescription>
        <detaileddescription>
<para>Detailed Description of test function 2</para>
<para><parameterlist kind="param"><parameteritem>
<parameternamelist>
<parametername>param1</parametername>
</parameternamelist>
<parameterdescription>
<para>Parameter 1 </para>
</parameterdescription>
</parameteritem>
<parameteritem>
<parameternamelist>
<parametername>param2</parametername>
</parameternamelist>
<parameterdescription>
<para>Parameter 2</para>
</parameterdescription>
</parameteritem>
</parameterlist>
<parameterlist kind="retval"><parameteritem>
<parameternamelist>
<parametername>uint32_t</parametername>
</parameternamelist>
<parameterdescription>
<para>The return value is a 32 bit unsigned integer</para>
</parameterdescription>
</parameteritem>
</parameterlist>
<parameterlist kind="exception"><parameteritem>
<parameternamelist>
<parametername>assert</parametername>
</parameternamelist>
<parameterdescription>
<para>Test Exception </para>
</parameterdescription>
</parameteritem>
</parameterlist>
</para>
        </detaileddescription>
        <inbodydescription>
        </inbodydescription>
        <location file="test.h" line="312" column="10" declfile="test.h" declline="312" declcolumn="10"/>
      </memberdef>
```

and this is the result of that code...

```rst
.. c:function:: uint32_t function_test2(uint32_t param1, uint32_t param2)

    Test function 2. 
    
    Detailed Description of test function 2

    :param param1: Parameter 1
    :type param1: uint32_t

    :param param2: Parameter 2
    :type param2: uint32_t

    :returns: The return value is a 32 bit unsigned integer
    :rtype: uint32_t

    :raises AssertionError: Test Exception
```

kdschlosser, 2024-09-02:
The doxygen syntax isn't changing at all. It is staying exactly how it is. Things like the `@deprecated` directive in doxygen is a good thing, but it would be better if the version was able to be passed to it. we could do a simple check to see if the first word is a version number and if it is then we can use it in the the title for a caption box if there is a caption or simply have a specially formatted line indicating that the object is deprecated with the version in which it was deprecated. 

It is typically a nice thing to have a policy in place that states deprecated items will be removed in a specific number of major or minor version changes. 

Supplying that version to a user allows them to be able to know what kind of a time frame they are having to work with before needing to update. It also provides the maintainers a way to check for things that should be removed. 

Doxygen is still working as it should the only thing that is getting done is the script that generates the rst files is looking to see if the version is there and it puts it in a different area instead of in the caption text.

The `@retval ` directive is for returned values plain and simple. The documentation in doxygen for that directive is not explicit in it being used for parameters or for actual returned values nor is it explicit is what goes in the "name" field it doesn't say "parameter name" and it also doesn't say "type name" it just says "name", so the name of what? The type that is being returned? or the name of the parameter? A return value is a return value, doesn't matter if it is being returned via a parameter or being returned directly from the call to the function. f it is being returned directly from the function call something has to be placed in the "name". The only name that can be provided is the type name.

Once again, not altering anything in doxygen. and not trying to change what the directives do. I am trying to align the doxygen directives with the ones that are in sphinx to have things work 100% correctly and also offer additional abilities like being able to reference portions of the sphinx documentation from the doxygen docs. It would be nice to be able to provide a link to the section on how to set up a display from the documentation for the `lv_display_create` function.

I still have not decided if we should replace breathe. I have been trying to locate a way to push the data that breathe collects from doxygen through the parser for sphinx. This is something I know is able to be done, it's just a matter of finding it. The issue is there is a quite a bit of voodoo magic code in both sphinx and also in breathe which makes it very hard to follow the data path.

The purpose of this thread was only for me to write down things I have come across so I can go back and look at them. That's all it's for.

kisvegabor, 2024-09-03:
Clear, thanks!

closed 2024-09-25 by lvgl-bot