lvgl/lvgl #6772: When rotating an object, the shadow will appear automatically?

M-D-777, 2024-08-31:
### LVGL version

v9.x

### What happened?

When rotating an obj, a shadow will automatically appear around the obj, even if I set the shadow width to 0.
When not rotating the obj, there is no shadow line.
[image]
I would like to not have this shadow or have the option to turn it off.
Because I want to use this simple square obj to cover other obj to achieve a polygon effect.
Like This:
[image]

### How to reproduce?

Both c and micropython versions can be reproduced
```obj = lv.obj(lv.screen_active())
obj.set_size(100, 100)
obj.set_pos(100, 100)
obj.set_style_radius(0, lv.PART.MAIN | lv.STATE.DEFAULT)
obj.set_style_shadow_width(0, lv.PART.MAIN | lv.STATE.DEFAULT)
# obj.set_style_shadow_spread(0, lv.PART.MAIN)
obj.set_style_border_width(0, lv.PART.MAIN | lv.STATE.DEFAULT)
obj.set_style_bg_color(lv.screen_active().get_style_bg_color(lv.PART.MAIN), lv.PART.MAIN | lv.STATE.DEFAULT)
obj.remove_flag(lv.obj.FLAG.SCROLLABLE)
obj.set_style_outline_width(0, lv.PART.MAIN | lv.STATE.DEFAULT)
obj.set_style_transform_pivot_x(50, 0)
obj.set_style_transform_pivot_y(50, 0)
obj.set_style_transform_rotation(450, 0)```

M-D-777, 2024-08-31:
https://github.com/lvgl/lv_micropython/tree/841ece1321d462745fff3ef054780ae569292683
I have confirmed that this version does not have this issue. Looking forward to the new version resolving this issue. Thank you

liamHowatt, 2024-09-04:
I reproduced it. I think it might not be a shadow. I think it's a rendering bug with rotation. When I set shadow opacity to transparent or shadow color to white, it's still there. I'm not sure what it would take to fix this. cc @kisvegabor

kisvegabor, 2024-09-04:
I've just opened a PR to fix it. See #6794

The problem was with mixing the black color of transparent pixels.

PR #6794 (2024-09-05): fix(draw): fix artifact when rotating ARGB8888 images
fixes #6772

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/ [... cut]
closed 2024-09-05 by FASTSHIFT