lvgl/lvgl #6774: Invalid conversion from 'int' to 'x' [-fpermissive]

Kirbyrawr, 2024-08-31:
### LVGL version

9.2

### What happened?

Not a bug per se but rather a question/doubt.

After updating from version 9.1 of LVGL to 9.2 i'm encountering this error:

```
src\library\StylesLibrary.cpp: In function 'void SetupStyles()':
src\library\StylesLibrary.cpp:106:73: error: invalid conversion from 'int' to 'lv_border_side_t' [-fpermissive]
  106 |     lv_style_set_border_side(&roller_selected_style, LV_BORDER_SIDE_TOP | LV_BORDER_SIDE_BOTTOM);
```

Based on what i saw in version 9.1 it was like this:
https://github.com/lvgl/lvgl/blob/release/v9.1/src/misc/lv_style.h#L115

And in version 9.2 like this: 
https://github.com/lvgl/lvgl/blob/release/v9.2/src/misc/lv_style.h#L99

Is it correct that now i need to cast it like this:
`lv_style_set_border_side(&roller_selected_style, static_cast<lv_border_side_t>(LV_BORDER_SIDE_TOP | LV_BORDER_SIDE_BOTTOM));`
Or i'm doing something wrong? 

Thanks.

supremeneuron, 2024-08-31:
I've encountered the same issue. I, too, would appreciate some insights.

bladerunner09, 2024-09-03:
It is not necessary to do an explicit conversion. Previously, your function took an argument of type int, but now it takes lv_border_side_t. In theory, you can declare and initialize a variable like lv_border_side_t, then substitute it.

Kirbyrawr, 2024-09-03:
> It is not necessary to do an explicit conversion. Previously, your function took an argument of type `int`, but now it takes `lv_border_side_t`. In theory, you can declare and initialize a variable like `lv_border_side_t`, then substitute it.

The function is not mine is of LVGL, they removed the int type and is now more strict it seems.

liamHowatt, 2024-09-04:
Okay, the problem is that the enum [used to be typedef'd as an integer type](https://github.com/lvgl/lvgl/blob/b25519abce8aaa5fd5dd2ab02a2acdce35ddf555/src/misc/lv_style.h#L128). Now it's a real `enum`. When enum members are or'd together like `LV_BORDER_SIDE_TOP | LV_BORDER_SIDE_BOTTOM` then the result type is an `int`, which C++ does not implicitly cast back to an `enum` like C does. If they aren't bitwise-or'd together there is no error. Also, if C is used there is no error.

We missed this because all of the test cases etc. are in C. It was a intentional decision to stop typedefing enums as integer types. I'm not sure what the best solution is, but `static_cast` is totally safe and correct solution in your case.

Kirbyrawr, 2024-09-04:
> Okay, the problem is that the enum [used to be typedef'd as an integer type](https://github.com/lvgl/lvgl/blob/b25519abce8aaa5fd5dd2ab02a2acdce35ddf555/src/misc/lv_style.h#L128). Now it's a real [...]

Thank you Liam, makes sense, i will close the issue then. 👍

closed 2024-09-04 by Kirbyrawr