lvgl/lvgl #6777: Forum is Down

kdschlosser, 2024-09-01:
### LVGL version

master

### What happened?

I am getting a 502 Bad Gateway Error when accessing the LVGL forum. I have flushed my local DNS resolver cache and that didn't help at all. It looks like it is connecting to the site which is weird....

```
GET
https://forum.lvgl.io/
[HTTP/2 502  492ms]

This page is in Quirks Mode. Page layout may be impacted. For Standards Mode use “<!DOCTYPE html>”. forum.lvgl.io
Error: An unexpected error occurred spoofer.js:1:38935
GET
https://forum.lvgl.io/favicon.ico
[HTTP/2 204  134ms]

```

You can see it grabs the `favicon.ico` amnd then it tanks. It states there is some kind of an issue with `spoofer.js` at line 38935.

I get a cookie for the forum and everything...

vwheeler63, 2024-09-01:
I too am getting "502 Bad Gateway" error:

```
<html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx</center>
</body>
</html>
```

cybershoe, 2024-09-02:
Experiencing this problem as well

kisvegabor, 2024-09-02:
Just fixed it. I don't know what happened to be honest. We needed to reboot the VPN and reconfigure the Forum.

closed 2024-09-02 by kisvegabor