lvgl/lvgl #6778: cpp  fatal error: ../../src/themes/lv_theme_private.h: No such file or directory

SalarAmirSalari, 2024-09-01:
### LVGL version

9.1.0 - 9.3.0

### What happened?

I am encountering an issue with the LVGL library:
```
fatal error: ../../src/themes/lv_theme_private.h: No such file or directory
    1 | #include "../../src/themes/lv_theme_private.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1
Compilation error: exit status 1
```
Everything is configured as explained in the LVGL documentation (including ```lv_conf.h``` and ```User_Setup.h```). After checking the directory, the file exists, and I have no clue what is wrong (I am a beginner, btw).

I am getting this error on both Windows and Arch Linux, so I suspect it might be an issue with the library itself.

amirmohammadrsa, 2024-09-02:
Same issue

C47D, 2024-09-02:
What file is the error on?

liamHowatt, 2024-09-03:
Is it `examples/styles/lv_example_style_14.c`? It's a strange issue. I can't tell what's wrong. If you hadn't already tried on two different systems, I would've suggested doing a clean build.

bladerunner09, 2024-09-03:
Try correcting the file path.
#include "../../src/themes/lv_theme_private.h" -> #include "../src/themes/lv_theme_private.h"
or
#include "../../src/themes/lv_theme_private.h" -> #include "src/themes/lv_theme_private.h"

liamHowatt, 2024-09-03:
Good idea! Also try `#include "lvgl/src/themes/lv_theme_private.h"` How are you building/compiling your project?

vwheeler63, 2024-09-03:
Forgive me if I am missing something, but from `examples/styles/lv_example_style_14.c`
```
#include "../../src/themes/lv_theme_private.h"
```
is actually a correct path.  The only way to cause that error is if the file is actually not there....  This would not be hard to have happen since the file was only added a month ago.  @SalarAmirSalari , when you look in your local `lvgl/src/themes` directory, is the `lv_theme_private.h` file actually there?  If not, then you probably have a "snapshot" of LVGL before this file was created.

- Are you working from your LVGL fork (which does have the file)?  Or directly cloned from `lvgl/lvgl`?
- Is it in your local `lvgl/src/themes/` directory?

JeroenRegtien, 2024-09-03:
I had the same problem. On my Mac OSC installation I had to edit the "example_style_14.c" file and insert an extra "../" in the path for "lv_theme_private.h". The error disappeared but then a new compilation error occurred for file "lv_example_scale_7.c". Same issue here, add an extra "../" in the path of header file "lvgl_private.h".

My lvgl demo for the gigadisplay shield then compiles again, but displays garbage on the screen. I am concerned the latest lvgl library release (I was using 9.2.0) is corrupt or hasn't been checked properly before release.

I reverted back to 9.0.0 and then had no compilation nor run time problems... I then moved to 9.1.0 and this worked fine too.
So my suggestion is to avoid 9.2.0 until these errors are fixed. I have however no idea how to find that out...

liamHowatt, 2024-09-04:
As vwheeler63 said, the most important question is: is the file in the directory it should be in? If it's not, are you using some kind of fork that was only partially updated to v9.3?

sanimalp, 2024-09-06:
this would only happen if you copied examples into src? right? 

The paths are correct in the files you are referring to, in the context of this repository. `example_style_14.c` and `lv_example_scale_7.c` have correct paths from the root of the repository, but if you copied your example folder into source, it would cause these problems. Your options are: remove the example folder from source, or fix the paths if you intend to keep the folder in src. 

here is a prominent tutorial that suggests the copy of examples into src, as a possible source for this bug https://randomnerdtutorials.com/lvgl-cheap-yellow-display-esp32-2432s028r/

liamHowatt, 2024-09-09:
Thanks sanimalp.

You can use the examples by compiling them where they are and calling `lv_example_style_14()` from your code or you can move or repurpose the examples of course, but the include paths may need to be updated, as sanimalp said.

LuckierBread, 2024-09-12:
This happened to me because my examples and demos folders were copied into my src folder. they are normally outside the src. I did this because I needed to access lv_demos.h but I found it better to access it via `#include <../demos/lv_demos.h>`
It saved me form having to rewrite a bunch of other files.

ouening, 2024-09-14:
Same issue in vesion 9.2

FeeJayHere, 2024-10-03:
Hello, I would like to add some info to this:
The problem is happening when using Arduino IDE
When installing lvgl version 9.2 from library manager, after configuring TFT_eSPI and lv_conf.h, LVGL docs have a note here: https://docs.lvgl.io/master/integration/framework/arduino.html#use-the-examples-and-demos to copy examples and demos folders to src. After copying these folders, the error mentioned appear.
I think that the docs for master (latest) should be updated or code changed to fix that

kissa96, 2024-10-24:
Hi, is this still a valid and open issue?

kisvegabor, 2024-10-30:
Closing it in lack of feedback. Let us know if the issue is still ongoing.

razarahil, 2024-11-19:
I'm still getting this error when compiling.

qq71680264, 2024-11-21:
> Closing it in lack of feedback. Let us know if the issue is still ongoing.

Arduino IDE 2.3.3
ESP32 Package 3.0.7
LVGL library 9.2.2

Still getting this error when compiling.

kisvegabor, 2024-11-27:
Hey,
I've check this issue again and found a solution, please test it with https://github.com/lvgl/lvgl/pull/7366

minou65, 2024-11-30:
Hi I have the same issue, also with this mr
> Hey, I've check this issue again and found a solution, please test it with #7366

kisvegabor, 2024-12-02:
Oh, thank! Maybe with this? https://github.com/lvgl/lvgl/pull/7397 :crossed_fingers: 

@fbiego Could you also try if https://github.com/lvgl/lvgl/pull/7397 can be compiled in Arduino?

fbiego, 2024-12-03:
I have tried compiling #7397 in Arduino. @kisvegabor 

### Without copying the demos to src
`#include <../demos/lv_demos.h> /* include demos */`
It results in an error
> undefined reference to lv_demo_ebike

### When demos is copied to src

`#include <demos/lv_demos.h> /* include demos */`

all files in demos folder with `../../src/` needs to be changed to `../../`
eg
`#include "../../src/lvgl_private.h"`
to
`#include "../../lvgl_private.h"`

The second option compiles successfully

kisvegabor, 2024-12-03:
Thank you @fbiego. I've updated the PR. I wonder if there is a way to add an Arduino build test to our CI. Maybe with Arduino CLI?

fbiego, 2024-12-03:
> Thank you @fbiego. I've updated the PR. I wonder if there is a way to add an Arduino build test to our CI. Maybe with Arduino CLI?

I'm not familiar with Arduino CLI, I'll research on that

minou65, 2024-12-04:
Hi @kisvegabor 
> Oh, thank! Maybe with this? #7397 🤞
> 
> @fbiego Could you also try if #7397 can be compiled in Arduino?

I can confirm that with this MR I do not recive anymore this errors. I am always not able to build the examples from waveshare. But that is an other story

kisvegabor, 2024-12-05:
Thank you for the feedback. The PR is merged so I close this issue. 🎉

closed 2024-12-05 by kisvegabor

BoraMD, 2025-02-07:
I got similar error on 
MacOS 15.1.1
Arduino IDE 2.3.4
LVGL 9.2.0
ESP32 3.1.1 

When I just try LVGL_Arduino sketch I can compile and it shows Hello Arduino text.
However if I try example or demo it gives the same error:
/Users/boramutludogan/Documents/Arduino/libraries/lvgl/src/examples/styles/lv_example_style_14.c:1:10: fatal error: ../../src/themes/lv_theme_private.h: No such file or directory
    1 | #include "../../src/themes/lv_theme_private.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

There is lv_theme_private.h file in that directory.
Tried changing the path according to above suggestions but it gives more errors.

BoraMD, 2025-02-07:
Also I managed to make demo work. Copied demos folder to src and tried lv_demo_widgets(); and enabled demo widgets on lv_conf it works.

Mr-Newlove, 2025-03-26:
I am still having this issue on v9.2.2.

https://forum.lvgl.io/t/wt32-sc01-plus-arduino-v9-widget-demo-struggle/20641

my current arduino code:

```
#include <LovyanGFX.hpp>
//#include "Libraries\lvgl\9.2.2\lvgl\src\themes\lv_theme_private.h"
#include <lvgl.h>
//#include <demos/lv_demos.h>
//#include <examples/lv_examples.h>
#include <SPI.h>

#define screenWidth 480
#define screenHeight 320

class LGFX : public lgfx::LGFX_Device
{
    lgfx::Panel_ST7796      _panel_instance;
    lgfx::Bus_Parallel8 _bus_instance;   // 8-bit parallel bus instance (ESP32 only)
    lgfx::Light_PWM     _light_instance;
    lgfx::Touch_FT5x06           _touch_instance; // FT5206, FT5306, FT5406, FT6206, FT6236, FT6336, FT6436
public:
    LGFX(void);
};

LGFX::LGFX(void)
{
    { // Bus config for LCD
        auto cfg = _bus_instance.config();    // inital bus configuration data structure
        // 8 bit parallel
              //cfg.i2s_port = I2S_NUM_0;     // Select the I2S port to use (I2S_NUM_0 or I2S_NUM_1) (Use the I2S LCD mode of ESP32)
        cfg.freq_write = 20000000;    // Bus transmission clock (maximum 20MHz, rounded to an integer division of 8MHz?)
        cfg.pin_wr = 47;              // WR Pin
        cfg.pin_rd = -1;              // RD Pin
        cfg.pin_rs = 0;              // RS(D/C) Pin
        cfg.pin_d0 = 9;              // D0 Pin
        cfg.pin_d1 = 46;              // D1 Pin
        cfg.pin_d2 = 3;              // D2 Pin
        cfg.pin_d3 = 8;              // D3 Pin
        cfg.pin_d4 = 18;              // D4 Pin
        cfg.pin_d5 = 17;              // D5 Pin
        cfg.pin_d6 = 16;              // D6 Pin
        cfg.pin_d7 = 15;              // D7 Pin
        _bus_instance.config(cfg);    // Set this config to the bus instance
        _panel_instance.setBus(&_bus_instance);      // Assign this bus to the panel
    }

    { // LCD control settings
        auto cfg = _panel_instance.config();    // Intial LCD configuration data structure

        cfg.pin_cs = -1;  // CS pin, (-1 to disable for parallel interface)
        cfg.pin_rst = 4;  // RST pin, (-1 to disable for parallel interface)
        cfg.pin_busy = -1;  // BUSY pin, (-1 to disable for parallel interface)

        // General default values for LCD, you may needc to comment some out to get it working

        cfg.panel_width = 320;  // LCD pixel width
        cfg.panel_height = 480;  // LCD pixel height
        cfg.offset_x = 0;  // LCD pixel ofset X(width)
        cfg.offset_y = 0;  // LCD pixel ofset Y(height)
        cfg.offset_rotation = 0;  // LCD rotation 0~7 (4~7 are upside down)
        cfg.dummy_read_pixel = 8;  // Number of dummy reads bits before first pixel read
        cfg.dummy_read_bits = 1;  // Number of dummy read bits before reading non-pixel data
        cfg.readable = true;  // Set to true if data can be read
        cfg.invert = true;  // Panel inverted
        cfg.rgb_order = false;  // Set to true if panel Red and Blue are swapped
        cfg.dlen_16bit = false;  // Set to true if panel transmists data in 16 bit units via parallel or SPI
        cfg.bus_shared = true;  // Set to true of bus is shared with an SD card(so bus can preform drawJpGFile etc)
[... 220 lines cut ...]
        if (base_dsc->id1 == 2) {
            line_dsc->width = 2;
            line_dsc->dash_gap = 0;
            line_dsc->dash_width = 0;
        }
        else {
            line_dsc->width = 2;
            line_dsc->dash_gap = 6;
            line_dsc->dash_width = 6;
        }

        if (base_dsc->id1 == 1 || base_dsc->id1 == 3) {
            line_dsc->color = lv_palette_main(LV_PALETTE_GREEN);
        }
        else {
            line_dsc->color = lv_palette_lighten(LV_PALETTE_GREY, 1);
        }
    }
}
```

with error:
```The lvgl library encountered an unknown path resolve error.Z*: 1:10: fatal error: ../../src/themes/lv_theme_private.h: No such file or directory```

I have copied the example and demo files into scr. I have tried many different alternative locations for the lv_theme_private.h file. I am not sure what do do.

TechOverflow, 2025-05-22:
In `lvgl/src/examples/styles/lv_example_style_14.c` change:
from `#include "../../src/themes/lv_theme_private.h"`
to `#include "../../themes/lv_theme_private.h"`

In `lvgl/src/examples/widgets/scale/lv_example_scale_7.c` change:
from `#include "../../../src/lvgl_private.h"`
to `#include "../../../lvgl_private.h"`

The `src/` makes no sense as these examples are already in `src/` when you copy `examples/` into `src/` as per the installation instructions.

There may be other bad links in other files. These were just the ones I encountered in the example I was trying to compile.

PR #7366 (2024-12-02): fix(arduino): allow including lvgl_private.h even if the examles and demos are in the src folder
fixes #6778

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig) [... cut]