lvgl/lvgl #6781: undefined reference to lv_bidi_xxx

xlla, 2024-09-02:
### LVGL version

v9.3.0-dev

### What happened?

after merge #6775 , I can't build it for esp32 port, the error log is 
<pre>
...

.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(lv_mp.c.obj):(.rodata.mp_lv_bidi_process_paragraph_mpobj+0xc): undefined reference to `lv_bidi_process_paragraph'
.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(lv_mp.c.obj):(.rodata.mp_lv_bidi_get_visual_pos_mpobj+0xc): undefined reference to `lv_bidi_get_visual_pos'
.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(lv_mp.c.obj):(.rodata.mp_lv_bidi_get_logical_pos_mpobj+0xc): undefined reference to `lv_bidi_get_logical_pos'
.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(lv_mp.c.obj):(.rodata.mp_lv_bidi_detect_base_dir_mpobj+0xc): undefined reference to `lv_bidi_detect_base_dir'
.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(lv_mp.c.obj):(.rodata.mp_lv_bidi_process_mpobj+0xc): undefined reference to `lv_bidi_process'

collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
make: *** [deploy] Error 2

</pre>

in my lv_conf.h, 

<pre>
...
#define LV_USE_PRIVATE_API		1
...
#define LV_USE_BIDI 0
</pre>

I think the macro LV_USE_BIDI judgement should be put in misc/lv_bidi_private.h too, after add it , I can build and flash firmware.

in lv_bidi_private.h

<pre>
/**********************
 * GLOBAL PROTOTYPES
 **********************/

#if LV_USE_BIDI

...

#endif
/**********************
 *      MACROS
 **********************/

</pre>

liamHowatt, 2024-09-03:
The error you got usually happens when a C/C++ file is calling those functions. I'm surprised adding the `#if` in the private header fixed the problem for you. Are you using [lv_port_esp32](https://github.com/lvgl/lv_port_esp32)?

xlla, 2024-09-04:
I am using esp32 port in [littlevgl/lv_micropython](https://github.com/littlevgl/lv_micropython.git) ,  maybe those struct defines in the private header just used by gen_mpy.py to generate out correct stubs for callback function for micropython.

however,  it get compiled.

xlla, 2024-09-05:
I'm happy to share it.

PR #6801 (2024-09-05): fix(micropython): missing bidi private header feature guard
Fixes #6781

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`as [... cut]
closed 2024-09-05 by xlla