lvgl/lvgl #6782: LVGL help create

fenixfz, 2024-09-03:
### Introduce the problem

I needed help with just one more thing.
I'm trying to implement a test I made in Squareline Studio but the UI isn't starting and the screen stays white. Maybe someone could help me, this is my code.
I can run the demos but I can't implement it created by me
I have esp32 8048s070

#include <lvgl.h>
#include <Arduino_GFX_Library.h>
#include "ui.h"
#include "Arduino.h"
#include <TAMC_GT911.h>

#define TOUCH_SCL 20
#define TOUCH_SDA 19
#define TOUCH_INT -1
#define TOUCH_RST 38
#define TOUCH_ROTATION ROTATION_NORMAL
#define TOUCH_MAP_X2 0
#define TOUCH_MAP_X1 800
#define TOUCH_MAP_Y2 480
#define TOUCH_MAP_Y1 0

//---- TFT Set ------------------------------------------
#define GFX_BL 2
// 7"
Arduino_ESP32RGBPanel rgbpanel = new Arduino_ESP32RGBPanel(
41 / DE /, 40 / VSYNC /, 39 / HSYNC /, 42 / PCLK /,
14 / R0 /, 21 / R1 /, 47 / R2 /, 48 / R3 /, 45 / R4 /,
9 / G0 /, 46 / G1 /, 3 / G2 /, 8 / G3 /, 16 / G4 /, 1 / G5 /,
15 / B0 /, 7 / B1 /, 6 / B2 /, 5 / B3 /, 4 / B4 /,
0 / hsync_polarity /, 180 / hsync_front_porch /, 30 / hsync_pulse_width /, 16 / hsync_back_porch /,
0 / vsync_polarity /, 12 / vsync_front_porch /, 13 / vsync_pulse_width /, 10 / vsync_back_porch */);

Arduino_RGB_Display gfx = new Arduino_RGB_Display(
800 / width /, 480 / height /, rgbpanel, 0 / rotation /, true / auto_flush */);

int touch_last_x = 0;
int touch_last_y = 0;

TAMC_GT911 ts = TAMC_GT911(TOUCH_SDA, TOUCH_SCL, TOUCH_INT, TOUCH_RST, max(TOUCH_MAP_X1, TOUCH_MAP_X2), max(TOUCH_MAP_Y1, TOUCH_MAP_Y2));

//---- Funções de Inicialização --------------------------------------------------

void tft_init() {
// Inicializa o display TFT
gfx->begin();
gfx->fillScreen(0xFFFF); // white
Serial.println("TFT initialized");

}

void touch_init() {
// Inicializa o touch screen
ts.begin(); // Inicializa o touch screen
Serial.println("Touch initialized"); // Confirma a inicialização
}

//---- Main --------------------------------------------------

void setup() {
Serial.begin(115200);

pinMode(GFX_BL, OUTPUT); // Define o pino de controle da luz de fundo como saída
digitalWrite(GFX_BL, HIGH);
delay(10); // Espera 10ms para a luz de fundo estabilizar

tft_init(); // Inicializa o TFT
lv_init();
touch_init(); // Inicializa o touch screen

// Chama a função para inicializar a interface gerada pelo SquareLine Studio
ui_init();

Serial.println("Setup done");
}

void loop() {
lv_timer_handler(); /* Deixa o GUI fazer seu trabalho */
delay(5);
}

[image]
[image]
[image]

liamHowatt, 2024-09-04:
Would it be possible for you to find and share the contents of the `ui_init` function?

fenixfz, 2024-09-04:
> Would it be possible for you to find and share the contents of the `ui_init` function?

Good morning, of course, no problem.
Attached I also send the complete code.
and thanks for the help.

[teste.zip](https://github.com/user-attachments/files/16863005/teste.zip)

ui.h

// This file was generated by SquareLine Studio
// SquareLine Studio version: SquareLine Studio 1.4.2
// LVGL version: 8.3.11
// Project name: SquareLine_Project

#ifndef _SQUARELINE_PROJECT_UI_H
#define _SQUARELINE_PROJECT_UI_H

#ifdef __cplusplus
extern "C" {
#endif

#if defined __has_include
#if __has_include("lvgl.h")
#include "lvgl.h"
#elif __has_include("lvgl/lvgl.h")
#include "lvgl/lvgl.h"
#else
#include "lvgl.h"
#endif
#else
#include "lvgl.h"
#endif

#include "ui_helpers.h"
#include "ui_events.h"

  // SCREEN: ui_Screen1
  void ui_Screen1_screen_init(void);
  extern lv_obj_t* ui_Screen1;
  void ui_event_Button1(lv_event_t* e);
  extern lv_obj_t* ui_Button1;
  extern lv_obj_t* ui_Checkbox1;
  // SCREEN: ui_Screen2
  void ui_Screen2_screen_init(void);
  extern lv_obj_t* ui_Screen2;
  void ui_event_Button2(lv_event_t* e);
  extern lv_obj_t* ui_Button2;
  extern lv_obj_t* ui____initial_actions0;

  LV_IMG_DECLARE(ui_img_43bae6e22231aa5367dd662a05b2ab17_png);  // assets/43bae6e22231aa5367dd662a05b2ab17.png

  void ui_init(void);

#ifdef __cplusplus
} /*extern "C"*/
#endif

#endif

ui,c

// This file was generated by SquareLine Studio
// SquareLine Studio version: SquareLine Studio 1.4.2
// LVGL version: 8.3.11
// Project name: SquareLine_Project

#include "ui.h"
#include "ui_helpers.h"

///////////////////// VARIABLES ////////////////////

// SCREEN: ui_Screen1
void ui_Screen1_screen_init(void);
lv_obj_t * ui_Screen1;
void ui_event_Button1(lv_event_t * e);
lv_obj_t * ui_Button1;
lv_obj_t * ui_Checkbox1;

// SCREEN: ui_Screen2
void ui_Screen2_screen_init(void);
lv_obj_t * ui_Screen2;
void ui_event_Button2(lv_event_t * e);
lv_obj_t * ui_Button2;
lv_obj_t * ui____initial_actions0;
const lv_img_dsc_t * ui_imgset_43bae6e22231aa5367dd662a05b2ab[1] = {&ui_img_43bae6e22231aa5367dd662a05b2ab17_png};

///////////////////// TEST LVGL SETTINGS ////////////////////
#if LV_COLOR_DEPTH != 16
    #error "LV_COLOR_DEPTH should be 16bit to match SquareLine Studio's settings"
#endif
#if LV_COLOR_16_SWAP !=0
    #error "LV_COLOR_16_SWAP should be 0 to match SquareLine Studio's settings"
#endif

///////////////////// ANIMATIONS ////////////////////

///////////////////// FUNCTIONS ////////////////////
void ui_event_Button1(lv_event_t * e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t * target = lv_event_get_target(e);
    if(event_code == LV_EVENT_CLICKED) {
        _ui_screen_change(&ui_Screen2, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0, &ui_Screen2_screen_init);
    }
}
void ui_event_Button2(lv_event_t * e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t * target = lv_event_get_target(e);
    if(event_code == LV_EVENT_CLICKED) {
        _ui_screen_change(&ui_Screen1, LV_SCR_LOAD_ANIM_FADE_ON, 500, 0, &ui_Screen1_screen_init);
    }
}

///////////////////// SCREENS ////////////////////

void ui_init(void)
{
    lv_disp_t * dispp = lv_disp_get_default();
    lv_theme_t * theme = lv_theme_default_init(dispp, lv_palette_main(LV_PALETTE_BLUE), lv_palette_main(LV_PALETTE_RED),
                                               false, LV_FONT_DEFAULT);
    lv_disp_set_theme(dispp, theme);
    ui_Screen1_screen_init();
    ui_Screen2_screen_init();
    ui____initial_actions0 = lv_obj_create(NULL);
    lv_disp_load_scr(ui_Screen1);
}

kisvegabor, 2024-09-04:
It seems you haven't created and LVGL display. See [here](https://docs.lvgl.io/master/porting/display.html)

closed 2024-09-27 by lvgl-bot