lvgl/lvgl #6784: Task dependency

cosmindanielradu19, 2024-09-03:
### Introduce the problem

As it was stated in [perf(nxp): v9.1.0 upstream vglite optimization](https://github.com/lvgl/lvgl/pull/6646), this patch lowers the CPU usage.

This change shown that there is a misbehaving of lv_layer and it should be analyzed and fixed in LVGL 9.3 as it was stated in [this comment](https://github.com/lvgl/lvgl/pull/6646#discussion_r1718069520)

### Proposal

[task_dependency.patch](https://github.com/user-attachments/files/16847553/task_dependency.patch)

kisvegabor, 2024-09-04:
Thanks for opening this issue! I'll immerse in it again at the latest next week.

kisvegabor, 2024-09-09:
I found it! :tada: See my long comment in the code in #6825

cosmindanielradu19, 2024-09-09:
Thank you for the PR 👏
We will look into it during the next days

closed 2024-09-17 by kisvegabor

nicusorcitu, 2024-09-17:
We will test it soon and see how it performs.

nicusorcitu, 2024-09-17:
@cosmindanielradu19 please have it on TODO list.

cosmindanielradu19, 2024-09-19:
> @cosmindanielradu19 please have it on TODO list.

I have tested the PR, and everything works great. The performance is the expected one.

nicusorcitu, 2024-09-19:
We will cherry-pick this commit into our 9.2 branch.
@kisvegabor what is the plan for lvgl 9.3? Or will it be lvgl 10? Thanks.

kisvegabor, 2024-09-19:
Amazing! :tada: 

We can release v9.2.1 when the the FreeRTOS realted fixes are also merged. I can cherry pick some of the fixes (including this one) to v9.2.1.

There is no reason to release v10 yet as there are no planned breaking changes. We are aiming to release v9.3 in early 2025. See [here](https://github.com/lvgl/lvgl/issues/6763)

nicusorcitu, 2024-09-19:
What is the release date of 9.2.1?
I would like to push also some extra patches that broke the vglite support in 9.2 + some improvements.
Would be nice to have them all included in 9.2.1 too.

kisvegabor, 2024-09-19:
No problem, we can wait a few weeks. What about October 1?

nicusorcitu, 2024-09-19:
Perfect, we will push until Oct 1st. Thanks.

PR #6825 (2024-09-17): perf(draw): skip area independence tests with one draw unit
fixes #6784

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com [... cut]