lvgl/lvgl #6785: When updating the LVGL library from 9.1 to 9.3, the number of FPS decreased.

bladerunner09, 2024-09-03:
### LVGL version

v9.3.0

### What happened?

I tested the demo project in conjunction with FreeRTOS and noticed that in the new version, in the absence of animations, FPS drops to 6-10. If you add looping animation, the FPS rises to 40. As I understand it, this refers to part of the optimization, so as not to waste resources on rendering static objects. But now it is impossible to scroll through file system objects with such an FPS. Can you give advice on how to fix this bug?

Kirbyrawr, 2024-09-03:
This also happens to me. Do you use SDL2?
It went from 30 to 4 fps literally.

cc @kisvegabor

bladerunner09, 2024-09-04:
@Kirbyrawr I don't use SDL. I disabled the graphics engine, but it didn't affect FPS. I accidentally discovered that the looping animation `lv_anim_t` accesses the lvgl timer and the fps increases. 
Try adding something like this to your project:
`lv_anim_t a;
 lv_anim_init(&a);
 lv_anim_set_values(&a, 20, 100);
 lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
 lv_anim_set_exec_cb(&a, scale1_indic1_anim_cb);
 lv_anim_set_var(&a, arc);
 lv_anim_set_duration(&a, 4100);
 lv_anim_set_playback_duration(&a, 2700);
 lv_anim_start(&a);`
If it works, then the source of our problem is common.

Kirbyrawr, 2024-09-04:
Yeah you are right, i also noticed that when you do stuff the fps increase instead of decrease 😅

kisvegabor, 2024-09-04:
Wow, thank for reporting it. It mush be a LVGL draw task scheduling issue of some kind. 

Unfortunately I can't see it on my end. So let's try to narrow it. Please 
- Use SW rendering only
- Set `LV_USE_OS LV_OS_NONE`
- If possible run a profiler (e.g. `perf` on Linux). You can just pause the debugger multiple times. If the FPS is ~4 you should catch where it's spinning

Just to confirm, it's really slow and not just a wrong FPS value is displayed, right?

bladerunner09, 2024-09-04:
@kisvegabor
- `LV_USE_DRAW_SDL` set 0
- `LV_USE_OS` set `LV_OS_NONE`
- I rebuilt the project without FreeRTOS. Unfortunately, pausing the debugger did not help to understand where the program was spinning.
- LVGL version master
Yes, FPS number is absolutely accurate. Where is draw task scheduling located?

kisvegabor, 2024-09-04:
> Where is draw task scheduling located?

Mainly in `lv_draw.c` and `lv_draw_sw.c`. The core element is the `dispatching`.

Just tested again and it's still good on my end. This is my `lv_conf.h` file. Please test with it too.

[lv_conf.h.zip](https://github.com/user-attachments/files/16871176/lv_conf.h.zip)

bladerunner09, 2024-09-04:
@kisvegabor 
The project crashed with your `lv_conf.h` 😅 I use these settings:
[lv_conf.zip](https://github.com/user-attachments/files/16872514/lv_conf.zip)
If you remove the line `lv_anim_start(&a);`:
[image]
If we return the string `lv_anim_start(&a);`:
[image]

I know for sure that the result is not affected by FreeRTOS and graphics engine drivers.

kisvegabor, 2024-09-04:
How does it look like with `LV_USE_DRAW_2DGE 0`?

kisvegabor, 2024-09-04:
Are you sure that you are on the latest `master`?
1. It's seems like you are using Nuvoton's LVGL fork which might not be fully up to date
2. The include in `lv_conf.h` shouldn't work like this. See [here](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L21-L23)

kisvegabor, 2024-09-04:
Please check if #6798 resolves the issue.

bladerunner09, 2024-09-05:
There are no more questions, if set `#define OS_TYPE 3 /* 0:Win32, 1:uITRON4.0, 2:uC/OS-II, 3:FreeRTOS, 4:CMSIS-RTOS */` in the FATFS driver, then everything works correctly. Thank you again.

kisvegabor, 2024-09-05:
Awesome!

PR #6798 (2024-09-10): fix(perf): perf monitor FPS
The logic for pausing the timer was accidentally inverted. It should not pause the timer if the perf monitor is enabled. Always try to achieve target FPS while perf monitor is enabled.

Should fix #6785

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvg [... cut]