lvgl/lvgl #6795: Update the values in components which reflects on all screens

vishvanathdoijad, 2024-09-04:
### Introduce the problem

"I have prepared a couple of screens using LVGL 8.3. All screens include a status bar that displays time, signal strength, and battery status. Now, I'm stuck on how to update these fields with real data or dummy varying data."

### Proposal

For other screens, To update the label field, I used below code in main...! Is it correct?

char buff[5];
int start = 0;
int end = 5;
int step = 1;
int current = 0;
int count = 0;

static void process_gui_msg_q( lv_timer_t *timer ) {
    	if (count <= 1)
    	{
    		count++;
    	}
    	else {
    		count = 0;
    		current += step;
    		if (current == end)
    			current = start;
    	}
	lv_label_set_text_fmt(ui_Label37,"%d mA",current+1);       // dummy data for screen 2
	lv_label_set_text_fmt(ui_Label42,"%d V",current);              // dummy data for screen 3
	lv_label_set_text_fmt(ui_Label46,"%d mA",current+2);
	lv_label_set_text_fmt(ui_Label50,"%2d:%2d:%2d", current, current, current);
	lv_label_set_text_fmt(ui_Label44,"%d V",current+1);
       lv_label_set_text_fmt(ui_Label48,"%d mA",current+3);           // dummy data for screen 7
}

int main(){
lv_init();
.
.
All other inits;
.
.
ui_init();
lv_timer_create((lv_timer_cb_t)process_gui_msg_q, 100, NULL);    // Now its works, But is it correct practice?
while(1)
{
    	lv_task_handler();
        usleep(5000);
}
}
[image]

vishvanathdoijad, 2024-09-05:
"Yes, I have one doubt.
I can update the values in the labels located in the screens/ui_screen.c file, but I can't update the label values located in the components/ui_components.c file."
I have added #include ui_components.h in main.c but getting the error "lv_obj_t obj" undeclared.

[image]

liamHowatt, 2024-09-09:
Is that code generated by SquareLine Studio? Is there a standard way of changing things on events in SquareLine?

If ui_components.h is intended to be included to access the objects, it's strange you're getting that error. Could you send the full "lv_obj_t obj" undeclared error?

vishvanathdoijad, 2024-09-11:
Yes, Code generated by SquareLine Studio.
Yes I am getting "lv_obj_t obj" undeclared error.

vishvanathdoijad, 2024-09-12:
Description	Resource	Path	Location	Type
'cui_LabelTime' undeclared (first use in this function); did you mean 'ui_Label182'?	main.c	/Test_8_3/src	line 113	C/C++ Problem

liamHowatt, 2024-09-17:
Sorry for the delay. Is `cui_LabelTime` in ui_components.c? Is it in also declared in ui_components.h? If it's not in any header files, it cannot be accessed by name from another file, unfortunately.

vishvanathdoijad, 2024-09-17:
1. Yes, "cui_LabelTime" is declared locally in ui_components.c (I make it global declaration)
2. "cui_LabelTime" is not declared in ui_components.h, (I declared it in ui.h as a extern lv_obj_t * cui_LabelTime; like other extern declarations)
 
So I just update "ui_components.c" & "ui.h", Is it correct approach...?

kisvegabor, 2024-09-17:
cc @leidto

leidto, 2024-09-18:
SLS relevant questions like this are answered in the forum. Maybe the solution will be useful for someone else.
https://forum.squareline.io/

closed 2024-09-18 by vishvanathdoijad

vishvanathdoijad, 2024-10-04:
Here is the solution for updating the label in components.

 lv_obj_t * child = ui_comp_get_child(ui_ContainerStatusBar17, UI_COMP_CONTAINERSTATUSBAR_LABELTIME); // When label is in the component
lv_label_set_text_fmt(child,"%02d:%02d:%02d", (int)hr, (int)min, (int)sec);

ui_ContainerStatusBar17 is declared in ui.c & user_screen1.c file
UI_COMP_CONTAINERSTATUSBAR_LABELTIME is #define in ui_comp_statusbar.c & .h file

vishvanathdoijad, 2024-10-04:
> Here is the solution for updating the label in components.
>
> lv_obj_t * child = ui_comp_get_child(ui_ContainerStatusBar17, UI_COMP_CONTAINERSTATUSBAR_LABELTIME); // When label is in the component [...]

@kisvegabor

kisvegabor, 2024-10-14:
Sorry, but I really can't comment on it. As @leidto mentioned, please ask SLS related questions in SquareLine's Forum.