lvgl/lvgl #6797: Kconfig symbol LV_STDLIB_BUILTIN

faxe1008, 2024-09-04:
### LVGL version

v9.2.0

### What happened?

Getting a warning when building:
```
 warning: undefined symbol LV_STDLIB_BUILTIN:
```
The symbol is not present in `Kconfig`. Commit that introduced the regression:
https://github.com/lvgl/lvgl/commit/87c57e3843bdbb14c89318cd4d998675c011fbfb

Maybe a CI step which ensures that the Kconfig is complete would be good?

kisvegabor, 2024-09-04:
This line really doesn't make sense this way:
```
depends on LV_STDLIB_BUILTIN && LV_USE_SYSMON
```

In C it should be
```
LV_USE_STDLIB_MALLOC == LV_STDLIB_BUILTIN
```

Would it work in Kconfig?
```
depends on LV_USE_BUILTIN_MALLOC && LV_USE_SYSMON
```

I wonder how it worked in other systems like NuttX and ESP-IDF. 
cc @XuNeo @espzav

> Maybe a CI step which ensures that the Kconfig is complete would be good?

That would be very nice but I have no idea how to check if Kconfig is complete. Do you have any ideas?

XuNeo, 2024-09-05:
> depends on LV_USE_BUILTIN_MALLOC && LV_USE_SYSMON

I think this will work.

> That would be very nice but I have no idea how to check if Kconfig is complete. Do you have any ideas?

We have discussed this internally several times, and thinking that add NuttX build(the simplest one that enabled lvgl) in lvgl's CI would be great.

kisvegabor, 2024-09-05:
> We have discussed this internally several times, and thinking that add NuttX build(the simplest one that enabled lvgl) in lvgl's CI would be great.

Ok for me.  I mentioned in https://github.com/lvgl/lvgl/issues/5396#issuecomment-2331287028

espzav, 2024-09-05:
> depends on LV_USE_BUILTIN_MALLOC && LV_USE_SYSMON

I think, this should work in ESP-IDF Kconfig

kisvegabor, 2024-09-09:
Can some open PR who can also test it?

liamHowatt, 2024-09-10:
@XuNeo I am happy to open a PR for it unless you have a special interest in doing it.

XuNeo, 2024-09-11:
> @XuNeo I am happy to open a PR for it unless you have a special interest in doing it.

Thank you Liam, @liamHowatt . Please help me out.

liamHowatt, 2024-09-12:
> Can some open PR who can also test it?

Sorry, I asked XuNeo because I thought you were asking for a NuttX CI workflow but I think you were just asking to fix the warning.

I was unable to reproduce the `LV_STDLIB_BUILTIN` warning with ESP IDF :slightly_frowning_face:

faxe1008, 2024-09-14:
While fiddeling with Kconfiglib prototyping for a CI verification I found more kconfig warnings:
`warning: the int symbol LV_DRAW_DMA2D_HAL_INCLUDE (defined at Kconfig:447) has a non-int default stm32h7xx_hal.h (undefined)
warning: the int symbol LV_USE_DRAW_DMA2D_INTERRUPT (defined at Kconfig:452) has a non-int default n (undefined)`

liamHowatt, 2024-09-17:
This would be an amazing solution for catching general kconfig issues before they go into `master`. I've been encountering those DMA2D warnings, and they're my own fault :stuck_out_tongue_closed_eyes:

I think a CI PR using this snippet would be great, if you feel like opening it. You can create a YML file in `.github/workflows` similar to the other checkers that call a python script, like check_conf.yml.

One question: is it possible to make this checker sensitive to your original problem, the `LV_STDLIB_BUILTIN` undefined symbol?

kisvegabor, 2024-09-17:
Amazing! Although it doesn't detect the sync issues between lv_conf.h and Kconfig (which is also a common problem) but at least will know that Kconfig is valid.

faxe1008, 2024-09-18:
@kisvegabor added the CI step to the PR: https://github.com/lvgl/lvgl/pull/6851

with a bit more love the script can be expanded to also catch discrepancies between lv_conf.h and Kconfig. The lib offers to access all of the Kconfig symbols and deps in a structured manner.

PR #6851 (2024-09-20): fix(Kconfig): Fix non existant LV_STDLIB_BUILTIN
LV_USE_MEM_MONITOR depends on LV_STDLIB_BUILTIN which does not exist in the kconfig, therefore generating warnings.

Resolves #6797.

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](htt [... cut]
closed 2024-09-20 by XuNeo