lvgl/lvgl #6804: [Question] How to complete disable (i.e. not include at compile) all functionality related to input/touch? Basically output only

teaalltr, 2024-09-05:
### Introduce the problem

Hi, I don't need all the input-related functionality. I have a touchscreen-enabled 5 inches display but I don't need the input-related code to be included in the build. Is there some flag or define to completely exclude input stuff from being compiled and included in build?
Thank you

kisvegabor, 2024-09-12:
There is no such a config option now, however it seems it could save some memory, however it would also increase complexity and reduce readability.

Is the flash usage your concern ?

teaalltr, 2024-09-12:
@kisvegabor Hi, thanks for your reply. Yes, I'd need to reduce the flash footprint to a minimum

kisvegabor, 2024-09-19:
I will keep this in mind and if a similar request comes from other people too, we can prioritizing it. Sorry that we can't jump into it now. 

However, this change is not complicated, just takes a longer time to properly disable all indev related code. If you would like jump into it please open PR and I can help if you stuck somewhere. 

We are also thinking about creating a very simple version of LVGL. Just basic rectangle fill, image blending and texts. Could something like this be enough for you? If not, what features would you still need in a highly resource constrained environment?

teaalltr, 2024-09-19:
@kisvegabor thanks, a simpler version of LVGL would be welcome, for some of our products we use micros with very small flash and just draw some simple, grid/rectangles based layout with numbers and labels on the cells. We still need i18n features like Chinese fonts etc but I guess this could be shrunk too. Text colors and possibility to set different sizes are also needed for us. Input for us is not needed but you could also include it with option to disable it, and maybe just a simple touch events handling would be enough (maybe with single tap/click/point support). We'd also need to have multiple screens for different fw functions, settings etc. All this would be enough for us. Regarding the touch disablement, thanks for opening the possibility!

kisvegabor, 2024-09-30:
Thank you for the input! I'll keep these in mind. Probably we will start working on it early next year.

closed 2024-10-24 by lvgl-bot