lvgl/lvgl #6805: A reflection on how to access inner fields of some of the widgets struct

lorenzo-arena, 2024-09-05:
### Introduce the problem

Hi everyone!

While transitioning from v9.1 to v9.2 in my project I had some troubles due to the refactoring which took place in 1d14386b99cdff731cc3367d1d8b5e96ffb8fcea. While this was a neat refactor and I always like the idea of having has much encapsulation as possible, in some of my widgets I had to access private fields of LVGL base widgets, like animimage and labels. This can be solved by just adding the right private include, like `#include <lv_animimage_private.h>`, but I'd like to be able to follow up on that encapsulation effort and reach a cleaner solution.

As an example, going again to the animimage issue, I have a widget where I want to override the values used for the animation. When setting the animimage source the values are automatically set in the animation contained inside the animimage, going from `0` to the number of frames, and I'd like to go instead from a specific frame to `0`. This was previously done by accessing the internal `anim` field of the animimage widget.

### Proposal

I basically have 3 proposals (I could also contribute to the last 2 options):
- this could be accepted as is, and for more advanced use cases having to include the "private" header files could be considered "fine"
- we could add some getter methods for the inner fields of a widget (referring again to the animimage example, something like `lv_anim_t * lv_animimg_get_animation(lv_obj_t * img);`)
- we could add more specific methods to extend the configurability of the widget (for the animimage example, I would add something like `void lv_animimg_set_values(lv_obj_t * img, int32_t start, int32_t end);`)

kisvegabor, 2024-09-11:
Thank you for bringing it up. The general idea is to use the `private` headers only for advanced use cases. In 99+% of the cases the public API should be enough.

However, if the public API can't cover the use cases we should really extended it. In this case I would add `lv_animimg_get_animation` as it provides the  highest level of control and flexibility.

closed 2024-09-11 by lorenzo-arena

kisvegabor, 2024-09-11:
Fantastic, thank you!