lvgl/lvgl #6811: lv_obj_set_style_size(chart, 0, 0, LV_PART_INDICATOR); not working fine

ataweg, 2024-09-06:
### LVGL version

master

### What happened?

In my application, with the head revision of LVGL, only individual points were displayed in a chart instead of lines (lv_obj_set_style_size(chart, 0, 0, LV_PART_INDICATOR)). To rule out a problem on my side, I tried lv_example_chart_1.c with the help of lv_port_pc_visual_studio.

In order to find out from which commit the incorrect behavior occurs, I updated lvgl bit by bit. It turns out that the incorrect behavior occurs between commit a8c8275b560fa74b8bdd72d1d180c98d22a33fc8 (good) and the following 1d14386b99cdff731cc3367d1d8b5e96ffb8fcea (fail) from August 2nd.

>git checkout a8c8275b560fa74b8bdd72d1d180c98d22a33fc8
HEAD is now at a8c8275b5 feat(os): use recursive mutex by default (#6573)

F:\working\_Temp\lv_port_pc_visual_studio\LvglPlatform\lvgl\examples\widgets\chart\lv_example_chart_1.c

 [image]
 
Added “    lv_obj_set_style_size(chart, 0, 0, LV_PART_INDICATOR);” after line 14

[image]

>git checkout 1d14386b99cdff731cc3367d1d8b5e96ffb8fcea
Previous HEAD position was a8c8275b5 feat(os): use recursive mutex by default (#6573)
HEAD is now at 1d14386b9 refactor(API): don't expose private symbols in lvgl.h. phase-out "_lv*" names (#6068)

 [image]

Added “    lv_obj_set_style_size(chart, 0, 0, LV_PART_INDICATOR);” after line 14

 [image]

kisvegabor, 2024-09-12:
Hi,

Could you try with the latest `master`. It work well on my end.

ataweg, 2024-09-12:
I still have the problem. Attached are the modified files so that the latest version (7f57f37560c29ae6007b828fb93a61764f483e1f) of lvgl works.

for for ...\lv_port_pc_visual_studio
[lvgl_issues_6811.zip](https://github.com/user-attachments/files/16978532/lvgl_issues_6811.zip)

for ...\lv_port_pc_visual_studio\LvglPlatform\lvgl
[lvgl_issues_6811-lvgl.zip](https://github.com/user-attachments/files/16978535/lvgl_issues_6811-lvgl.zip)

ataweg, 2024-09-15:
When setting in lv_example_chart_1() 
````lv_obj_set_style_size(chart, 0, 0, LV_PART_INDICATOR);````

I dug a little deeper into lvgl and found  something not optimal in draw_series_line():
```
    if(LV_MIN(point_w, point_h) > line_dsc.width / 2) line_dsc.raw_end = 1;
    if(line_dsc.width == 1) line_dsc.raw_end = 1;
```
when I replave above with 
````line_dsc.raw_end = 1;````

I got 
[image]
Not good, but I have lines.

ataweg, 2024-09-16:
Running the example on an ESP32 is ok. 
Is it a grafic driver issue?

kisvegabor, 2024-09-25:
I can't imagine how can it be a Windows driver issue. Dump question. Are you sure that you are using the same LVGL version. Using `git describe` is a giid way to check it

ataweg, 2024-09-25:
I mean the graphical driver from lv_port_pc_visual_studio.

>git describe
v9.2.0-80-g48cbe7bd7
>git log -1
commit 48cbe7bd7653af00e14c28c339f27c5a4c408159 (HEAD -> master, origin/master, origin/HEAD)
Author: Gabor Kiss-Vamosi <kisvegabor@gmail.com>
Date:   Wed Sep 25 11:39:51 2024 +0200
    perf(benchmark): use XRGB8888 image withn 32bit color depth (#6914)

It is noticeable that with the major conversion to private headers (https://github.com/lvgl/lvgl/commit/a8c8275b560fa74b8bdd72d1d180c98d22a33fc8 -->  https://github.com/lvgl/lvgl/commit/1d14386b99cdff731cc3367d1d8b5e96ffb8fcea ) this problem has occurred.

kisvegabor, 2024-10-03:
It was really a large update but I wonder how can ir work on my end with SDL. 

cc @MouriNaruto

MouriNaruto, 2024-10-03:
Windows driver should work as the same as SDL driver if SDL driver also uses the LVGL Software render.

Kenji Mouri

kissa96, 2024-10-18:
Hi, how does this issue hold?

ataweg, 2024-10-18:
Hi, the issue is still there with lv_port_pc_visual_studio commit  15ea92a227970cb766527071cc7737dfb5cf5e58 .  (The latest version has issues with Mile.Project.Windows).

Were you able to test it under the same conditions?
I use 
-   Microsoft Windows [Version 10.0.22631.4317]
 - Microsoft Visual Studio Community 2022 (64-Bit) - Current Version 17.11.5

I would be very surprised if this is a specific problem with my setup.

Mair, 2024-10-27:
As reported by someone doing the LVGL, this is also affecting lines drawn in scales.
[image]

kissa96, 2024-10-28:
Visual Studio has a known issue regarding line drawing.

kisvegabor, 2024-10-28:
@MouriNaruto found the root of the problem and I've just opened a PR to fix it: https://github.com/lvgl/lvgl/pull/7171

Please confirm if its working well.

ataweg, 2024-10-29:
Now it works as it should.
[image]

kisvegabor, 2024-10-30:
Amazing, thank you for the feedback!

closed 2024-10-30 by kisvegabor