lvgl/lvgl #6812: type 'lv_draw_task_t' incomplete

Harry863, 2024-09-06:
### LVGL version

V9.2.0

### What happened?

Hello,

I use lv_draw_dsc_base_t to draw my table.

After updating V9.1.0 to V9.2.0 I got the following error during the compilation.

error: invalid use of incomplete type 'lv_draw_task_t' {aka 'struct lv_draw_task_t'}

In this line.

lv_draw_dsc_base_t * base_dsc = (lv_draw_dsc_base_t*)draw_task->draw_dsc;

Thank you
Harry

### How to reproduce?

lv_draw_task_t * draw_task = lv_event_get_draw_task(e);
lv_draw_dsc_base_t * base_dsc = (lv_draw_dsc_base_t*)draw_task->draw_dsc;

lhdjply, 2024-09-07:
In LVGL 9.2, some less frequently used structures in the application layer have been made private. If you want to resolve this issue, there are three methods:
1. use the function lv_draw_task_get_draw_dsc as an alternative
2. enable `LV_USE_PRIVATE_API` in lv_conf.h
3. include the corresponding header file `lv_draw_private.h`

Harry863, 2024-09-07:
Perfect! l
     v_draw_task_get_draw_dsc solvs the problem.

----------------------------------------------------------------------------------------
But it seems like V9.2.0 needs much more RAM1.

Here the memory usage with V9.1
   teensy_size: Memory Usage on Teensy 4.0:
   teensy_size:   FLASH: code:356764, data:57528, headers:8616   free for files:1608708
   teensy_size:    RAM1: variables:58656, code:350776, padding:9672   free for local variables:105184
   teensy_size:    RAM2: variables:12416  free for malloc/new:511872

And here the memory usage with V9.2 
   teensy_size: Memory Usage on Teensy 4.0:
   teensy_size:   FLASH: code:402716, data:57528, headers:8744   free for files:1562628
   teensy_size:    RAM1: variables:58720, code:396728, padding:29256   free for local variables:39584
   teensy_size:    RAM2: variables:12416  free for malloc/new:511872

In lv_conf.h the following options are activated
   //#define LV_USE_STDLIB_MALLOC    LV_STDLIB_BUILTIN
   //#define LV_USE_STDLIB_STRING    LV_STDLIB_BUILTIN
   //#define LV_USE_STDLIB_SPRINTF   LV_STDLIB_BUILTIN
   #define LV_USE_STDLIB_MALLOC    LV_STDLIB_CLIB
   #define LV_USE_STDLIB_STRING    LV_STDLIB_CLIB
   #define LV_USE_STDLIB_SPRINTF   LV_STDLIB_CLIB

kisvegabor, 2024-09-12:
40kB extra RAM is not intentional for sure.  What does `code` mean in case of RAM?

Harry863, 2024-09-12:
Code:                           Size of the reserved (blocked) memory.
Padding:                     Memory which is free in the last not full block.
Free for local variable: Free memory for additional blocks. 

Teensy reserves memory block wise. Each block is 32KB.

kisvegabor, 2024-09-19:
I see, thank you! 

Can you run a tool like `nm` (Linux) to see what uses that much memory?
This my cheat sheet to it:
```
functions
  nm -S --size-sort path_to_your.elf -t d | grep -P "^[0-9]+ [0-9]+ [tT]" | tail -20

const data:
  nm -S --size-sort path_to_your.elf -t d | grep -P "^[0-9]+ [0-9]+ [rR]" | tail -20

RAM usage
  nm -S --size-sort path_to_your.elf -t d | grep -P "^[0-9]+ [0-9]+ [bBdD]" | tail -20
```

You can also send you elf file so that I can take a look.

Harry863, 2024-09-21:
Hallo Gabor,

To achieve the smallest possible elf file, I have separated all stuff regarding LVGL in a separate projects.
And compiled it with LVGL 9.1 and 9.2 versions.
The big difference in RAM usage is not there.

Result with LVGL 9.1
teensy_size:   FLASH: code:368748, data:112072, headers:9132   free for files:7636512
teensy_size:    RAM1: variables:69476, code:226516, padding:2860   free for local variables:225436
teensy_size:    RAM2: variables:12448  free for malloc/new:511840

Result with LVGL 9.2
teensy_size:   FLASH: code:369132, data:112072, headers:8748   free for files:7636512
teensy_size:    RAM1: variables:69540, code:226900, padding:2476   free for local variables:225372
teensy_size:    RAM2: variables:12448  free for malloc/new:511840

So I have changed my current project to the version 9.2 too

If you still would like to have the elf files, then please give me a sign.

Thank you. 
Harry

kisvegabor, 2024-10-03:
Huh, I'm happy that it's not coming from LVGL :slightly_smiling_face: 

`nm` still might help finding what happened in your project.

kissa96, 2024-10-18:
Is this still relevant? If not, can you please close the issue?

kisvegabor, 2024-11-06:
Closing due to inactivity.

closed 2024-11-06 by kisvegabor