lvgl/lvgl #6814: How to input "^" with lv_keyboard?

lanistor, 2024-09-07:
### Introduce the problem

Is there a built-in keyboard supporting "^"? Or must redefine the buttons of keyboard?

liamHowatt, 2024-09-09:
Do you know about [lv_keyboard_set_map](https://docs.lvgl.io/master/API/widgets/keyboard/lv_keyboard.html#_CPPv420lv_keyboard_set_modeP8lv_obj_t18lv_keyboard_mode_t)?

kisvegabor, 2024-09-10:
It's really missing form the default layout, however I can't see where can we fit it. :slightly_frowning_face:

liamHowatt, 2024-09-11:
Maybe we could add another `lv_keyboard_mode_t` with more symbols?

kisvegabor, 2024-09-11:
Are there more missing symbols? I wouldn't add a new page for a single symbol.

lanistor, 2024-09-27:
> Are there more missing symbols? I wouldn't add a new page for a single symbol.

I found these missing symbols : `~`, `^`, `\`` `|`.

closed 2024-10-24 by lvgl-bot