lvgl/lvgl #6815: LV_CONF_PATH got expanded (on lv_conf_internal.h) causes error when path contain "linux" (Linux env.)

fb87, 2024-09-07:
### LVGL version

v9.2.0

### What happened?

I got funny case where the path to `lv_conf.h` contains `linux` string which surrounded by escape characters(`/work/lvgl-linux/lv_conf.h`), and C pre-processor just treated the path as macro and then expanded the `linux` to `1` which resulted in an error due to incorrect path `/work/lvgl-1/lv_conf.h` generated. And I've just got to know that (at least in Linux) has `linux` defined elsewhere (not by me :smile:) which leads to this issue.

### How to reproduce?

Bellow is the codesnip copied from `lv_conf_internal.h` where `LC_CONF_PATH` got expanded. Quite easy to reproduce, just set `LV_CONF_PATH` to a string which contain `linux` string surrounded by escape characters e.g `-linux-` then do the compilation.

```bash
root@2ffd8fe5df8f:~# cat hello.c 
#define LV_CONF_PATH lvgl-linux/cfg/lv_conf.h

#if linux
#warning "Linux defined"
#endif

#ifdef LV_CONF_PATH                           /*If there is a path defined for lv_conf.h use it*/
    #define __LV_TO_STR_AUX(x) #x
    #define __LV_TO_STR(x) __LV_TO_STR_AUX(x)
    #include __LV_TO_STR(LV_CONF_PATH)
    #undef __LV_TO_STR_AUX
    #undef __LV_TO_STR
#elif defined(LV_CONF_INCLUDE_SIMPLE)         /*Or simply include lv_conf.h is enabled*/
    #include "lv_conf.h"
#else
    #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
#endif
root@2ffd8fe5df8f:~# gcc -E hello.c 
# 0 "hello.c"
# 0 "<built-in>"
# 0 "<command-line>"
# 1 "/usr/include/stdc-predef.h" 1 3 4
# 0 "<command-line>" 2
# 1 "hello.c"
hello.c:4:2: warning: #warning "Linux defined" [-Wcpp]
    4 | #warning "Linux defined"
      |  ^~~~~~~
hello.c:10:38: fatal error: lvgl-1/cfg/lv_conf.h: No such file or directory
   10 |     #include __LV_TO_STR(LV_CONF_PATH)
      |                                      ^
compilation terminated.
root@2ffd8fe5df8f:~# gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

kisvegabor, 2024-09-12:
Wow, very interesting situation. I wonder what can we do with it. Could defining `LV_CONF_PATH` as a string work?
```c
#define LV_CONF_PATH "lvgl-linux/cfg/lv_conf.h"
#include LV_CONF_PATH
``` 

Actually I don't know why it's not a string.

fb87, 2024-09-12:
No, the `#define __LV_TO_STR_AUX(x) #x` will add double quotes around the string one more time and then an other compile error introduced.

kisvegabor, 2024-09-19:
Honestly, I don't know why it needs to be defined without quotes. It should be way simpler to have `#define LV_CONF_PATH "lvgl-linux/cfg/lv_conf.h"`. 

@XuNeo What do you think, can we introduce such a breaking change somehow? In summary change:
```c
#define LV_CONF_PATH path/to/lv_conf.h
```
to
```c
#define LV_CONF_PATH "path/to/lv_conf.h"
```

XuNeo, 2024-09-20:
Hi Gabor,

I have never used customized path to lv_conf.h. This break change seems easy to fix. I think we can introduce it and make a good comment && documentation.

```patch
diff --git a/src/lv_conf_internal.h b/src/lv_conf_internal.h
index 0c9006b6e..ddacbc5cd 100644
--- a/src/lv_conf_internal.h
+++ b/src/lv_conf_internal.h
@@ -49,11 +49,8 @@
 /* If lv_conf.h is not skipped, include it. */
 #if !defined(LV_CONF_SKIP) || defined(LV_CONF_PATH)
     #ifdef LV_CONF_PATH                           /* If there is a path defined for lv_conf.h, use it */
-        #define __LV_TO_STR_AUX(x) #x
-        #define __LV_TO_STR(x) __LV_TO_STR_AUX(x)
-        #include __LV_TO_STR(LV_CONF_PATH)
-        #undef __LV_TO_STR_AUX
-        #undef __LV_TO_STR
+        /* Note: make sure your path is a string, E.g. #define LV_CONF_PATH "lv_con.h" */
+        #include LV_CONF_PATH
     #elif defined(LV_CONF_INCLUDE_SIMPLE)         /* Or simply include lv_conf.h is enabled. */
         #include "lv_conf.h"
     #else
```

kisvegabor, 2024-09-20:
I agree! We just need to correctly document these kind of niche breaking changes. 

A new colleague will join us soon, so I leave it open for him as a good getting started issue. :slightly_smiling_face: 
cc @kissa96

closed 2024-11-27 by FASTSHIFT

zahical, 2025-02-24:
First, thanks, FASTSHIFT, for fixing it. 

Since the fix is not yet part of any official release (the latest, as of time of writing, v9.2.2 doesn't have it)
I just want to share a possible workaround for those using lvgl via its CMake integration.

1. Do **not** define LV_CONF_PATH
2. Instead, set the the undocumented LV_CONF_DIR to the **folder** where your custom lv_conf.h resides.
3. Ensure LV_CONF_INCLUDE_SIMPLE is enabled (it's on by default)

For example (assuming lv_conf.h is next to CMakeLists.txt):
```
set(LV_CONF_DIR "${CMAKE_CURRENT_SOURCE_DIR}")

include(FetchContent)
FetchContent_Declare(
    lvgl
    GIT_REPOSITORY https://github.com/lvgl/lvgl.git
    GIT_TAG v9.2.2
)
FetchContent_MakeAvailable(lvgl)
```

The approach works because lvgl's CMake script unconditionally adds LV_CONF_DIR as an include directory for the lvgl target and this way the simple `#include "lv_conf.h"` (enabled by the LV_CONF_INCLUDE_SIMPLE option) is able to locate it.

PR #7335 (2024-11-27): fix(lv_conf): get rid of the LV_CONF path building macros
A clear and concise description of what the bug or new feature is.

The macros for building LV_CONF_PATH symbol causes problems on platforms like Linux, which does not generate it correctly, so get rid of them, and instruct the user to define the symbol as a string when configuring LVGL.

Fixes: #6815 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/test [... cut]