lvgl/lvgl #6818: Enhance dropdown lists for smaller displays

zalexzperez, 2024-09-07:
### Introduce the problem

Currently, dropdowns have four opening directions.
However, in small displays, dropdown lists have too little space and it's made worse by the way lists open up.
Take the next image as an example of the problem. It's a small 1.47" 172x320 display, so your monitor will likely make it look way bigger than it really is:

[image]
When the list shows up there's only two options displayed even though they're tiny-looking:
[image]

At the moment, one can improve the available list space by increasing parent panel size and moving the dropdown widget close to a border like this:
[image]

### Proposal

It would be nice to have an option to make the list take up as much space as required by the number of options, even if it results in the dropdown button getting covered. 
If all the options don't fit on the screen, then, a scrollbar would be shown.

Another idea, perhaps easier to implement, is to simply allow dropdowns to look like a scrollable popup list window like those seen in smartphone UIs:

[image]

Or even with some dropdown header title:
[image]

kisvegabor, 2024-09-12:
Hi,

Interesting idea! I've tried out a few things:
- Set the `min-height` of  the list. It could work, but it's hard to center the list if there are only a few option and the list height is smaller
- Add `LV_DIR_CENTER`. Also could work, but shall it center on the screen or on the drop-down?

After thinking about it more it seems to me it's should be an entirely different widget, and not the drop down. Even on the smartphones it looks very different and has nothing to do with an ordinary drop-down.

kisvegabor, 2024-09-19:
It seems like you don't need to use drop-downs at all. The users can click a simple button, a message box appears where you can create and `lv_list` or just an other set of buttons.

zalexzperez, 2024-09-23:
Considering I'm using a UI editor (EEZ Studio), and the fact I need to display the chosen option all the time like in dropdowns, it would take some effort. I think I'll just increase font size for the list and settle for only viewing two options at a time.