lvgl/lvgl #6819: software rotation is not longer available in 9.x

kdschlosser, 2024-09-08:
### LVGL version

master branch

### What happened?

in `lv_display_t` there is the field `sw_rotate` which used to control software rotation. This field looks to be no longer used by LVGL and it is also inaccessible due to it not being apart of the public API. There is no function to set this parameter even if it did actually do something.

The problem is RGB displays that do not have a hardware rotation feature. Currently LVGL is able to work properly with displays that support hardware rotation but it doesn't work with displays that don't.

I believe it would be a simple fix because LVGL does provide rotation of `lv_obj_t` a function that would set `sw_rotate` and then when calling `lv_display_set_rotation` that function would check `sw_rotate` and it would rotate the layers that are attached to a display. Since the layers are `lv_obj_t` structures the `lv_obj_set_style_transform_rotation` can be used to set the rotation of those layers when the `lv_display_set_rotation` function is called and the `sw_rotate` flag is set.

MacDragon, 2024-09-09:
The expectation is currently to handle the rotation in buffer flush currently - tiling cache optimised flush can be pretty efficient at this in SW, and better yet if there's a DMA available that can do it. Doing a seperate rotation step between drawing and sending to FB would be quite slow, it either needs to be done at draw time, or at flush to FB time to be in any way performant.

kdschlosser, 2024-09-09:
the issue with doing it at flush time is the need for another buffer, With an RGB display you can't use the second frame buffer to copy the rotated data to because it needs to be synchronized with the first buffer. when dealing with a higher resolution RGB display that 3rd buffer is going to be very costly memory wise because it needs to be the same size as the display. you can't do an in place modification of the buffer data.

It might be able to be done by using a temporary buffer that is of smaller size but now instead of doing a single copy of data from one buffer to the other you have to do 2 copies one to the temp buffer and then one back to the frame buffer. That is going to slow it down even more. And yes you can use DMA if it is available but it is only able to be done chunks at a time which is going to impact performance quite a bit due to the repeated transactions that have to be made. 

would have been nicer if LVGL would place the pixels correctly when the data is being rendered to the buffer. Otherwise there really is no point to even storing the rotation in LVGL because all it is using it for is to identify what the width and height is. May as well just swap those when the rotation is done and not bother storing the rotation at all. The user can take care of tracking what the rotation is.

kdschlosser, 2024-09-10:
OoOO and just to let you know. DMA transfers are not going to work because the overhead is going to outweigh doing a straight up copy from one buffer to the other. The reason why is the reversal of the pixel order when making the copy. You cannot copy say an entire stride at one shot because the pixel order is going to be backwards. so that means a DMA transfer would have to be started for each pixel and that causes a huge amount of overhead between the ISR for the transfer being done, the amount of memory used for the DMA transfer VS what actually is getting transferred and the iteration still needing to be done over the entire buffer to be able to do it. There is no gains to be had by using DMA for doing the pixel rotation.

MacDragon, 2024-09-10:
That depends on the DMA hardware, the hardware we're using has DMA capable of handling copies with stride/reversal etc.

kisvegabor, 2024-09-17:
I've opened https://github.com/lvgl/lvgl/pull/6866 to remove `sw_rotate`.

In general I believe letting the user do the rotation in the `flush_cb` leaves more room for customization. 

Note that the fixed (90, 180, 270) rotations are way faster than anti-aliased layer rotation with an arbitrary angle.

kdschlosser, 2024-09-17:
@kisvegabor 

> In general I believe letting the user do the rotation in the flush_cb leaves more room for customization.

I fully agree, but... there should be some streamlined convenience functions that the user is able to use to do the rotation. The functions(s) should be able to take an input buffer, an output buffer an area where the input buffer needs to be placed in the output buffer. If it is a single function to handle rotation then it would also need to be passed the rotation value. 

If you look at this code.. 

https://github.com/lvgl/lvgl/blob/87f6e9d691f20703dd47d4d1454c568e552af116/src/drivers/sdl/lv_sdl_window.c#L228

You can see there is use of a temp buffer. This is unneeded and actually wasteful. It is also inefficient to use a temp buffer when handling rotation. This is because the data has to be copied from one buffer to the temp buffer and then the data once again gets copied from the temp buffer into the frame buffer. There are 2 copy operations and 2 iterations over the data when it only needs to be done a single time. Copy the data from the input buffer to the frame buffer at the same time the data is being rotated. 

Here is some example code for doing that...
This code is used in a much larger project and I have made some changes  so you get the general idea. 

```c
 __attribute__((always_inline)) static inline void cpy_pixel_8bpp(uint8_t *to, const uint8_t *from)
{
    *to++ = *from++;
}

__attribute__((always_inline)) static inline void cpy_pixel_16bpp(uint8_t *to, const uint8_t *from)
{
    *to++ = *from++;
    *to++ = *from++;
}

__attribute__((always_inline)) static inline void cpy_pixel_24bpp(uint8_t *to, const uint8_t *from)
{
    *to++ = *from++;
    *to++ = *from++;
    *to++ = *from++;
}

typedef void (*cpy_pixel)(uint8_t *to, const uint8_t *from);

void rgb_tx_color(lv_display_t *disp, lv_area_t *area, void *color)
{
    int x_start = (int)area->x1;
    int y_start  = (int)area->y1;
    int x_end = (int)area->x2;
    int y_end = (int)area->y2;

    bool last_flush = lv_display_flush_is_last(disp);
    size_t color_size = (size_t)((x2 - x1 + 1) * (y2 - y1 + 1) * 2)
    lv_display_rotation_t rotation = lv_display_get_rotation(disp);

    // buf 1 and buf 2 are full frame buffers just render the buffers.
    // In order to have the rotation work with an RGB display a 3rd buffer 
    // would be needed that is the same size. That is a lot of additional 
    // memory use and the copy would be slow because the buffers would reside in
    // spiram.
    if (color == buf_1 || color == buf_2) {
        LV_UNUSED(x_start);
        LV_UNUSED(y_start);
        LV_UNUSED(x_end);
        LV_UNUSED(y_end);
        LV_UNUSED(rotation);
        LV_UNUSED(last_flush);

        esp_lcd_panel_io_tx_color(panel_io, -1, color, color_size);
    } else {
        // I am using a FreeRTOS task that handles copying the data from one 
        // of the frame buffers to the other. I do not want that process to stop
        // LVGL from writing new data to a partial buffer. The copying is done
        // on the second core of the ESP32 
        xSemaphoreTake(flushCopySeph, portMAX_DELAY);
        uint8_t *fb;

        if (cur_fb_index == 0) {
[... 81 lines cut ...]
        default:
            LV_UNUSED(cpy_func);
            LV_UNUSED(offset);
            to = fb + (y_start * h_res + x_start) * bytes_per_pixel;
            for (int y = y_start; y < y_end; y++) {
                memcpy(to, from, copy_bytes_per_line);
                to += bytes_per_line;
                from += copy_bytes_per_line;
            }
            bytes_to_flush = (y_end - y_start) * bytes_per_line;
            flush_ptr = fb + y_start * bytes_per_line;
            break;
        }

        LV_UNUSED(bytes_to_flush);
        LV_UNUSED(flush_ptr);

        if (last_flush) {
            esp_lcd_panel_io_tx_color(panel_io, lcd_cmd, fb, (size_t)(v_res * h_res * bytes_per_pixel));
        }

        lv_display_flush_ready(disp);
    }
}
```

closed 2024-09-18 by kisvegabor

kisvegabor, 2024-09-19:
Please take a look at this PR https://github.com/lvgl/lvgl/pull/6835

It eliminates the the use of temp buffer.

kdschlosser, 2024-09-19:
it's not just sdl that uses the temp buffer. the linux fb driver does as well.

kisvegabor, 2024-09-30:
If the performs better in your benchmarks I'll add it for fbdev too.

PR #6866 (2024-09-18): fix(display): remove the unused sw_rotate field
fixes #6819

Leftover from v8

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py [... cut]