lvgl/lvgl #6820: Jaggy Objects

HenryKnightCoder, 2024-09-08:
### LVGL version

v9.2.0

### What happened?

I plot a black square at (0, 0) of a 1448x1072 display (8bit color depth, i.e. RGB232)

``` c
lv_obj_t * scr = lv_scr_act();
lv_obj_set_style_bg_color(scr, lv_color_white(), LV_PART_MAIN);
lv_obj_t * square = lv_obj_create(scr);
lv_obj_set_size(square, 100, 100);
lv_obj_set_style_border_color(square, lv_color_black(), LV_PART_MAIN);
lv_obj_set_style_border_width(square, 2, LV_PART_MAIN);
lv_obj_set_style_bg_color(square, lv_color_black(), LV_PART_MAIN);
lv_obj_set_pos(square, 0, 0);
```

and inside flush function I print x, y, w, h, and the first 8 pixels of each line:

``` c
void my_flush_cb(lv_display_t* my_drv, const lv_area_t* area, lv_color_t* color_p)
{
  vTaskDelay(pdMS_TO_TICKS(10));
  uint32_t w = area->x2 - area->x1 + 1;
  uint32_t h = area->y2 - area->y1 + 1;
  LoadImageInfo lvgl_img =
  {
      .ImageData = (uint8_t*)color_p,
      .X = area->x1,
      .Y = area->y1,
      .Width = w,
      .Height = h,
  };
  printf("\n%d %d %d %d\n", lvgl_img.X, lvgl_img.Y, lvgl_img.Width, lvgl_img.Height);
  
  // print the first 8 pixels of each line
  for(uint8_t i = 0; i < 5; i++){
      printf("%d %d %d %d %d %d %d %d\n", lvgl_img.ImageData[i * lvgl_img.Width], lvgl_img.ImageData[i * lvgl_img.Width + 1], lvgl_img.ImageData[i * lvgl_img.Width + 2], lvgl_img.ImageData[i * lvgl_img.Width + 3], lvgl_img.ImageData[i * lvgl_img.Width + 4], lvgl_img.ImageData[i
  * lvgl_img.Width + 5], lvgl_img.ImageData[i * lvgl_img.Width + 6], lvgl_img.ImageData[i * lvgl_img.Width + 7]);
  }

  \\ rest of the code, sending pixels to the display
}
```

I expect to have all 0x00 for the first few pixels of the rows but I see an unexpected behavior causing jaggy plot:

``` bash
0 0 1448 5
255 255 255 255 255 255 174 94
255 255 255 255 190 30 0 0
255 255 255 126 0 0 3 0
255 255 126 0 0 0 0 0
255 190 0 0 0 0 0 0

0 5 1448 5
255 30 0 0 0 0 0 0
174 0 3 0 0 0 0 0
94 0 0 0 0 0 0 0
30 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0

0 10 1448 5
0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0
```
I am checking lvgl pixels before sending them to my display driver so the issue should be on LVGL side.

p/s: I have tested the display driver by creating lvgl_img by hand and everything is working perfectly.

[image]

kisvegabor, 2024-09-12:
Hi,

It's strange that the skew part resets in every ~5 line.

Do you use PARTIAL buffering mode? If so, what is the buffer size?

HenryKnightCoder, 2024-09-20:
Hi Kisvegabor,

Yes, the size of buffer is 5x of the width of display. Here I am refreshing the whole display.

additionally, if I only change part of the display (like updating a font in the middle of the display), this jaggedness is way way worse.

FirmwareFury, 2024-10-14:
NP. Thank you for following this up. The rest is just calling the display driver function which sends the pixels in a given rectangle to the display:

``` c
void my_flush_cb(lv_display_t* my_drv, const lv_area_t* area, lv_color_t* color_p)
{
  vTaskDelay(pdMS_TO_TICKS(10));
  uint32_t w = area->x2 - area->x1 + 1;
  uint32_t h = area->y2 - area->y1 + 1;
  LoadImageInfo lvgl_img =
  {
      .ImageData = (uint8_t*)color_p,
      .X = area->x1,
      .Y = area->y1,
      .Width = w,
      .Height = h,
  };
  printf("\n%d %d %d %d\n", lvgl_img.X, lvgl_img.Y, lvgl_img.Width, lvgl_img.Height);
  
  // print the first 8 pixels of each line
  for(uint8_t i = 0; i < 5; i++){
      printf("%d %d %d %d %d %d %d %d\n", lvgl_img.ImageData[i * lvgl_img.Width], lvgl_img.ImageData[i * lvgl_img.Width + 1], lvgl_img.ImageData[i * lvgl_img.Width + 2], lvgl_img.ImageData[i * lvgl_img.Width + 3], lvgl_img.ImageData[i * lvgl_img.Width + 4], lvgl_img.ImageData[i
  * lvgl_img.Width + 5], lvgl_img.ImageData[i * lvgl_img.Width + 6], lvgl_img.ImageData[i * lvgl_img.Width + 7]);
  }

  Paint_DrawImage(&lvgl_img);

  lv_disp_flush_ready(eink_drv);
}
```
The `Paint_DrawImage` is tested (and has been used for a long time outside LVGL) and is working as expected (it shows `color_p` pixels on display, starting at `(x1, y1)` with the given size of `(w, h)`.  So if color_p is all zeros, it shows a rectangle with no jaggy edges.

Note that from what `flush` function prints, it is clear that there is a shift in each row and extra non-zero pixels are sent to the flush function by LVGL

P/S: each printed section is ` printf("\n%d %d %d %d\n", lvgl_img.X, lvgl_img.Y, lvgl_img.Width, lvgl_img.Height);` followed by the first 8 pixels of each row (the LVGL buffer is set to 5 rows, so the flush function is called for every 5 rows):

```
0 0 1448 5
255 255 255 255 255 255 174 94
255 255 255 255 190 30 0 0
255 255 255 126 0 0 3 0
255 255 126 0 0 0 0 0
255 190 0 0 0 0 0 0

0 5 1448 5
255 30 0 0 0 0 0 0
174 0 3 0 0 0 0 0
94 0 0 0 0 0 0 0
30 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0

0 10 1448 5
0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0
0 0 0 0 0 0 0 0
```

kisvegabor, 2024-10-18:
I'm testing it on my end like this:
```c
  hal_init(40, 25); //40x25 display
  //black rounded rectangle on a white background
  lv_obj_set_style_bg_color(lv_screen_active(), lv_color_white(), 0);
  lv_obj_t * obj = lv_obj_create(lv_screen_active());
  lv_obj_set_size(obj, 20, 15);
  lv_obj_set_style_border_width(obj, 0, 0);
  lv_obj_set_style_bg_color(obj, lv_color_black(), 0);
  lv_obj_remove_flag(obj, LV_OBJ_FLAG_SCROLLABLE);
```
and print the frame buffer in the `flush_cb` (DIRECT_MODE):
```
00. FF FF FF EE 7E 2E 00 00 00 00 00 00 00 00 2E 7E EE FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
01. FF FF BE 0E 00 00 00 00 00 00 00 00 00 00 00 00 0E BE FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
02. FF BE 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0E BE FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
03. EE 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0E EE FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
04. 7E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7E FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
05. 2E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2E FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
06. 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
07. 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
08. 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
09. 2E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2E FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
10. 7E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7E FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
11. EE 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0E EE FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
12. FF BE 0E 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0E BE FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
13. FF FF BE 0E 00 00 00 00 00 00 00 00 00 00 00 00 0E BE FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
14. FF FF FF EE 7E 2E 00 00 00 00 00 00 00 00 2E 7E EE FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
15. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
16. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
17. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
18. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
19. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
20. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
21. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
22. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
23. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
24. FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
```

It worked well in partial mode too. So it looks good in both cases. Rendering to a canvas also works. The related CI tests are also passing.  In light of all these I'm quite confident that it's not an LVGL issue.

---

By looking into your logs again, I see the same shifting pattern in yours and mine too. But it's due to the radius of the rectangle. By default it has ~5px radius. 

If rendering is correct, I can only imagine that `Paint_DrawImage` has an off by one issue which was not discovers yet somehow.
To confirm it, pleas try this:
- fill a screen sized buffer with 0xff and send it to `Paint_DrawImage` without LVGL
- fill a 20x20 buffer with 0x00 and send it `Paint_DrawImage` with x = 3; y = 4; w = 20; h = 20 coordinates.

closed 2024-11-15 by lvgl-bot