lvgl/lvgl #6822: Lack Of Pre_Process controlls

NobodyMore, 2024-09-09:
### Introduce the problem

Items like textarea don't have functionality or required events to control changes before them happening. for example if i want to do something when i press LV_KEY_RIGHT at the end of string inside a text area. problem is if i check it in LV_EVENT_READY or LV_EVENT_KEY , then i cannot check if i got to end of string with RIGHT key or was at the end of it before pressing the button. 
doing it through indev will make it super complicated when making a multi page program with a bunch of custom widgets.

TNX for the amazing library.

### Proposal

I have a few suggestions.

1- We really need a event system on groups. i want to have events when adding or removing objects or sending KEY events to its focused child.

2- This one is the way i made it right now. i call an event at start of processing LV_EVENT_KEY inside the lv_textarea.c 

3- double event-ing the events. events can have phases and we should able to detect them and handle them before and after them receiving it.

kisvegabor, 2024-09-12:
Hey,

I think what you need is the `LV_EVENT_KEY | LV_EVENT_PREPROCESS`. This way you can catch the events before the widget processes them. In the event, if needed, you can call `lv_event_stop_processing(e)` to not execute any further events. 

Does it help in your case?

NobodyMore, 2024-09-14:
> I think what you need is the `LV_EVENT_KEY | LV_EVENT_PREPROCESS`. This way you can catch the events before the widget processes them. In the event, if needed, you can call `lv_event_stop_processing(e)` to not execute any further events.

hello., thank you for your time.
i tried this. and it works. since i made this post i did it a little different so i can do some other things like controlling group and focus in a simpler fashion.
i added my text areas under another object and added that to group indev. then i made that object hold the focused object and made it so that inputs were transferred to the text areas from this parent object. this helped me set changes before letting it change.

now im gonna change a few of my changes with LV_EVENT_KEY | LV_EVENT_PREPROCESS since it is more on the convention.

again thanks for the time.

closed 2024-09-14 by NobodyMore

kisvegabor, 2024-09-14:
Thank you for the feedback! 🙂