lvgl/lvgl #6827: rotate180_rgb565 code optimization to yield >= 25% increase in speed

kdschlosser, 2024-09-10:
### Introduce the problem

Here is the existing function that is located in `src/draw/sw/lv_draw_sw.c`

```c
static void rotate180_rgb565(const uint16_t * src, uint16_t * dst, int32_t width, int32_t height, int32_t src_stride,
                             int32_t dest_stride)
{
    if(LV_RESULT_OK == LV_DRAW_SW_ROTATE180_RGB565(src, dst, srcWidth, srcHeight, srcStride, dstStride)) {
        return ;
    }

    src_stride /= sizeof(uint16_t);
    dest_stride /= sizeof(uint16_t);

    for(int32_t y = 0; y < height; ++y) {
        int32_t dstIndex = (height - y - 1) * dest_stride;
        int32_t srcIndex = y * src_stride;
        for(int32_t x = 0; x < width; ++x) {
            dst[dstIndex + width - x - 1] = src[srcIndex + x];
        }
    }
}
```

Now the first thing is I am not even sure how this is even managing to compile without pitching a fit about the `srcWidth`, `srcHeight`, `srcStride` and `dstStride` variables not existing. Those are passed as parameters to the `LV_DRAW_SW_ROTATE180_RGB565` macro.

But never mind that at the moment. I am not sure as to why there is all of this crazy math that is being used in order to rotate the display 180°. This can be simplified using the code below

### Proposal

```c
static void rotate180_rgb565(const uint16_t * src, uint16_t * dst, int32_t width, int32_t height, int32_t src_stride,
                             int32_t dest_stride)
{
    if(LV_RESULT_OK == LV_DRAW_SW_ROTATE180_RGB565(src, dst, srcWidth, srcHeight, srcStride, dstStride)) {
        return ;
    }
    // not sure if these need to be here or not. I believe it might matter if the macro above is not defined in userland
    LV_UNUSED(srcStride);
    LV_UNUSED(dstStride);

    uint32_t num_pixels = height * width;

    for(int32_t i = 0; i < num_pixels; i++) {
        dst[~i + num_pixels] = src[i];
    }
}
```

The code above yields a 25% increase in speed at a minimum. The results of the speed test are with the code converted to python running the test 5000 times over each function. I believe the `~` operator does the same in C code that it does in Python. In Python `~0` is `-1`.

so basically we are taking a  function that has 2 division and 2 addition, 4 subtraction, 2 multiplication done in for loops and reducing it to a bit operation and an addition done in a for loop. There is no need to muck about with the strides because when rotating 180 degrees the strides are going to be identical. all that is being done is the last pixel is being moved to the first pixel location and vice versa through the entire buffer. 

at a 25% speed increase at a minimum this is something that I feel should be done. This can also be done for the ARGB8888 and RGB888 180° functions as well. The RGB888 function will have some more addition in it.

MacDragon, 2024-09-10:
The reason is that the function is not intended to only rotate a whole display frame buffer, but insert a smaller drawn canvas rotated into a larger frame buffer.

kdschlosser, 2024-09-10:
It doesn't matter if you are doing partial updating or not. The strides will always be the same when doing a 180 degree rotation. The buffers must always be the same size in order to do the rotations. 

What would be the use case to do a 180 degree rotation using 2 different sized buffers and having different strides???

MacDragon, 2024-09-10:
lv_draw_sw_rotate called in e.g. SDL and linux FB flush_cb's in library when LV_DISPLAY_RENDER_MODE_PARTIAL with small input buffer and output location in full FB, thus two different strides.

kdschlosser, 2024-09-10:
You failed to notice this...

https://github.com/lvgl/lvgl/blob/d824618366e474eef57081d647031965e6edc136/src/drivers/sdl/lv_sdl_window.c#L226

There is a temporary buffer that gets created that is identical in size to the partial buffer and that temporary buffer is where the rotated data gets copied to. As you can see that buffer is the EXACT same size as the buffer that is holding the data. The same thing happens in the linux fb driver.. 

https://github.com/lvgl/lvgl/blob/d824618366e474eef57081d647031965e6edc136/src/drivers/display/fb/lv_linux_fbdev.c#L274

There is nothing that gets broken from the change above. it will work 100% as it should without any issue.

MacDragon, 2024-09-10:
the lv_area_t describes the output coordinates within the fullsize framebuffer in flush_cb for a partial render, and they match the size of the partial buffer given - that is enough information to work out where to render it.

kdschlosser, 2024-09-10:
Yeah the area that is passed to the flush callback. But that area is never passed to the rotate functions. The full frame buffer is not passed to the rotation functions either. So the data has to be copied from the partial buffer to a temp buffer and after that is done the data from the temp buffer needs to get copied to the full frame buffer,. why not copy the data from the partial buffer directly to the full frame buffer?

MacDragon, 2024-09-11:
I mean that it can be done that way with the information given - in our own flusb_cb that we use it does the a rotated copy straight into FB without any additional buffers, I had assumed that they functions were working the same way, hence not looking at them too carefully.

kdschlosser, 2024-09-11:
Hey if you want to change the code for those rotation functions that will do the rotation and the copy to a larger buffer setting the position of the data properly I am pretty sure you will not see anyone telling you no it won't be accepted. It sure would make things a hell of a lot easier on the user if needing to do a software rotation and not to mention it would also make it a hell of a lot faster too.

kisvegabor, 2024-09-11:
Probably we could rotate directly into the frame buffers too. Just need to go to to the first pixel of the frame buffer (top left corner of the rotated image) and and use it the destination buffer.

kisvegabor, 2024-09-12:
> That's where the 3rd much smaller partial buffer comes into play. 

Seems doable. It's something that can be managed in the flush_cb and not modification to LVGL is needed.

> I know that LVGL has the lv_display_flush_is_last.. what is the purpose of this function? Is it the way to determine if all of the updates have been written? If it is this would be the way to identify when to swap the frame buffers buffers

Exactly!

---

I've opened a PR to update SDL: #6835

closed 2024-09-24 by FASTSHIFT

kdschlosser, 2024-09-24:
Ummm. Dunno why this got closed because it is not yet finished...

There is also the Linux fb driver that needs to be updated in a similar fashion mas the SDL driver.

kisvegabor, 2024-09-25:
It was closed because  #6835 was merged. 

@kdschlosser can you confirm that the rotation update that I added for SDL is a  good direction. If so, I can add the same for fbdev and maybe other too.

kdschlosser, 2024-09-25:
I will check it out a little bit later today. I will run the benchmark using the old code vs the new and see if there is any improvement.

kdschlosser, 2025-07-02:
I wanted to drop some code in here for folks that might want to perform a copy from a partial buffer into a full buffer while handling rotation at the same time. Optionally performing an RGB565 byte swap can be done at the same time as well as dithering for RGB888 and RGB565. All optional pieces are able to be turned on and off at runtime instead of at compile time.

<details>

<summary>rotation.h</summary>

```
#include <stdint.h>
#include <stddef.h>
#include <stdbool.h>
#include <string.h>

#ifndef __ROTATION_H__
    #define __ROTATION_H__

    #define LCD_ROTATION_0    (0)
    #define LCD_ROTATION_90   (1)
    #define LCD_ROTATION_180  (2)
    #define LCD_ROTATION_270  (3)

    typedef struct _lcd_tx_data_t {
        uint8_t *data;
        int x_start;
        int y_start;
        int x_end;
        int y_end;
        uint32_t rotation: 2;
        uint32_t dither: 1;
        uint32_t last_update: 1;
        uint32_t byte_swap: 1;
        uint32_t sw_rotation: 1;
        uint32_t bytes_per_pixel: 4;
        uint32_t dst_width: 11;
        uint32_t dst_height: 11;
    } lcd_tx_data_t;

    void copy_pixels(void *dst, lcd_tx_data_t *tx_data);

#endif
```

</details>

<details>

<summary>rotation.c</summary>

```c

#include <stdint.h>
#include <stddef.h>
#include <stdbool.h>
#include <string.h>

#include "rotation.h"

#define _CALC_THRESHOLD(x, y)  (uint8_t)(((y & 7) << 3) + (x & 7))

uint8_t* red_thresh = NULL;
uint8_t* green_thresh = NULL;
uint8_t* blue_thresh = NULL;

void dither_deinit(void)
{
    if (red_thresh != NULL) m_free(red_thresh);
    if (green_thresh != NULL) m_free(green_thresh);
    if (blue_thresh != NULL) m_free(blue_thresh);

    red_thresh = NULL;
    green_thresh = NULL;
[... 813 lines cut ...]
        case LCD_ROTATION_270:
            uint32_t dst_width_minus_one = dst_width - 1;

            for (uint32_t y = y_start; y < y_end; y++) {
                j = dst_width_minus_one - y;
                i = y * src_bytes_per_line;
                
                for (uint32_t x = x_start; x < x_end; x++) {
                    copy_32bpp(src + (i + x - offset),
                               dst + ((x * dst_width) + j));
                }
            }
            break;

        default:
            LCD_UNUSED(i);
            LCD_UNUSED(j);
            LCD_UNUSED(src_bytes_per_line);
            LCD_UNUSED(offset);
            break;
    }
}
```

</details>

PR #6835 (2024-09-24): feat(sdl): speed up rotation
fixes #6827

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig) [... cut]